blueman-project/blueman #908: Failed to change profile to headset_head_unit

iamrahatul, 2018-08-12:
I was trying to change my audio profile from A2DP to HSP/HFP to check if the latency decreases or not.
But it keeps showing this alert -

> Failed to change profile to headset_head_unit

infirit, 2018-08-12:
You do not provide the basic information we asked for, which distro, bluez and blueman version :disappointed:.

However this profile is provided by the ofono stack, do you have it installed?

arijitthehacker, 2018-08-25:
I am also facing similar kind of issue. I am using Ubuntu 16.04 with latest blueman & bluez. How to check their versions ?

infirit, 2018-08-31:
@arijitthehacker you can run below in a terminal. But without the ofono stack you will not get anywhere.

```
apt-cache show bluez blueman
```

xsellier, 2018-09-05:
Hey guys, I have the very same issue:

```
Package: bluez
Version: 5.50-1
Installed-Size: 4772
Maintainer: Debian Bluetooth Maintainers <team+pkg-bluetooth@tracker.debian.org>
Architecture: amd64
Replaces: bluez-audio (<= 3.36-3), bluez-input, bluez-network, bluez-serial, bluez-utils (<= 3.36-3), udev (<< 170-1)
Depends: libasound2 (>= 1.0.17), libc6 (>= 2.15), libdbus-1-3 (>= 1.9.14), libdw1 (>= 0.127), libglib2.0-0 (>= 2.31.8), libreadline7 (>= 6.0), libudev1 (>= 196), kmod, udev, lsb-base, dbus
Suggests: pulseaudio-module-bluetooth
Conflicts: bluez-audio (<= 3.36-3), bluez-utils (<= 3.36-3)
Breaks: udev (<< 170-1)
Description-en: Bluetooth tools and daemons
 This package contains tools and system daemons for using Bluetooth devices.
 .
 BlueZ is the official Linux Bluetooth protocol stack. It is an Open Source
 project distributed under GNU General Public License (GPL).
Description-md5: ef25d6a9f4a57e78f32faa7b58ef4e59
Multi-Arch: foreign
Homepage: http://www.bluez.org
Tag: admin::hardware, admin::kernel, devel::lang:c, devel::library,
 hardware::TODO, implemented-in::c, protocol::TODO, role::devel-lib,
 role::program, use::driver, use::synchronizing
Section: admin
Priority: optional
Filename: pool/main/b/bluez/bluez_5.50-1_amd64.deb
Size: 1029032
MD5sum: d258a7ab97c68537469389816eeff2ba
SHA256: e906dca050d0ae1b8a3dc65e974f5da4f9f86f416c865a21bf13c0e88fc57675

Package: blueman
Version: 2.1~alpha2-1
Installed-Size: 3058
Maintainer: Christopher Schramm <debian@cschramm.eu>
Architecture: amd64
Depends: libbluetooth3 (>= 4.91), libc6 (>= 2.4), libglib2.0-0 (>= 2.31.8), libpython3.6 (>= 3.6.0~b2), dconf-gsettings-backend | gsettings-backend, python3 (<< 3.7), python3 (>= 3.6~), python3:any (>= 3.3.2-2~), dbus, bluez (>= 5), bluez-obexd, python3-dbus, python3-gi, notification-daemon, librsvg2-common, gnome-icon-theme, libpulse-mainloop-glib0, gir1.2-gtk-3.0, gir1.2-gdkpixbuf-2.0, gir1.2-glib-2.0, gir1.2-pango-1.0, python3-cairo, python3-gi-cairo, gir1.2-appindicator3-0.1
Recommends: policykit-1, pulseaudio-module-bluetooth
Description-en: Graphical bluetooth manager
 Blueman is a GTK+ bluetooth management utility for GNOME using bluez D-Bus
 backend.
Description-md5: 0d53e070801b0cf7f6808bfd0b84a60f
Homepage: https://github.com/blueman-project/blueman
Tag: interface::graphical, interface::x11, role::program, uitoolkit::gtk
Section: x11
Priority: optional
Filename: pool/main/b/blueman/blueman_2.1~alpha2-1_amd64.deb
Size: 722100
MD5sum: 698b575cd6dc9a4573762cc189dad053
SHA256: 1fd4130354ef5361b8f240d4bf0d9a1cd5d7be9cf428181e0555073c15c169b4
```

infirit, 2018-09-09:
We can't really do anything about this as it is not something we can influence from blueman. It requires packages like ofono and possibly configuration. A little background so people understand.

BlueZ5 does not provide audio support (it did in BlueZ4). This has been moved into a framework to allow anyone to register for audio. Pulseaudio supports BlueZ5 and registers with it that it can handle audio, but only a2dp out of the box. Other profiles like hfp/hsp are implemented as a plugin to pulseaudio and provided by ofono.

There is another implementation that uses alsa directly called bluealsa [1] that does not require pulseaudio. It is harder to setup and conflicts with pulseaudio.

Pulseaudio provides documentation what is needed to get hsp/hfp working with it on [2].

Closing but feel free to keep the discussion going and I'll help where I can.

[1] https://github.com/Arkq/bluez-alsa
[2] https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/Bluetooth/

closed 2018-09-09 by infirit

xsellier, 2018-09-09:
@infirit Thank you. This tutoriel helped: https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/Bluetooth/

andrii-pukhalevych, 2021-04-13:
Nice instruction for Ubuntu is here: https://zambrovski.medium.com/using-bluetooth-headset-on-ubuntu-790ce6eecc2