blueman-project/blueman #912: blueman 2.0.6 prompts for root password on login

delxg, 2018-08-21:
blueman: 2.0.6
BlueZ: 5.50
Distribution: Arch Linux
Desktop environment: XFCE/lightdm

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

After upgrading to blueman 2.0.6 I get prompted for my root password when logging in. It seems this is caused by #896. The workaround is [documented](https://github.com/blueman-project/blueman/wiki/PolicyKit).

I would like to request that the default policies be changed from `<allow_active>auth_admin_keep</allow_active>` to `<allow_active>yes</allow_active>`.

My rationale is:
- This would be a more useful default for most users.
- People with special requirements can still lock things down with additional configuration.
- This behaviour would be consistent with NetworkManager, which provides roughly analogous functions to blueman.

Thanks :)

infirit, 2018-08-21:
Hi,

>It seems this is caused by #896

It is and the sole reason for the release.

>The workaround is documented

It is not a workaround, it is a necessary security step for a distribution. It should not allow any users to run dbus services as root.

><allow_active>auth_admin_keep</allow_active> to <allow_active>yes</allow_active>.

These parts run as root and need to be protected so they need auth_admin. NetworkManager does the same and uses auth_admin in a very similar way. On gentoo I actually have a polkit rules file installed for NetworkManager and I am in the correct group so I do not have to enter a password. I suspect you have as well.

You should open a bug with arch so that the maintainer can install a polkit rules for blueman.

delxg, 2018-08-22:
Hi, thanks for your fast reply! :)

I did raise [Arch bug #59736](https://bugs.archlinux.org/task/59736), however Arch has a policy of not deviating from upstream defaults, so this is unlikely to be changed there.

The [default NetworkManager policy](https://gitlab.freedesktop.org/NetworkManager/NetworkManager/blob/master/data/org.freedesktop.NetworkManager.policy.in.in) does not require any group configuration. This is because it uses `<allow_active>yes</allow_active>` which according to the [polkit(8) manpage](https://www.freedesktop.org/software/polkit/docs/latest/polkit.8.html) grants authorisation to "clients in active sessions on local consoles".

I believe that granting access to users with active sessions on a local console is a good default for blueman as well as NetworkManager.

infirit, 2018-08-22:
As @cschramm pointed out it only has yes for some but a lot have auth_admin*.

>I believe that granting access to users with active sessions on a local console is a good default for blueman as well as NetworkManager.

I am not comfortable having this set yes. This part of blueman is doing quite a lot of low level system changes. If a normal user is not allowed to add iptables rules, which blueman does among other things, than we should not allow it as well. If the distro does not want to add polkit rules then it is up to the users like you to decide if you are allowed or not.

I am working on making blueman use NM for all network configuration (NAP, PANU, DUN) and avoid running these low level tools as root. Part is in a PR already the other part is a bit more involved and I am working on it for 2.1.

delxg, 2018-08-23:
Thanks for your replies.

I've looked into it more an learned that the Arch networkmanager package passes `--enable-modify-system` which causes `org.freedesktop.NetworkManager.settings.modify.system` to have `<allow_active>yes</allow_active>`

However NetworkManager also has the concept of "personal" connections. These are connections which are only usable by one person. The policy is hardcoded with `<allow_active>yes</allow_active>`.

My opinion is that a default install of NetworkManager and blueman should be optimised for the common case where all local sessions are trusted.

If you're not comfortable with this then that's fair enough though. We can disagree :)

Thanks for listening.

Zeioth, 2018-08-27:
I'm also experiencing this issue on arch. Waiting for the new release.

UbuntuTarnow, 2018-09-06:
In any other operating system, Ubuntu, Rehat, Windows and Mac OS etc.. to greet new standard users with a prompt for administrator password would be unacceptable.  If they figured out a way to not have the problem, Manajro should also be able to solve the problem.

friederbluemle, 2018-10-07:
Just wondering, what is the recommended solution for normal users? I've been seeing this annoying error when starting blueman-applet for several weeks now. Everything appears to be working normally. I'm on latest Arch Linux. Thanks :bow:

friederbluemle, 2018-10-07:
Nevermind, I figured it out. I just created a new file `/etc/polkit-1/rules.d/90-blueman.rules` with the content over here: https://github.com/blueman-project/blueman/wiki/PolicyKit

cgag, 2018-12-20:
Was dealing with this on fedora 29 as of today and those policykit rules solved it for me, thanks @friederbluemle.

JayeHorn, 2019-05-01:
> Nevermind, I figured it out. I just created a new file `/etc/polkit-1/rules.d/90-blueman.rules` with the content over here: https://github.com/blueman-project/blueman/wiki/PolicyKit

Can someone tell me how to do this exactly?

infirit, 2019-05-01:
@JayeHorn what part of the instructions in the link do you not understand?

JayeHorn, 2019-05-01:
> @JayeHorn what part of the instructions in the link do you not understand?
Was dealing with this on fedora 29 as of today and those policykit rules solved it for me, thanks @friederbluemle.
Nevermind, I figured it out. I just created a new file /etc/polkit-1/rules.d/90-blueman.rules with the content over here: https://github.com/blueman-project/blueman/wiki/PolicyKit

I've tried what friederbluemle put in his reply and was unsuccessful. I have a whole company that uses ubuntu 16.04 and are getting this error. I can't seem to figure out a way to fix this issue. Please help?!?

cschramm, 2019-05-02:
@JayeHorn: That issue does not exist in Ubuntu as 16.04 has blueman 2.0.6 which does not even have that check and later versions in Ubuntu ship correct policies.

infirit, 2019-05-02:
@JayeHorn fedora 29 comes with blueman 2.1 alpha3 which has our example polkit rule in the rpm. You have to be in the wheel group for it to work though. If adding your user to the wheel group does not work please open a new issue for your specific problem.