blueman-project/blueman #913: Failed to apply network settings - Dbus

mrpewpew, 2018-08-24:
blueman: 2.0.6
BlueZ: 5.50-3
Distribution: Arch Linux
Desktop environment: no DE, just using i3wm

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

After updating linux kernel to 4.18 and rebooting, a window appeared with the message : 
`Failed to apply network settings - You might not be able to connect to the Bluetooth network via this machine`

The exception is : 
`org.freedesktop.Dbus.Python.dbus.exceptions.DbusException: Not authorized`

Any idea how to fix it ?

infirit, 2018-08-24:
See #912

closed 2018-08-24 by infirit

mrpewpew, 2018-08-24:
Yes thank you, it worked !

JayeHorn, 2019-05-01:
> Yes thank you, it worked !

How did you get this to work?