blueman-project/blueman #914: create directory

Hinnawy, 2018-08-25:
blueman:
BlueZ:
Distribution: UBUNTU 18.04
Desktop environment:Lenovo-Z50-70

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow

ℹ Please use some service like [Pastebin](http://pastebin.com/) or [GitHub Gist](https://gist.github.com/) to post your logs and keep the thread clean and readable. Make sure to describe the exact steps you took when producing it
I lost the directory for receiving downloads. Trying to configure it using blueman-services I  do not know how to complete the  task. I was only prompted to use blueman services by a message repeatedly appears every time I start the computer.

infirit, 2018-08-26:
Not sure what you tried but you can run ``blueman-services`` from a terminal. Or you can open ``blueman-manager`` and go to the ``view`` menu. The last option is click on the blueman status icon.

It should open the below window and on the Transfer tab you can configure it.

[image]

Hinnawy, 2018-08-27:
This is what I discovered and have already done. Thank you.

closed 2018-08-27 by Hinnawy