blueman-project/blueman #915: Get connection info failed

wdlaszlo, 2018-08-28:
blueman: 2.1.alpha1
BlueZ: 5.50-1
Distribution: Fedora
Desktop environment: i3/lightdm

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I'm trying to run blueman-applet in i3 and I get some exceptions from python. I have added to `/etc/sudoers` a new line with `%wheel ALL=(root) NOPASSWD: /usr/bin/blueman-applet` to run without sudo. 

Logs: https://gist.github.com/wdlaszlo/cf2717eec4bfc2fecf6962f4fbaa466c

wdlaszlo, 2018-08-29:
I ran the applet as root because of `DBusException: Not authorized`. I saw that running applet as sudo vanish this exception. 
```
Failed to apply network settings
You might not be able to connect to the Bluetooth via this machine
g-io-error-quark: GDBus.Error:org.freedesktop.DBus.Python.dbus.exceptions.DBusException: Not authorized (36)
```
https://i.imgur.com/wVc3Kit.png

These are my logs if I ran applet without sudo (without line from sudoers).
https://gist.github.com/wdlaszlo/726a6b2ad3a55cc11f3b2e19896cc632

infirit, 2018-08-29:
The authorization error comes from the blueman-mechanism root dbus service. It is the part of blueman that handles the system configuration that requires root, creating a bridge interface for example.

The checks it does are done through PolicyKit and typically an agent will handle the ui asking for the root password. So, do you have a polkit agent installed? If not, install one and you will be prompted for a password.

However it is more common for distributions to ship a polkit rules file with the package that allow specific user/groups to run these dbus methods. See https://github.com/blueman-project/blueman/wiki/PolicyKit for creating a rules file.

wdlaszlo, 2018-08-30:
Thank you for answer.

After adding a new rule to PolicyKit I was able to run the applet without sudo with no warning.

closed 2018-08-30 by wdlaszlo