blueman-project/blueman #934: Blueman deconnecting from speaker

NeyzanRumi, 2018-10-12:
blueman: 2.1-0.9.alpha2
BlueZ: 5.50
Distribution: Fedora 28
Desktop environment: LXQt

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Blueman deconnects almost every day from a bluetooth speaker and either reconnects after a few minutes by itself or I have to reconnect it manually. See the blueman debuglog at the below link:
https://gist.github.com/NeyzanRumi/088713da39a9058f26735c34de358d44

infirit, 2018-10-12:
Can you open the plugin dialog and go to the powermanager plugin. In there is a configuration setting to auto power the adapter. If it is enabled can you disabled to see if this make a difference?

NeyzanRumi, 2018-10-13:
@infirit I disabled it and now am waiting to see what will happen.

NeyzanRumi, 2018-10-13:
Up until now the device hasn't disconnected and my output is in the below link:
https://gist.github.com/NeyzanRumi/87c60f27dab207b1498e5e7b0ed7facd

I'll see again tomorrow...

infirit, 2018-10-14:
This option will check the killswitch and the whether the adapter is powered and unblock/power on. I see that blueman is registering and setting the killswitch in your first log which appears to cause the disconnection.

If you are confident this workaround works can you post the output of ``rfkill -r list`` with the option disabled and when the device is playing.

NeyzanRumi, 2018-10-14:
The connection crashed today again. See the output in the link below:
https://gist.github.com/NeyzanRumi/ea74c45d910d3e87fe2178f805a9cf70

While the device is playing `rfkill -r list` gives me this:
```
0: ideapad_wlan: Wireless LAN
        Soft blocked: no
        Hard blocked: no
1: ideapad_bluetooth: Bluetooth
        Soft blocked: no
        Hard blocked: no
3: phy0: Wireless LAN
        Soft blocked: no
        Hard blocked: no
5: hci0: Bluetooth
        Soft blocked: no
        Hard blocked: no
```

NeyzanRumi, 2018-11-02:
After upgrading to Fedora 29 the crashes have become more frequent. There are also a bunch of "critical" messages in the debug log, like the following:

```
(blueman-manager:20885): GLib-GIO-CRITICAL **: 11:49:52.760: g_dbus_connection_call_sync_internal: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
Traceback (most recent call last):
  File "/usr/lib64/python3.7/site-packages/gi/overrides/Gio.py", line 296, in __call__
    return self._unpack_result(result)
  File "/usr/lib64/python3.7/site-packages/gi/overrides/Gio.py", line 302, in _unpack_result
    result = result.unpack()
AttributeError: 'NoneType' object has no attribute 'unpack'

(blueman-tray:19774): GLib-GIO-CRITICAL **: 11:50:02.683: g_dbus_connection_call_sync_internal: assertion 'G_IS_DBUS_CONNECTION (connection)' failed
Traceback (most recent call last):
  File "/usr/lib64/python3.7/site-packages/gi/overrides/Gio.py", line 296, in __call__
    return self._unpack_result(result)
  File "/usr/lib64/python3.7/site-packages/gi/overrides/Gio.py", line 302, in _unpack_result
    result = result.unpack()
AttributeError: 'NoneType' object has no attribute 'unpack'
```

See [here](https://gist.github.com/NeyzanRumi/79a39f8c183e3c68e4fd0e1d33de1b2a) the full debug log after the upgrade to Fedora 29 and after a crash occurred.

infirit, 2018-11-02:
So the disconnect happen regardless of autopower being disabled?

The errors are unrelated but I have seen in experiments with the dbus proxy classes. Fedora has a more recent pygobject version than I have which may be related. I'll have a look but fedora 29 ships with very recent software versions and with them bugs.

NeyzanRumi, 2018-11-02:
Yes, blueman disconnects regardless of autopower being enabled or disabled.

infirit, 2018-11-27:
The AttributeErrors are fixed in pygobject 3.30.3

see https://gitlab.gnome.org/GNOME/pygobject/issues/267 for the details.

Carl393, 2019-01-05:
try this, it worked for me
https://superuser.com/questions/924559/wifi-connection-troubles-solved-why-does-my-fix-work

cschramm, 2020-08-01:
@infirit: Not sure why we're tracking this as bug. Anything blueman is doing wrong here? :thinking:

infirit, 2020-08-01:
Don't remember, closing.

closed 2020-08-01 by infirit