blueman-project/blueman #935: Blueman pairs and connects, but does not dial modem

lxs602, 2018-10-12:
Hi,

blueman: 2.0.5
BlueZ: 5.50 (ppa)
Distribution: Ubuntu Budgie 18.04 x86_64
Desktop environment:  Budgie 

I have been attempting to use bluetooth to connect to the modem on my phone. It pairs with blueman and other services seem to work, but not dialling out on the modem. 

The main error I can see seems to be (from syslog/other log):
org.blueman.Mechanism[800]: Unable to init server: Could not connect: Connection refused

I have attached all the logs I can think of below. I hope it is useful.
dpkg -l | grep blue  -  https://pastebin.com/aABmi35W
blueman-manager  -  https://pastebin.com/SNpaw22Y
btmon                    -  https://pastebin.com/TyqthqhH
syslog                    -   https://pastebin.com/rrVKEmev
sdptool-browse     -    https://pastebin.com/JkKTAwqC
lsusb                      -   https://pastebin.com/WQgdJn4W
dmesg                   -    https://pastebin.com/EJV8VKas
journalctl               -    https://pastebin.com/YqbACtfE

I started pairing and dialup at the following times:
Pairing:  	        18:56:48 
DUN connect:  	18:57:10 

Journalctl didn't log for about 15 minutes from about 18:52, I have no idea why. The same error as above occurs at 20:14:25:
org.blueman.Mechanism[821]: Unable to init server: Could not connect: Connection refused

I am able to use **minicom** (actually cutecom) to send 'ATZ' to the modem, and I get:

OK

Bluetooth dial up works on Windows, which I am using to write this now unfortunately.

I'd be grateful for whatever you can suggest.

Thanks folks.

infirit, 2018-10-12:
Yes, DUN is broken in 2.0. Neither of us had phones that had DUN so had no way to test.

I found my old n900 phone which has DUN so I have been fixing thing in master. If you are able (and willing) to apply patches I could provide a single patch that makes it work for me.

lxs602, 2018-10-13:
Hi yes I'm very happy to add patches. I can buy you a test phone on if you like... :) It would have to be an old one though

infirit, 2018-10-14:
Thanks for the offer, I have a software solution now that emulates a bluetooth modem on any android device. Its called BlueDUN+. It does not require root or any configuration changes.

I'll prepare a 2.0 branch for the dun fixes, test a bit and when ready post a patch here.

infirit, 2018-10-31:
Unfortunately been busy and there are a lot of fixes that went into master for serial/rfcomm, dun and python3 that are not in 2.0. They are not difficult to backport but takes time as master has diverged somewhat from 2.0 (so not trivial as I intentionally thought :disappointed:).

@cschramm, happy to do this but then it would be nice if I can push this into 2.0 and prepare a new release. We have a willing victim for testing patches :smiling_imp:.

infirit, 2018-11-01:
The blueman dun code works which you can test with https://github.com/blueman-project/blueman/pull/939 (add ``.patch`` url for a patch). It dials and my machine uses the phone for internet.

The NetworkManager bits are not done yet which I'll work on in the next couple of days. But you are using blueman's dun so please have a go and let me know if it works.

lxs602, 2018-11-05:
Okey coke. Will try it out.

infirit, 2018-11-05:
The NM part works for me now as well.

lxs602, 2018-11-06:
Hi, I get the message below when I try to patch. Is it anything significant?

luke@luke-Satellite-L635:/usr/local/blueman-master$ sudo patch -p1 < 939.patch 
patching file blueman/plugins/applet/PPPSupport.py
Hunk #1 FAILED at 31.
1 out of 1 hunk FAILED -- saving rejects to file blueman/plugins/applet/PPPSupport.py.rej
patching file blueman/plugins/applet/PPPSupport.py
Hunk #1 FAILED at 39.
1 out of 1 hunk FAILED -- saving rejects to file blueman/plugins/applet/PPPSupport.py.rej
patching file blueman/main/PPPConnection.py
Reversed (or previously applied) patch detected!  Assume -R? [n]

lxs602, 2018-11-06:
Also this, after continuing after the message above anyway:
(Names and email redacted with ****)

> |From 222d0532cd9164500056cd56fa6542ba17f94ad0 Mon Sep 17 00:00:00 2001
> |From: *********** <*********@*********>
> |Date: Tue, 30 Oct 2018 21:52:49 +0100
> |Subject: [PATCH 09/13] mechanism Rfcomm: Decode output to codepoints
> |
> |---
> | blueman/plugins/mechanism/Rfcomm.py.in | 2 +-
> | 1 file changed, 1 insertion(+), 1 deletion(-)
> |
> |diff --git a/blueman/plugins/mechanism/Rfcomm.py.in b/blueman/plugins/mechanism/Rfcomm.py.in
> |index 867b1103..6d49f2fb 100644
> |--- a/blueman/plugins/mechanism/Rfcomm.py.in
> |+++ b/blueman/plugins/mechanism/Rfcomm.py.in
> --------------------------

lxs602, 2018-11-06:
Sorry, didn't give the full message:

1 out of 1 hunk FAILED -- saving rejects to file blueman/main/PPPConnection.py.rej
can't find file to patch at input line 989
Perhaps you used the wrong -p or --strip option?
The text leading up to this was:
--------------------------
|
|From 222d0532cd9164500056cd56fa6542ba17f94ad0 Mon Sep 17 00:00:00 2001
|From: ********************** <**********@***********>
|Date: Tue, 30 Oct 2018 21:52:49 +0100
|Subject: [PATCH 09/13] mechanism Rfcomm: Decode output to codepoints
|
|---
| blueman/plugins/mechanism/Rfcomm.py.in | 2 +-
| 1 file changed, 1 insertion(+), 1 deletion(-)
|
|diff --git a/blueman/plugins/mechanism/Rfcomm.py.in b/blueman/plugins/mechanism/Rfcomm.py.in
|index 867b1103..6d49f2fb 100644
|--- a/blueman/plugins/mechanism/Rfcomm.py.in
|+++ b/blueman/plugins/mechanism/Rfcomm.py.in
--------------------------
File to patch:

lxs602, 2018-11-06:
I haven't done lots of patching, so I hope the above are not silly problems.

infirit, 2018-11-06:
>silly oversights of mine...

Not silly but maybe an oversight :smile:

It looks like you are trying to apply the patch against master. The patch however is against the current stable 2.0 so you can patch the version that is in ubuntu. I am not too familiar with debian packages but you should be able to do something like https://cs-people.bu.edu/doucette/xia/guides/debian-patch.txt

I would recommend patching the package in ubuntu as installing blueman correctly from source can be tricky.

lxs602, 2018-11-09:
I have tested it for about 1/2 hour and it is working well, with network-manager integration.

Are there any other conditions or new features to be tested?

infirit, 2018-11-09:
>working well, with network-manager integration

Thanks for testing :+1: 

>Are there any other conditions

If you can try changing to blueman's dun by changing it in "Local Services".

lxs602, 2018-11-10:
Blueman's dun fails, unfortunately. I will attach syslog and journalctl below. Any other logs you need (btmon)?

lxs602, 2018-11-10:
**journatlctl -r**

Nov 10 17:37:05 luke-Satellite-L635 gedit[6893]: remote volume monitor with dbus name org.gtk.vfs.GoaVolumeMonitor is not supported
Nov 10 17:37:05 luke-Satellite-L635 gvfs-goa-volume[2292]: monitor says it's not supported
Nov 10 17:37:04 luke-Satellite-L635 bluetoothd[860]: src/adapter.c:resume_discovery()
Nov 10 17:37:04 luke-Satellite-L635 bluetoothd[860]: src/device.c:device_bonding_failed() status 14
Nov 10 17:37:04 luke-Satellite-L635 bluetoothd[860]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Nov 10 17:37:04 luke-Satellite-L635 bluetoothd[860]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 00:19:2D:04:02:64 type 0 status 0xe
Nov 10 17:37:04 luke-Satellite-L635 bluetoothd[860]: plugins/policy.c:disconnect_cb() reason 2
Nov 10 17:37:04 luke-Satellite-L635 bluetoothd[860]: src/adapter.c:adapter_remove_connection()
Nov 10 17:37:04 luke-Satellite-L635 bluetoothd[860]: src/adapter.c:dev_disconnected() Device 00:19:2D:04:02:64 disconnected, reason 2
Nov 10 17:37:02 luke-Satellite-L635 ModemManager[865]: <info>  (tty/rfcomm0): released by modem /sys/devices/virtual/tty/rfcomm0
Nov 10 17:37:02 luke-Satellite-L635 org.blueman.Applet[2027]: rfcomm channel: 1
Nov 10 17:37:02 luke-Satellite-L635 nautilus-autostart.desktop[2351]: RuntimeError: object at 0x7f26755a1730 of type FolderColorMenu is not initialized
Nov 10 17:37:01 luke-Satellite-L635 plank.desktop[2353]: RuntimeError: object at 0x7f34eae9a780 of type FolderColorMenu is not initialized
Nov 10 17:34:49 luke-Satellite-L635 whoopsie[1637]: [17:34:49] Not online; processing later (/var/crash/_usr_bin_blueman-applet.1000.crash).
Nov 10 17:34:49 luke-Satellite-L635 systemd[1]: Started Process error reports when automatic reporting is enabled.
Nov 10 17:34:49 luke-Satellite-L635 whoopsie-upload-all[5324]: All reports processed
Nov 10 17:34:49 luke-Satellite-L635 whoopsie-upload-all[5324]: /var/crash/_usr_bin_software-properties-gtk.1000.crash already marked for upload, skippi
Nov 10 17:34:49 luke-Satellite-L635 whoopsie-upload-all[5324]: /var/crash/_usr_bin_budgie-polkit-dialog.1000.crash already marked for upload, skipping
Nov 10 17:34:49 luke-Satellite-L635 whoopsie-upload-all[5324]: Marking /var/crash/_usr_bin_blueman-applet.1000.crash for whoopsie upload
Nov 10 17:34:49 luke-Satellite-L635 whoopsie-upload-all[5324]: Collecting info for /var/crash/_usr_bin_blueman-applet.1000.crash...
Nov 10 17:34:49 luke-Satellite-L635 whoopsie-upload-all[5324]: /var/crash/_usr_bin_software-properties-gtk.0.crash already marked for upload, skipping
Nov 10 17:34:49 luke-Satellite-L635 whoopsie-upload-all[5324]: /var/crash/_usr_bin_nautilus.1000.crash already marked for upload, skipping
Nov 10 17:34:49 luke-Satellite-L635 whoopsie-upload-all[5324]: /var/crash/_usr_bin_budgie-daemon.1000.crash already marked for upload, skipping
Nov 10 17:34:49 luke-Satellite-L635 whoopsie-upload-all[5324]: Marking /var/crash/_usr_lib_blueman_blueman-mechanism.0.crash for whoopsie upload
Nov 10 17:34:49 luke-Satellite-L635 whoopsie-upload-all[5324]: Collecting info for /var/crash/_usr_lib_blueman_blueman-mechanism.0.crash...
Nov 10 17:34:49 luke-Satellite-L635 whoopsie-upload-all[5324]: /var/crash/_usr_bin_xfce4-power-manager.1000.crash already marked for upload, skipping
Nov 10 17:34:36 luke-Satellite-L635 whoopsie[1637]: [17:34:36] Not online; processing later (/var/crash/_usr_lib_blueman_blueman-mechanism.0.crash).
Nov 10 17:33:45 luke-Satellite-L635 NetworkManager[884]: <info>  [1541871225.7891] ignoring modem 'rfcomm0' (no associated Bluetooth device)
Nov 10 17:33:45 luke-Satellite-L635 kernel: Bluetooth: TIOCGSERIAL is not supported
Nov 10 17:33:45 luke-Satellite-L635 ModemManager[865]: <info>  Modem: state changed (unknown -> disabled)
Nov 10 17:33:45 luke-Satellite-L635 ModemManager[865]: <info>  Modem for device at '/sys/devices/virtual/tty/rfcomm0' successfully created
Nov 10 17:33:45 luke-Satellite-L635 kernel: Bluetooth: TIOCGSERIAL is not supported
Nov 10 17:33:45 luke-Satellite-L635 kernel: Bluetooth: TIOCGSERIAL is not supported
Nov 10 17:33:45 luke-Satellite-L635 ModemManager[865]: <info>  Creating modem with plugin 'Nokia' and '1' ports
Nov 10 17:33:41 luke-Satellite-L635 systemd[1]: Starting Process error reports when automatic reporting is enabled...
Nov 10 17:33:41 luke-Satellite-L635 blueman-mechanism[5288]: TypeError: 'map' object is not subscriptable
Nov 10 17:33:41 luke-Satellite-L635 blueman-mechanism[5288]:     raise PPPException("Bad modem response %s, expected CONNECT" % response[0])
Nov 10 17:33:41 luke-Satellite-L635 blueman-mechanism[5288]:   File "/usr/lib/python3/dist-packages/blueman/main/PPPConnection.py", line 93, in connect
Nov 10 17:33:41 luke-Satellite-L635 blueman-mechanism[5288]:     self.commands[item_id][1](response)
Nov 10 17:33:41 luke-Satellite-L635 blueman-mechanism[5288]:   File "/usr/lib/python3/dist-packages/blueman/main/PPPConnection.py", line 100, in __cmd_
Nov 10 17:33:41 luke-Satellite-L635 blueman-mechanism[5288]:     callback(ret, exception, *user_data)
Nov 10 17:33:41 luke-Satellite-L635 blueman-mechanism[5288]:   File "/usr/lib/python3/dist-packages/blueman/main/PPPConnection.py", line 230, in on_don
Nov 10 17:33:41 luke-Satellite-L635 blueman-mechanism[5288]:     on_done(lines, None)
Nov 10 17:33:41 luke-Satellite-L635 blueman-mechanism[5288]:   File "/usr/lib/python3/dist-packages/blueman/main/PPPConnection.py", line 216, in on_dat
[... cut ...]
Nov 10 17:31:50 luke-Satellite-L635 plank.desktop[2353]: RuntimeError: object at 0x7f34eae9a780 of type FolderColorMenu is not initialized
Nov 10 17:31:50 luke-Satellite-L635 plank.desktop[2353]: RuntimeError: object at 0x7f34eae9a780 of type FolderColorMenu is not initialized
Nov 10 17:31:49 luke-Satellite-L635 plank.desktop[2353]: RuntimeError: object at 0x7f34eae9a780 of type FolderColorMenu is not initialized
Nov 10 17:31:49 luke-Satellite-L635 plank.desktop[2353]: ('file scheme: ', 'file')
Nov 10 17:31:49 luke-Satellite-L635 plank.desktop[2353]: ('Handling file: ', 'file:///media/luke/FD88-8964/Bluetoothlog')

lxs602, 2018-11-10:
SYSLOG

Nov 10 17:32:48 luke-Satellite-L635 org.blueman.Mechanism[882]: Unable to init server: Could not connect: Connection refused
Nov 10 17:32:48 luke-Satellite-L635 org.blueman.Mechanism[882]: Unable to init server: Could not connect: Connection refused
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechanism: Starting blueman-mechanism
Nov 10 17:32:48 luke-Satellite-L635 dbus-daemon[882]: [system] Successfully activated service 'org.blueman.Mechanism'
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechani[5288]: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechanism: loading Rfcomm
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechanism: loading RfKill
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechanism: loading Network
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechanism: loading Ppp
Nov 10 17:32:48 luke-Satellite-L635 org.blueman.Mechanism[882]: Unable to init server: Could not connect: Connection refused
Nov 10 17:32:49 luke-Satellite-L635 kernel: [  382.694929] Bluetooth: TIOCGSERIAL is not supported
Nov 10 17:32:48 luke-Satellite-L635 org.blueman.Mechanism[882]: Unable to init server: Could not connect: Connection refused
Nov 10 17:32:57 luke-Satellite-L635 ModemManager[865]: <info>  Creating modem with plugin 'Nokia' and '1' ports
Nov 10 17:32:57 luke-Satellite-L635 ModemManager[865]: <info>  Modem for device at '/sys/devices/virtual/tty/rfcomm0' successfully created
Nov 10 17:32:57 luke-Satellite-L635 kernel: [  390.966831] Bluetooth: TIOCGSERIAL is not supported
Nov 10 17:32:57 luke-Satellite-L635 kernel: [  390.967003] Bluetooth: TIOCGSERIAL is not supported
Nov 10 17:32:58 luke-Satellite-L635 ModemManager[865]: <info>  Modem: state changed (unknown -> disabled)
Nov 10 17:32:58 luke-Satellite-L635 kernel: [  391.790572] Bluetooth: TIOCGSERIAL is not supported
Nov 10 17:32:58 luke-Satellite-L635 NetworkManager[884]: <info>  [1541871178.7949] ignoring modem 'rfcomm0' (no associated Bluetooth device)
Nov 10 17:33:14 luke-Satellite-L635 ModemManager[865]: <info>  (tty/rfcomm0): released by modem /sys/devices/virtual/tty/rfcomm0
Nov 10 17:33:15 luke-Satellite-L635 org.blueman.Applet[2027]: rfcomm channel: 1
Nov 10 17:33:16 luke-Satellite-L635 bluetoothd[860]: src/adapter.c:dev_disconnected() Device 00:19:2D:04:02:64 disconnected, reason 2
Nov 10 17:32:48 luke-Satellite-L635 org.blueman.Mechanism[882]: Unable to init server: Could not connect: Connection refused
Nov 10 17:32:48 luke-Satellite-L635 org.blueman.Mechanism[882]: Unable to init server: Could not connect: Connection refused
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechanism: Starting blueman-mechanism
Nov 10 17:32:48 luke-Satellite-L635 dbus-daemon[882]: [system] Successfully activated service 'org.blueman.Mechanism'
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechani[5288]: gtk_icon_theme_get_for_screen: assertion 'GDK_IS_SCREEN (screen)' failed
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechanism: loading Rfcomm
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechanism: loading RfKill
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechanism: loading Network
Nov 10 17:32:48 luke-Satellite-L635 blueman-mechanism: loading Ppp
Nov 10 17:32:48 luke-Satellite-L635 org.blueman.Mechanism[882]: Unable to init server: Could not connect: Connection refused
Nov 10 17:32:49 luke-Satellite-L635 kernel: [  382.694929] Bluetooth: TIOCGSERIAL is not supported
Nov 10 17:32:48 luke-Satellite-L635 org.blueman.Mechanism[882]: Unable to init server: Could not connect: Connection refused
Nov 10 17:32:57 luke-Satellite-L635 ModemManager[865]: <info>  Creating modem with plugin 'Nokia' and '1' ports
Nov 10 17:32:57 luke-Satellite-L635 ModemManager[865]: <info>  Modem for device at '/sys/devices/virtual/tty/rfcomm0' successfully created
Nov 10 17:32:57 luke-Satellite-L635 kernel: [  390.966831] Bluetooth: TIOCGSERIAL is not supported
Nov 10 17:32:57 luke-Satellite-L635 kernel: [  390.967003] Bluetooth: TIOCGSERIAL is not supported
Nov 10 17:32:58 luke-Satellite-L635 ModemManager[865]: <info>  Modem: state changed (unknown -> disabled)
Nov 10 17:32:58 luke-Satellite-L635 kernel: [  391.790572] Bluetooth: TIOCGSERIAL is not supported
Nov 10 17:32:58 luke-Satellite-L635 NetworkManager[884]: <info>  [1541871178.7949] ignoring modem 'rfcomm0' (no associated Bluetooth device)
Nov 10 17:33:14 luke-Satellite-L635 ModemManager[865]: <info>  (tty/rfcomm0): released by modem /sys/devices/virtual/tty/rfcomm0
Nov 10 17:33:15 luke-Satellite-L635 org.blueman.Applet[2027]: rfcomm channel: 1
Nov 10 17:33:16 luke-Satellite-L635 bluetoothd[860]: src/adapter.c:dev_disconnected() Device 00:19:2D:04:02:64 disconnected, reason 2
Nov 10 17:33:16 luke-Satellite-L635 bluetoothd[860]: src/adapter.c:adapter_remove_connection()
Nov 10 17:33:16 luke-Satellite-L635 bluetoothd[860]: plugins/policy.c:disconnect_cb() reason 2
Nov 10 17:33:16 luke-Satellite-L635 bluetoothd[860]: src/adapter.c:bonding_attempt_complete() hci0 bdaddr 00:19:2D:04:02:64 type 0 status 0xe
Nov 10 17:33:16 luke-Satellite-L635 bluetoothd[860]: src/device.c:device_bonding_complete() bonding (nil) status 0x0e
Nov 10 17:33:16 luke-Satellite-L635 bluetoothd[860]: src/device.c:device_bonding_failed() status 14
Nov 10 17:33:16 luke-Satellite-L635 bluetoothd[860]: src/adapter.c:resume_discovery()
Nov 10 17:33:31 luke-Satellite-L635 bluetoothd[860]: src/adapter.c:connected_callback() hci0 device 00:19:2D:04:02:64 connected eir_len 11
[... cut ...]
Nov 10 17:37:38 luke-Satellite-L635 dbus-daemon[2027]: [session uid=1000 pid=2027] Activating service name='org.freedesktop.thumbnails.Thumbnailer1' requested by ':1.71' (uid=1000 pid=2343 comm="/usr/lib/tracker/tracker-miner-fs " label="unconfined")
Nov 10 17:37:38 luke-Satellite-L635 tracker-miner-f[2343]: Could not set mount point in database 'urn:nepomuk:datasource:ACFC3030FC2FF2EE', GDBus.Error:org.freedesktop.Tracker1.SparqlError.Internal: UNIQUE constraint failed: nie:DataObject.nie:url
Nov 10 17:37:38 luke-Satellite-L635 org.freedesktop.thumbnails.Thumbnailer1[2027]: Registered thumbailer /usr/bin/totem-video-thumbnailer -s %s %u %o

lxs602, 2018-11-10:
That's not so easy to read. I have uploaded them to pastebin as well:

SYSLOG:  https://pastebin.com/jzf256Pa
JOURNALCTL: https://pastebin.com/zRTLMPam

infirit, 2018-11-10:
Thanks, yeah a pastebin is usually a good idea.

There is only one ``map`` call in PPPConnection and its behaviour changed in python3. Below hopefully will take care of the error. I do expect the connection to still fail though as this error is hiding it.

```diff
diff --git a/blueman/main/PPPConnection.py b/blueman/main/PPPConnection.py
index 771a82d3..e64c17a7 100644
--- a/blueman/main/PPPConnection.py
+++ b/blueman/main/PPPConnection.py
@@ -210,7 +210,7 @@ class PPPConnection(GObject.GObject):
 
         if found:
             lines = filter(lambda x: x != "", lines)
-            lines = map(lambda x: x.strip("\r\n"), lines)
+            lines = list(map(lambda x: x.strip("\r\n"), lines))
             dprint("<-- ", lines)
 
             on_done(lines, None)
```

lxs602, 2018-11-20:
Sorry for the slow reply, as something important came up.

As before, connection failed with blueman's DUN, then succeeded with network-manager.

blueman-manager:     https://pastebin.com/PmKpV7Aq
journalctl -r:                https://pastebin.com/HSQZnp76
syslog:                        https://pastebin.com/G00bG8xQ

If newer phones don't use DUN, what do they use instead for tethering?

infirit, 2018-12-08:
@lxs602 not sure what is going on but thanks for trying out blueman dun support. It works for me :disappointed:. 

>If newer phones don't use DUN, what protocol do they use instead? I have never owned a smartphone.

Newer phones use a PAN connection. This works much simpler with a bridge connection that is registered with BlueZ. These PAN connection also show up automatically in NetworkManager.

We merged the two PR to master and 2-0-stable branches. I like to keep this issue open as blueman's DUN fails for you.

lxs602, 2018-12-10:
Are there any other logs I can send? I can change dependency versions, or recompile them, etc., if it would help. 

Perhaps I should have said, if it is relevant, but the bluetooth modem is external USB.

It is not a very big problem, though, so thank you for the reply.

PS. I bought a slightly newer laptop recently, with internal bluetooth. Blueman does not load, though I am sure it did at first... I have submitted a separate bug report below. 

https://github.com/blueman-project/blueman/issues/963

closed 2022-09-05 by github-actions[bot]