blueman-project/blueman #940: Allow applet actions without user password

squalou, 2018-11-12:
blueman: 2.0.6-1
BlueZ: 5.50-3
Distribution: Arch
Desktop environment: XFCE

My issue is probably more a lack of knowledge / documentation not found, so first of all I apoligize if it's tha case.

Everything works fine, but it's a bit annoying :
- when my xfce essions starts blueman-applet starts ... and asks for my passwors
- when I turn BT on (or off) : again it asks for my password.

Maybe I can tweak that with some sudo configuration of some sort ?

I've grown used to gnome integration where the password is not asked, so there's probably something doable.

Thanks for any advice,

S.

squalou, 2018-11-12:
A few more details just in case : 
dialog pop up whith these actions marked as asking for privielges : 
```
org.blueman.network.setup
org.blueman.rfkill.setstate
```
I already tried  somehting like this in /etc/sudoers :
```
user host = (root) NOPASSWD: /usr/bin/rfkill

```

but as I expected it .. it works to call sudo rfkill without password, but has no direct effect on the applet.

infirit, 2018-11-12:
See https://wiki.archlinux.org/index.php/Blueman#Permissions.

I would suggest to undo your other changes as they are not required if you follow above.

Closing but feel free to comment further :smile:

closed 2018-11-12 by infirit

squalou, 2018-11-12:
last comment ... it works out of the box on xubuntu so I'm obviously missing something ;)