blueman-project/blueman #941: Audio sink connected to wrong pulseaudio instance

reitzig, 2018-11-22:
blueman: 2.0.5 (via )
BlueZ: 5.48
Distribution: Ubuntu 18.04 LTS
Desktop environment: gdm3, i3 4.14.1

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I have two graphical user sessions at the same time. Each gets its own pulseaudio:

```
# ps -ef | grep pulse
gdm       1346  1287  0 Nov21 ?        00:00:03 /usr/bin/pulseaudio --daemonize=no
user1   8690     1  0 00:42 ?        00:00:03 /usr/bin/pulseaudio --start --log-target=syslog
user2     10683     1  1 08:59 ?        00:00:36 /usr/bin/pulseaudio --start --log-target=syslog
```

Logged in as `user2`, I start `blueman-applet` after killing all processes related to blueman:

```
# ps -ef | grep blue
root       998     1  0 Nov21 ?        00:00:00 /usr/lib/bluetooth/bluetoothd
user2     11773  9450  0 09:19 ?        00:00:00 /usr/lib/bluetooth/obexd
user2     14069 14007  0 09:46 ?        00:00:00 /usr/bin/python3 /usr/bin/blueman-applet
```

I connect to a Bluetooth speaker. It does not show up as audio sink for `user2` (checked in `pavucontrol`), but it _does_ for `user1`.

I have used this speaker with both accounts before; in this case, the session of `user1` is older.

infirit, 2018-11-24:
Blueman only handles the bluetooth connection but not how the audio bits are handled (except selecting a profile). Pulseaudio and BlueZ handle it between them, but..

If I remember correctly only one client can connect to BlueZ as an audio transport. What you looking for can only work if you run Pulseaudio systemwide as root.

reitzig, 2018-12-02:
@infirit 	So, in your opinion, should I open an issue with Pulseaudio or Bluez? (The latter doesn't seem to have a tracker? Ugh...)

cschramm, 2019-03-30:
@reitzig: Did you take further steps? I recommend filing an issue with pulseaudio, in case you haven't done that. I think they *do* support multiple users by using the system-wide mode, however at least [their wiki page](https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/SystemWide/) recommends disabling module loading and thus bluethooth support in that case for security reasons.

closed 2019-06-05 by stale[bot]

reitzig, 2019-06-09:
@cschramm Sorry, I did not. The use case went away for me.

As I've stated above, pulseaudio works just fine with multiple users: I can switch between logged-in users and get audio in all of them. That's an out-of-the-box Ubuntu configuration, iirc. Imho, something on the bluetooth end of things needs to either be able to connect to multiple pulseaudio instances, or at least choose better which one to connect to (note user2 started bluetooth, but user1 got the connection).