blueman-project/blueman #944: only mac address showing; no device names

bennowak, 2018-11-28:
blueman:
BlueZ:
Distribution: Debian 9
Desktop environment: XFCE

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

bennowak, 2018-11-28:
Names of devices all show as "UNKNOWN".  Only MAC addresses are shown.

infirit, 2018-11-28:
For us to start we need logs for of blueman-applet and possibly blueman-manager. See https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging for how to do this.

bennowak, 2018-12-01:
Why does it say that I "closed" this?  I did no such thing!

bennowak, 2018-12-01:
How do I get the logs of blueman-manager?

cschramm, 2018-12-01:
We probably don't need logs from `blueman-manager`. `blueman-applet` should be enough. But did you actually see such devices when you recorded it? You probably need to run a discovery.

bennowak, 2018-12-08:
Sorry haven't had time lately, but will update soon.

jpmorris, 2019-01-30:
Anyone else who is having this issue: double-check you are in paring mode in  yoru device.

The entire interaction is somewhat confusing

- If a user paired a first for the first time with blueman-applet the device might default into paring mode and they see their device listed.
- That is just added confusion, but even if they dont start with a new device, the entire screen is filled with LE devices, as was mentioned above. (Unlike, say windows, OSX, iOS, Android).  So the user assumes that the developers haven't worked in the names or there's a bug.  The user wonders: which one of these 30 (in my case) mac addresses is my device?  In actuality the user didn't set the device to paring mode (by holding the button down even longer past the point that device powers on) and these are just LE devices.

It would be nice to have a UI solution similar to what is seen in almost any other operating system.

infirit, 2019-02-02:
We rely on the high level dbus api provided by BlueZ. In the api we can read several properties on a device and we use the Alias to show the name in the list. For these devices BlueZ set this to the device mac but with dashes instead of colons. It is a limitation of the dbus api.

BlueZ has been adding more properties we could use, just needs someone to figure out how to convert the byte array data to something useful.

closed 2019-04-27 by stale[bot]

arjk4259, 2020-04-19:
I have the same trouble with Linux XXXXX 5.3.0-46-generic #38~18.04.1-Ubuntu SMP Tue Mar 31 04:17:56 UTC 2020 x86_64 x86_64 x86_64 GNU/Linux

maniac0s, 2020-05-05:
I have the same problem. Why can't I see any names. I don't know the MAC address of my headphones and I have 10 devices listed here. How is it possible to find the right dame?

infirit, 2020-05-05:
If you only see a bt address then the device does not provide a Name. The only reason you see them is because BlueZ guarantees in the dbus api that the device Alias is always populated. 

Phones usually hide these devices because you are not meant to find them. So I opened #1198 to remind me to implement hiding them in blueman. If people want to contribute to the discussion or work on it  please go to #1198.

arjk4259, 2020-05-06:
> I have the same problem. Why can't I see any names. I don't know the MAC address of my headphones and I have 10 devices listed here. How is it possible to find the right name?

Upgrading to Ubuntu fixed the problem for me. It is possible the Bluetooth manager included with Ubuntu’s version 20.04 Lts fixes it.

jsperezg, 2020-06-25:
No, it doesn't. It's BS. Just a list of mac addresses.

maniac0s, 2020-07-06:
Same here still on latest Debian

yilmazbingo, 2020-12-22:
I have the same issue
[image]

cschramm, 2020-12-23:
It's still the same advice here: If all devices show up with addresses only, most probably none of them is pairable. #1258 hides devices with no name but the result of that will be that nothing will be found at all unless you're dealing with a very exotic case where the address is actually provided as the name of a pairable device. You can look up the OUIs, i.e. the first three bytes, of the addresses to find the manufacturer and possibly identify the device you're looking for, but it's either not even in the list as you did not discover it or it is but it's not pairable, so that you cannot do anything with it.

yilmazbingo, 2020-12-26:
@cschramm  I have 2 laptops on my desk. one is mac, and other one kali linux. my iphone and headphones are discoverable by macbook, I can pair with mac, but it wont be pairable with kali linux? I dont think that it is because they are pairable. 

Your advice "You can look up the OUIs, i.e. the first three bytes, of the addresses to find the manufacturer and possibly identify the device you're looking for" will be helpful. Thanks

dm17, 2021-05-29:
I was having this but perhaps a lot more bluetooth devices are visible in Linux that other platforms... When I got the device into pairing mode, then its name did start showing in blueman.

Neo1968, 2021-08-28:
> It's still the same advice here: If all devices show up with addresses only, most probably none of them is pairable. #1258 hides devices with no name but the result of that will be that nothing will [...]

That's about as helpful as horshoes on a dog.  The bluetooth names show up using other devices and under other operating systems.  The names were showing up under ubuntu's unity interface, but won't show under kubuntu's kde.  So the devices I'm looking for are pairable.

cschramm, 2021-08-28:
@Neo1968

> That's about as helpful as horshoes on a dog.

:thought_balloon:

> The names were showing up under ubuntu's unity interface, but won't show under kubuntu's kde.

I suppose with KDE you mean bluedevil. Where does blueman come into play?

Neo1968, 2021-08-28:
I just login to the unity desktop, find my device, logout, login to plasma and boom, its there. That works for me.

cschramm, 2021-08-29:
@Neo1968: What you describe does not seem to have anything to do with blueman. It sounds like you're reporting a bluedevil issue. You're posting to a thread in the blueman project.

Neo1968, 2021-08-29:
I did a search for Linux not showing bluetooth device names and i got this thread.

geos-github, 2023-08-13:
Setup: PC with up-to-date Manjaro XFCE, Blueman-Manager 2.3.5. Bluetooth USB adapter, lsusb says "Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)".
When I perform a search in Blueman-Manager the Bluetooth Dongle is activated (green LED is on). When I put an old Android phone with BT turned on and set to temporarily discoverable near the BT adapter, it is being found by Blueman manager and correctly displayed as a device type smartphone, but instead of its name its  Bluetooth MAC address is displayed. This happens with other Bluetooth devices, too (e.g. a Mac, which is correctly recognized as desktop computer), so it looks like it is not specific to the device found.
Interestingly I get the following message in dmesg:

Bluetooth: hci0: unexpected cc 0x041a length: 7 > 1

This happens for every search which hits an "unknown" device.

Interestingly, for device (also an Android phone) which has been paired before, Blueman-manager search does display the full device name and the above message does not show up in dmesg, even when the pairing has been removed in the meantime on the phone. That device (including its name) remains stored in /var/lib/bluetooth/(adapter's MAC)/cache/, so I guess Blueman-manager simply reads out the name from that cache insteady of trying to query it which somehow fails.
After deleting the entry from the cache a Blueman-manager search on that previously paired phone again displays it with its MAC only, and the error message is added to dmesg.

The key question is thus: what does the above error message mean, i.e. what is going wrong when trying to retrieve a newly discovered device's name?