blueman-project/blueman #945: unable to suspend to ram while connected to wireless headphones

treaki, 2018-11-29:
blueman:blueman-manager version 2.0.5 starting

BlueZ:
Source: bluez (5.47-1)
Version: 5.47-1+b1

Distribution:
~$ lsb_release -a
No LSB modules are available.
Distributor ID:	Debian
Description:	Debian GNU/Linux 7.11 (wheezy)
Release:	7.11
Codename:	wheezy
(just some debian testing)
Desktop environment: mostly mate

when i am connected to my wirelless bluetoth headphones and i close the lid of my thinkpad. as usually, it dose try to go to suspend to ram but wakes up again, if i disable bluetooth before (hardware switch) it keeps sleeping. Please fix that (maybe also a debian problem) so i can suspend my thinkpad without switching off bluetooth befor

cschramm, 2018-12-01:
I'd be rather surprised if this actually has to do with blueman. In any case, the least we need is an information why the system does not suspend in the first place. I guess `dmesg` should show some reason.

infirit, 2018-12-01:
Typically software is not the cause of failing to suspend. More likely is that the kernel driver for the bluetooth adapter has a bug or is just not able to do this when something is connected.

closed 2019-01-04 by cschramm