blueman-project/blueman #947: feature request: show battery status of device

treaki, 2018-12-01:
for the version see my previews issue #945

When i am connected to my wireless headphones using my android phone i can see its battery status (little bar next to the Bluetooth icon in the notification bar). It would be very grate if blueman-manager could also show this next to the three bars at the right of the devie (package lost, receive and send signal (or so))

thanks in advance and keep up the good work

infirit, 2018-12-01:
This information is not exposed over the bluez dbus api. Recently for LE device https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/battery-api.txt was added, it is however not very reliable for the LE devices (always 100%) I have. I don't see this happen anytime soon.

cschramm, 2018-12-01:
We should be able to read the value from the GATT API https://git.kernel.org/pub/scm/bluetooth/bluez.git/tree/doc/gatt-api.txt, Battery Service (0x180F), Battery Level Characteristic (0x2A19). If I'm not mistaken, I have a mouse that provides it. Will check.

infirit, 2018-12-01:
Yes, this is what is exposed by the battery api and is LE only :smile: We have nothing exposed for older devices.

shanmukhateja, 2019-10-28:
I'd like to state that certain wireless headsets which support iOS devices have non-standard AT commands sent in HSP for battery indication.

There are ongoing talks over at PulseAudio for ways to support this feature [link](https://gitlab.freedesktop.org/pulseaudio/pulseaudio/issues/722).

Android vanilla has added support for this recently [link](http://androidxref.com/9.0.0_r3/xref/packages/apps/Bluetooth/src/com/android/bluetooth/btservice/RemoteDevices.java#733). 

Also, bluez-alsa project has it as well. [link](https://github.com/Arkq/bluez-alsa/blob/ed3f513a1f85d38daa4a80a868686bd709d1ac68/src/rfcomm.c#L443).

Fry-kun, 2020-04-23:
More devices support a reasonable battery level indication nowadays, please consider adding something here.
I'm sure I'm not the only one who's been using BT earbuds with a laptop lately.. they report their charge level to my phone perfectly fine, but there's nothing on my laptop :(

joker234, 2020-05-16:
This project works very well with my devices, but it would be nice to use the existing connection for these information and display this in the blueman-applet. https://github.com/TheWeirdDev/Bluetooth_Headset_Battery_Level

kred-jon, 2020-08-04:
still waiting default indicator from blueman, think its a must now.

areographe, 2020-10-18:
Hi, I just ran into this issue when my Apple Magic Keyboard battery reached 0% - on Windows and macOS, it's possible to check the battery level when I connect the keyboard over Bluetooth, whereas on Linux I need to remember to run `upower --dump`, which I never do. Is it possible to add this feature to the `blueman` applet?

yagodorea, 2020-10-27:
Any updates? That would be a really nice feature.

cschramm, 2020-10-29:
I'm currently "focused" on finishing #239 to finally get a version 2.2 out and would look into this afterward.

cschramm, 2020-11-02:
You're welcome to try and test #1417 and #1418.

rf-bcit, 2021-10-02:
+1 Vote for this feature

b-zee, 2022-01-12:
> You're welcome to try and test #1417 and #1418.

Thanks a lot for your work on this! I was curious to know the current state of affairs...

I run BlueZ 5.62 and Blueman 2.2.3, but I see no battery status. What are the steps to take to get this working? Is experimental mode for BlueZ necessary? The plugin menu shows a Battery plugin and is enabled...

cschramm, 2022-01-12:
@b-zee: Both features are in blueman 2.2 and the necessary API got added in BlueZ 5.48. Note that it only targets the standard Bluetooth LE Battery service and nothing legacy. The device needs to provide the `0000180f-0000-1000-8000-00805f9b34fb` (`0x180f`) Battery Service (check e.g. "Info" in the device's menu in blueman).

closed 2022-01-12 by cschramm

kred-jon, 2022-01-13:
solve it by gnome extension: https://github.com/MichalW/gnome-bluetooth-battery-indicator

Vudentz, 2022-06-27:
> @b-zee: Both features are in blueman 2.2 and the necessary API got added in BlueZ 5.48. Note that it only targets the standard Bluetooth LE Battery service and nothing legacy. The device needs to [...]

Just check for the presence of org.bluez.Battery1 since there could be other sources providing battery information (e.g. via AT commands over RFCOMM/HFP) not only BAS.

sonnysasaka, 2022-06-27:
+1 to @Vudentz comment above. Clients should use BlueZ's battery API (org.bluez.Battery1) to get battery information. This information does not only provide battery values from LE BAS but also from other sources, if there are battery providers. I assume that recent versions of [pulseaudio](https://gitlab.freedesktop.org/pulseaudio/pulseaudio/-/merge_requests/482) and [pipewire](https://gitlab.freedesktop.org/pipewire/pipewire/-/merge_requests/504) .

cschramm, 2022-06-27:
Thanks for the hint, @Vudentz. That's already done on the main branch (upcoming blueman 2.3). The dependency on the presence of a Battery Service only applies to blueman 2.2.