blueman-project/blueman #948: "You might not be able to connecto to the Bluetooth network via this machine"

WhyNotHugo, 2018-12-01:
blueman: 2.0.6
BlueZ: 5.50
Distribution: ArchLinux
Desktop environment: Xorg 1.20.3

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Whenever I start `blueman-applet`, I get a popup stating: 

> "You might not be able to connecto to the Bluetooth network via this machine"

There are two problems here (though I personally just care about the latter):

1. This tells me there's a problem with something. But I've no clue as to what the problem is, how it affects me, how to fix it, etc. UX-wise, this is really bad, because a typical user can only shrug, and click `OK`, without the slightest clue as to how it should be addressed nor what this means.
2. I don't really know how to get rid of this popup. `blueman` works for everything that I use it for, so, TBH, I just want to get rid of the popup warning everytime I run `blueman`.

Here's the console output for `blueman-applet`: https://gist.github.com/WhyNotHugo/00f92f516e439780994076e7ea5e86a1

infirit, 2018-12-01:
You are probably missing a polkit rules file, see https://github.com/blueman-project/blueman/issues/912 for another issue opened and the arch wiki entry with a rules file example https://wiki.archlinux.org/index.php/Blueman#Permissions

cschramm, 2018-12-02:
I think at least in `Network.ReloadNetwork` we're actually checking permissions too eagerly as `NetConf.apply_settings` might not actually try to create a bridge if the network is not configured. We should probably either move the check from `NetConf.apply_settings` to `Network.ReloadNetwork` or the authorization check into `NetConf.apply_settings`.

cschramm, 2018-12-02:
@WhyNotHugo: Can you check if #950 avoids the message?

WhyNotHugo, 2018-12-02:
I can confirm that #950 avoids this error message without the extra permissions file.

That was really fast, thanks!

PR #950 (2018-12-03): Avoid authorization check
We might ask for network setup authorization even if no network settings are configured and we do not actually set anything up. Moving the check in front of the confirmation should avoid that.

Fixes #948
commit facd2b05 (2018-12-03): Avoid authorization check

We might ask for network setup authorization even if no network settings are configured and we do not actually set anything up. Moving the check in front of the confirmation should avoid that.

Fixes #948
closed 2018-12-03 by cschramm