blueman-project/blueman #949: Using blueman-manager without blueman-applet

WhyNotHugo, 2018-12-01:
blueman: 2.0.6
BlueZ: 5.50
Distribution: ArchLinux
Desktop environment: 1.20.3

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Blueman works pretty well for me, but the usability is kinda confusing. If I start `blueman-manager` (which is the actual tool I need), it doesn't really work. For example, to connect home headsets, I need to:

* Run `blueman-applet` (why do I need to run this?).
  * Dismiss a warning that `blueman-applet` shows, see #948
* Run `blueman-manager`.
* Actually connect my headsets.
* After closing `blueman-manager`, `blueman-applet` is still there, but if I close it, everything continues working fine.

So I'm kinda wondering why manager depends on the applet, and if this can be avoided somehow. It kinda seems superfluous to manually have to start the applet for the manager works, when I don't really need to applet and it doesn't provide any use to me (note: I know it has uses, it's just that _I_ don't need it).

WhyNotHugo, 2018-12-02:
> `blueman-applet` is kind of a daemon that provides all the functionality we need on DBus. If you close it and continue using `blueman-manager`, `blueman-applet` will automatically get started when you commit any action.

I was using the latest stable (rather than master), and the "auto-starting" of `blueman-applet` is kinda a new thing in master, so I wasn't aware of that.  Thanks for pointing this out!

> Regarding the tray icon, we could provide a way to disable the StatusIcon plugin, however that'd be some work as we'd have to change the way other plugins that depend on it work. That's basically what you're asking for, right?

TBH, I'd rather not have to run the applet process entirely. This sounds kinda like running a deamon for a one-shot operation (connecting headphones, in my case). I'd rather open the blueman-manager, use it, close it, and not have anything linger.

However, if it's kinda necessary due to design limitation, I guess having the icon hide-able would be close enough for me. :slightly_smiling_face:

infirit, 2018-12-02:
> I was using the latest stable (rather than master), and the "auto-starting" of `blueman-applet` is kinda a new thing in master, so I wasn't aware of that. Thanks for pointing this out!

This is how its been in stable and its the core of blueman.

> TBH, I'd rather not have to run the applet process entirely. This sounds kinda like running a deamon for a one-shot operation (connecting headphones, in my case). I'd rather open the blueman-manager, use it, close it, and not have anything linger.

blueman-applet is used anytime you do anything in blueman-manager, there is no way around it.

blueman-applet provides functionality like NAP, PAN, Serial, tray icon, file transfer, authentication agent and more. 

> However, if it's kinda necessary due to design limitation, I guess having the icon hide-able would be close enough for me.

Its not a design limitation it is the design from the start. Your misunderstanding of the design has led you on the path of confusion :wink:

closed 2022-10-07 by cschramm