blueman-project/blueman #953: Lubuntu - Bluetooth Pairing_OK - Connection_NOK

AH40833V, 2018-12-03:
blueman:
[Bluetooth_issue - bluetoothd[1153].txt](https://github.com/blueman-project/blueman/files/2640237/Bluetooth_issue.-.bluetoothd.1153.txt)

BlueZ:
[image]

Distribution: 
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=16.04
DISTRIB_CODENAME=xenial
DISTRIB_DESCRIPTION="Ubuntu 16.04.5 LTS
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2018-12-03:
Most problems with a2dp are due to missing packages or some setup. See https://github.com/blueman-project/blueman/wiki/Troubleshooting#a2dp-audio-device-paired-but-fails-to-connect

If you still have problems see https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging and log your attempt to connect to the device.

rohoog, 2018-12-09:
I think I experience the same issue (fc28-xfce4). This used to work fine, but not anymore probably after updates of bluez/pulseaudio. The BT device (speaker) appears connected, but it fails to be registered with and used by pulseaudio. I don't think it is a blueman issue, but I lack the knowledge of who-does-what-and-who-is-to-blame.

rohoog, 2018-12-17:
Indeed, that selecting of the audio profile fails as well.

michael3638, 2019-04-24:
NOT stale. Same issue, beefree os-perixx mouse

cschramm, 2019-04-24:
Hi @michael3638, we need to see at least the logs as instructed on the troubleshooting page. Also your issue is most probably not the same as "os-perixx mouse" clearly doesn't sound like an audio device. :wink:

infirit, 2019-04-24:
Michael, this issue is about an audio device not mouse. If you have
problems with a mouse open a new issue and provide the necessary
information as requested in the new issue template.

closed 2019-06-30 by stale[bot]