blueman-project/blueman #961: Fix crash when an appearance is not in the gatt_appearance [patch]

rohoog, 2018-12-09:
blueman: blueman-2.1-0.9.alpha2.fc28
BlueZ: bluez-5.50-1.fc28
Distribution: fc28
Desktop environment: xfce4

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

diff --git a/blueman/DeviceClass.py b/blueman/DeviceClass.py
index bf3e1ca6..3fe74011 100644
--- a/blueman/DeviceClass.py
+++ b/blueman/DeviceClass.py
@@ -305,4 +305,6 @@ def get_minor_class(klass, i18n=False):
 
 
 def gatt_appearance_to_name(appearance):
-    return gatt_appearance[appearance]
+    if appearance in gatt_appearance:
+        return gatt_appearance[appearance]
+    return gatt_appearance[0];

infirit, 2018-12-09:
This just hides that we are missing id(s) or a bug. Please post,

* The value(s) of the Appearance property that causes the crash
* The backtrace/crash report
* blueman-applet log with --debug flag (see https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging)

rohoog, 2018-12-16:
The appearance value is 47806 from my 'LES3' BLE GPS watch (brand Lemfo)
(By the way, I now use the GIT version of blueman)
8<-------------------
[ronald@bto blueman]$ blueman-manager
blueman-manager version 2.1.alpha2 starting
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/blueman/main/Manager.py", line 141, in on_dbus_name_appeared
    self.List.display_known_devices(autoselect=True)
  File "/usr/local/lib/python3.6/site-packages/blueman/gui/DeviceList.py", line 285, in display_known_devices
    self.device_add_event(device)
  File "/usr/local/lib/python3.6/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 229, in device_add_event
    self.add_device(device)
  File "/usr/local/lib/python3.6/site-packages/blueman/gui/DeviceList.py", line 272, in add_device
    self.row_setup_event(tree_iter, device)
  File "/usr/local/lib/python3.6/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 274, in row_setup_event
    description = gatt_appearance_to_name(appearance)
  File "/usr/local/lib/python3.6/site-packages/blueman/DeviceClass.py", line 308, in gatt_appearance_to_name
    return gatt_appearance[appearance]
KeyError: 47806
----------------------->8
When I apply the patch to the DeviceClass python file, the watch appears as "Unknown" in the GUI window, which is expected and acceptable.
[blueman-applet.log](https://github.com/blueman-project/blueman/files/2683948/blueman-applet.log)

infirit, 2018-12-16:
@cschramm my LE mouse has appearance ``0x03c2`` and ``0x03c2 >> 6`` gives me 15 which corresponds to the GATT category. But doing the same for ``0xbabe`` (47806) I get 746 which is not even in the list. Am I missing something?

cschramm, 2018-12-17:
The first 10 bits specify the class and classes are only defined up to 81, so something's definitely wrong there. Also, `0xbabe` sounds like a placeholder... :thinking: I guess we should add a fallback for such "quality devices". :wink:

rohoog, 2018-12-17:
Yes, I think crashing with a keyError exception is less acceptable...

infirit, 2018-12-17:
Thanks @cschramm 

Opened #967 as potential fix for this.

infirit, 2018-12-18:
#967 merged so closing.

closed 2018-12-18 by infirit