blueman-project/blueman #963: Blueman does not start - ValueError: Namespace NM not available

lxs602, 2018-12-10:
blueman: 2.0.5
BlueZ: 5.50 (ppa)
Distribution: Ubuntu Budgie 18.04 x86_64
Desktop environment: Budgie

Hi,

I bought a 'new' laptop, and blueman will not start. 

I am sure it did when I first installed the system a few days ago, so perhaps the settings have been scrambled.

luke@luke-PORTEGE-R930:~$ blueman-applet
blueman-applet version 2.0.5 starting
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['DhcpClient', 'AppIndicator', 'DBusService', 'NetUsage', 'NMPANSupport', 'Headset', 'KillSwitch', 'PowerManager', 'PPPSupport', 'Networking', 'GameControllerWakelock', 'NMDUNSupport', 'RecentConns', 'Menu', 'ExitItem', 'AuthAgent', 'TransferService', 'DiscvManager', 'SerialManager', 'StandardItems', 'ShowConnected', 'StatusIcon'] 
/usr/lib/python3/dist-packages/blueman/plugins/applet/AppIndicator.py:8: PyGIWarning: AppIndicator3 was imported without specifying a version first. Use gi.require_version('AppIndicator3', '0.1') before import to ensure that the right version gets loaded.
  from gi.repository import AppIndicator3 as girAppIndicator
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
Traceback (most recent call last):
  File "/usr/bin/blueman-applet", line 121, in <module>
    BluemanApplet()
  File "/usr/bin/blueman-applet", line 63, in __init__
    self.Plugins.Load()
  File "/usr/lib/python3/dist-packages/blueman/main/PluginManager.py", line 90, in Load
    __import__(self.module_path.__name__ + ".%s" % plugin, None, None, [])
  File "/usr/lib/python3/dist-packages/blueman/plugins/applet/NMPANSupport.py", line 7, in <module>
    from blueman.main.NetworkManager import NMPANConnection
  File "/usr/lib/python3/dist-packages/blueman/main/NetworkManager.py", line 3, in <module>
    gi.require_version('NM', '1.0')
  File "/usr/lib/python3/dist-packages/gi/__init__.py", line 130, in require_version
    raise ValueError('Namespace %s not available' % namespace)
ValueError: Namespace NM not available

* [Yes ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2018-12-10:
I will add a fix so that it does not crash anymore but this is one of the annoying things of pygobject :disappointed: 

Under debian/ubuntu I think you need to install python-networkmanager

lxs602, 2018-12-20:
I installed python-networkmanager but it made no difference. 

I tried blueman on a ubuntu budgie 18.04 live CD and it ran, as I suspected, so I suppose I have broken the settings.

I will try the patch you made a link to.

infirit, 2018-12-21:
Sorry, wrong package name. It should be ``gir1.2-nm-1.0``.