blueman-project/blueman #966: adapter is greyed out in bluetooth manager

snikolov3, 2018-12-13:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

snikolov3, 2018-12-13:
lsmod output
Module                  Size  Used by
ccm                    20480  3
ip6t_REJECT            16384  1
nf_reject_ipv6         16384  1 ip6t_REJECT
nf_log_ipv6            16384  5
xt_hl                  16384  22
ip6t_rt                16384  3
nf_conntrack_ipv6      20480  8
nf_defrag_ipv6         36864  1 nf_conntrack_ipv6
ipt_REJECT             16384  1
nf_reject_ipv4         16384  1 ipt_REJECT
nf_log_ipv4            16384  5
nf_log_common          16384  2 nf_log_ipv4,nf_log_ipv6
xt_LOG                 16384  10
xt_limit               16384  13
xt_tcpudp              16384  18
xt_addrtype            16384  4
bnep                   20480  2
nf_conntrack_ipv4      16384  8
nf_defrag_ipv4         16384  1 nf_conntrack_ipv4
xt_conntrack           16384  16
arc4                   16384  2
edac_mce_amd           28672  0
snd_hda_codec_realtek   106496  1
kvm                   598016  0
snd_hda_codec_generic    73728  1 snd_hda_codec_realtek
iwlmvm                364544  0
irqbypass              16384  1 kvm
snd_hda_codec_hdmi     49152  1
crct10dif_pclmul       16384  0
crc32_pclmul           16384  0
ghash_clmulni_intel    16384  0
ip6table_filter        16384  1
pcbc                   16384  0
mac80211              778240  1 iwlmvm
snd_hda_intel          40960  4
snd_seq_midi           16384  0
ip6_tables             28672  1 ip6table_filter
snd_seq_midi_event     16384  1 snd_seq_midi
snd_hda_codec         126976  4 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec_realtek
btusb                  45056  0
nf_conntrack_netbios_ns    16384  0
btrtl                  16384  1 btusb
input_leds             16384  0
nf_conntrack_broadcast    16384  1 nf_conntrack_netbios_ns
btbcm                  16384  1 btusb
snd_hda_core           81920  5 snd_hda_codec_generic,snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_codec_realtek
snd_rawmidi            32768  1 snd_seq_midi
iwlwifi               282624  1 iwlmvm
btintel                16384  1 btusb
nf_nat_ftp             16384  0
snd_hwdep              20480  1 snd_hda_codec
nf_nat                 32768  1 nf_nat_ftp
bluetooth             548864  12 btrtl,btintel,btbcm,bnep,btusb
aesni_intel           188416  2
nf_conntrack_ftp       20480  1 nf_nat_ftp
snd_seq                65536  2 snd_seq_midi,snd_seq_midi_event
snd_pcm                98304  4 snd_hda_codec_hdmi,snd_hda_intel,snd_hda_codec,snd_hda_core
nf_conntrack          131072  8 xt_conntrack,nf_conntrack_ipv6,nf_conntrack_ipv4,nf_nat,nf_nat_ftp,nf_conntrack_netbios_ns,nf_conntrack_broadcast,nf_conntrack_ftp
aes_x86_64             20480  1 aesni_intel
crypto_simd            16384  1 aesni_intel
libcrc32c              16384  2 nf_conntrack,nf_nat
glue_helper            16384  1 aesni_intel
joydev                 24576  0
snd_seq_device         16384  3 snd_seq,snd_seq_midi,snd_rawmidi
wmi_bmof               16384  0
ecdh_generic           24576  1 bluetooth
cryptd                 24576  3 crypto_simd,ghash_clmulni_intel,aesni_intel
snd_timer              32768  2 snd_seq,snd_pcm
[... 110 lines cut ...]
sig_hashalgo:   md4
parm:           disable_scofix:Disable fixup of wrong SCO buffer size (bool)
parm:           force_scofix:Force fixup of wrong SCO buffers size (bool)
parm:           reset:Send HCI reset command on initialization (bool)

systool -v -m btusb output 

Module = "btusb"

  Attributes:
    coresize            = "45056"
    initsize            = "0"
    initstate           = "live"
    refcnt              = "0"
    srcversion          = "6E8AFA25DFF927A6A033399"
    taint               = ""
    uevent              = <store method only>
    version             = "0.8"

  Parameters:
    disable_scofix      = "N"
    force_scofix        = "N"
    reset               = "Y"

  Sections:

snikolov3, 2018-12-13:
lspci output

00:00.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Root Complex
00:00.2 IOMMU: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) I/O Memory Management Unit
00:01.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge
00:01.3 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge
00:02.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge
00:03.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge
00:03.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe GPP Bridge
00:04.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge
00:07.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge
00:07.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B
00:08.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) PCIe Dummy Host Bridge
00:08.1 PCI bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Internal PCIe GPP Bridge 0 to Bus B
00:14.0 SMBus: Advanced Micro Devices, Inc. [AMD] FCH SMBus Controller (rev 59)
00:14.3 ISA bridge: Advanced Micro Devices, Inc. [AMD] FCH LPC Bridge (rev 51)
00:18.0 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 0
00:18.1 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 1
00:18.2 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 2
00:18.3 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 3
00:18.4 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 4
00:18.5 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 5
00:18.6 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric Device 18h Function 6
00:18.7 Host bridge: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Data Fabric: Device 18h; Function 7
01:00.0 USB controller: Advanced Micro Devices, Inc. [AMD] Device 43d5 (rev 01)
01:00.1 SATA controller: Advanced Micro Devices, Inc. [AMD] Device 43c8 (rev 01)
01:00.2 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 43c6 (rev 01)
02:00.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 43c7 (rev 01)
02:01.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 43c7 (rev 01)
02:04.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 43c7 (rev 01)
02:08.0 PCI bridge: Advanced Micro Devices, Inc. [AMD] Device 43c7 (rev 01)
03:00.0 Ethernet controller: Intel Corporation I211 Gigabit Network Connection (rev 03)
04:00.0 Network controller: Intel Corporation Device 24fb (rev 10)
06:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller SM951/PM951 (rev 01)
07:00.0 VGA compatible controller: Advanced Micro Devices, Inc. [AMD/ATI] Ellesmere [Radeon RX 470/480] (rev ef)
07:00.1 Audio device: Advanced Micro Devices, Inc. [AMD/ATI] Ellesmere [Radeon RX 580]
08:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Device 145a
08:00.2 Encryption controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) Platform Security Processor
08:00.3 USB controller: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) USB 3.0 Host Controller
09:00.0 Non-Essential Instrumentation [1300]: Advanced Micro Devices, Inc. [AMD] Device 1455
09:00.2 SATA controller: Advanced Micro Devices, Inc. [AMD] FCH SATA Controller [AHCI mode] (rev 51)
09:00.3 Audio device: Advanced Micro Devices, Inc. [AMD] Family 17h (Models 00h-0fh) HD Audio Controller

lsmod | grep -i blue output

bluetooth             548864  12 btrtl,btintel,btbcm,bnep,btusb
ecdh_generic           24576  1 bluetooth

sudo lshw -class network output

  *-network                 
       description: Ethernet interface
       product: I211 Gigabit Network Connection
       vendor: Intel Corporation
       physical id: 0
       bus info: pci@0000:03:00.0
       logical name: enp3s0
       version: 03
       serial: e0:d5:5e:ae:78:00
       size: 1Gbit/s
       capacity: 1Gbit/s
       width: 32 bits
       clock: 33MHz
       capabilities: pm msi msix pciexpress bus_master cap_list ethernet physical tp 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
       configuration: autonegotiation=on broadcast=yes driver=igb driverversion=5.4.0-k duplex=full firmware=0. 6-1 ip=192.168.1.66 latency=0 link=yes multicast=yes port=twisted pair speed=1Gbit/s
       resources: irq:29 memory:fe300000-fe31ffff ioport:e000(size=32) memory:fe320000-fe323fff
  *-network
       description: Wireless interface
       product: Intel Corporation
       vendor: Intel Corporation
       physical id: 0
       bus info: pci@0000:04:00.0
       logical name: wlp4s0
       version: 10
       serial: 34:e1:2d:76:26:81
       width: 64 bits
       clock: 33MHz
       capabilities: pm msi pciexpress bus_master cap_list ethernet physical wireless
       configuration: broadcast=yes driver=iwlwifi driverversion=4.15.0-42-generic firmware=29.1044073957.0 ip=192.168.1.64 latency=0 link=yes multicast=yes wireless=IEEE 802.11
       resources: irq:77 memory:fe200000-fe201fff

snikolov3, 2018-12-13:
Bluetooth driver is integrated into motherboard, aorus b450 wifi board

snikolov3, 2018-12-13:
Bluetooth worked on ubuntu. Then had to reinstall ubuntu and bluetooth stopped working. Then now I am on linux mint and bluetooth is still not working

snikolov3, 2018-12-13:
Starting the blueman manager the adapter field is greyed out

cschramm, 2018-12-15:
https://github.com/blueman-project/blueman/wiki/Troubleshooting#no-adapter-or-not-able-to-turn-on-bluetooth

closed 2019-03-30 by cschramm