blueman-project/blueman #973: Mint 19.1 no icon

FlailAway, 2018-12-26:
blueman: 2.0.5
BlueZ:
Distribution: No idea what you mean
Desktop environment: Mint 19.1 all up to date

Used the CLI to install "sudo apt install blueman"

In the View menu it has "Show toolbar" and "Show status bar" both checked but when I close the app, I no longer have an icon anywhere. Not sure which is the toolbar and which is the status bar, but I'd like to see the icon on bar at the lower right of Mint (with Network, Date+Time etc) so I can click it to launch Blueman to send a file.

How do I do that?

Yes, I have done the following and more.
* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

FlailAway, 2018-12-26:
Ok, figured it out. It was the Mint Theme with a light bar across the bottom. The Blueman icon is white so I could not see it there. I changed the theme to one with a dark bar and I can see the icon. BUT, I need the light bar to see the Workspaces which I use a hundred times a day.

Is there some way to change the color of the Blueman icon as I need to keep the light colored bar?

cschramm, 2018-12-27:
Yep, that basically means your desktop theme is not compatible with your icon theme. Not sure whom I'd file a bug with but maybe both are from the same guys anyway?

Regarding terminology:

> Distribution: No idea what you mean

That would be Linux Mint 19.1

> Desktop environment: Mint 19.1 all up to date

and that would be either Cinnamon (probably), MATE, or Xfce.

FlailAway, 2018-12-27:
@infirit: Thanks, then it is a rock and a hard place thing as I can either see the Blueman icon or the four workspaces and that's a no brain-er for me. I understand the "not influence" approach and that's to be commended. So I will live with the "invisible," but I know here it is, icon. :)

@cschramm: Thanks for the comments, I promise to get that terminology correct the next time. As mentioned, I'll just live with the "invisible" icon. It's the white gap between Network and workspaces. ;)

infirit, 2018-12-28:
Typically the icon theme will make a symlink from the bluetooth icon to blueman. You could look for that and remove the symlink to get the blue icon we provide. Or if it is a proper icon remove it.

Look in ~/.local/share/icons/ or /usr/share/icons/ and look for these,
blueman-active.png
blueman-disabled.png
blueman-tray.png

Closing but feel free to continue the discussion.

closed 2018-12-28 by infirit

cschramm, 2019-01-05:
@FlailAway: I'd try to find out if the icon theme you're using is supposed to play well with your window theme. If that's the case, it's a bug in the icon theme and should get filed with them.