blueman-project/blueman #975: unsupported operand type(s)

m0ppers, 2018-12-31:
blueman: 2.0.7
BlueZ: 5.50
Distribution: arch linux
Desktop environment: sway

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2018-12-31:
Thanks for the report. Can you try below please.

Also I am interested in what the Class id is so can you post the log of blueman-manager, run it from a terminal and post it into a github gist :smile: .

```patch
diff --git a/blueman/gui/manager/ManagerDeviceList.py b/blueman/gui/manager/ManagerDeviceList.py
index b5ac8686..51af1c0c 100644
--- a/blueman/gui/manager/ManagerDeviceList.py
+++ b/blueman/gui/manager/ManagerDeviceList.py
@@ -250,7 +250,7 @@ class ManagerDeviceList(DeviceList):
             description = gatt_appearance_to_name(appearance)
         else:
             icon = get_icon(device.Icon, 48, "blueman")
-            description = get_major_class(device.Class).capitalize()
+            description = klass.capitalize()
 
         name = device.Alias
         address = device.Address
```

m0ppers, 2018-12-31:
that works as well :)

here is the gist:

(Added "Klass: " and "Device.class:" there)

https://gist.github.com/m0ppers/d7d7d0ac7a149adb4d433e16809ccc83

infirit, 2019-01-01:
Thanks. This was backported from master and I introduced this bug :sob:.

abhixec, 2019-01-06:
Thanks, that fixed the issue for me too on Archlinux.

Mange, 2019-02-07:
Also worked for me. Is there any chance to get this shipped soon so I don't have to manually apply this patch to all my machines?

blueman-manager 2.0.7

cschramm, 2019-02-07:
What do mean, @Mange? We released version 2.0.8 a month ago, so if you're talking about the Arch package that's on Arch. https://www.archlinux.org/packages/community/x86_64/blueman/ it's flagged out-of-date but there seems to be no issue or any other information on an update to 2.0.8.

Mange, 2019-02-08:
I apologize. :bowing_man: 

Since I couldn't get a newer version on Arch, and this issue was still open, I wrongly assumed that no fix had been released. I should have checked more closely, and will do so the next time.

closed 2019-02-08 by cschramm