blueman-project/blueman #976: Icon theme compatibility

cschramm, 2018-12-31:
Follow-up for #974.

Current state is that we advise gnome and mate in https://github.com/blueman-project/blueman/blob/master/Dependencies.md and supposedly check for gnome, mate, faenza, or adwaita in https://github.com/blueman-project/blueman/blob/master/configure.ac. That check seems flawed in multiple ways: a) Isn't it a runtime-only dependency, so it should be affected by `$enable_runtime_deps_check`? b) faenza and mate do not seem to actually provide pkgconfig configurations at all, so the check will fail in any case if neither adwaita nor gnome are present. c) The check does not match the alleged dependencies. d) None of the four icon themes provides all names that we use.

Icon names that I identified in our code:

```
application-exit
dialog-error
dialog-information
dialog-password
document-new
document-open
document-open-recent
edit-find
go-down
go-up
gnome-power-manager
help-about
menu-editor
modem
network
network-wireless
network-workgroup
preferences-desktop
process-stop
system-shutdown
```

Of those, `gnome-power-manager` (`PowerManager` plugin), `menu-editor` (`Menu` plugin), and `network` are not part of the Icon Naming Specification. `modem` is but faenza icon themes do not provide them.

That leaves us with the following ("more relevant" :wink:) icon themes:

||gnome-power-manager|menu-editor|network|
|-|-|-|-|
|[elementary-xfce](https://github.com/shimmerproject/elementary-xfce)|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|
|[Papirus](https://github.com/PapirusDevelopmentTeam/papirus-icon-theme)|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|
|[GTK+ 2 HighContrast](https://github.com/GNOME/gnome-themes-extra)|:heavy_check_mark:|||
|[gnome-icon-theme](https://github.com/GNOME/gnome-icon-theme)|||:heavy_check_mark:|
|[mate-icon-theme](https://github.com/mate-desktop/mate-icon-theme)|||:heavy_check_mark:|
|[adwaita-icon-theme](https://github.com/GNOME/adwaita-icon-theme)||||
|[EFL](https://git.enlightenment.org/core/efl.git/)||||
|[Oxygen](https://github.com/KDE/oxygen-icons)|||

Proposed action items:

* Rethink the checks in configure.ac. Should we just drop them?
* Drop, replace, or just keep (treating it as optional) `menu-editor` in the `Menu` applet plugin. The plugin does not necessarily need an icon; others do not have one as well.
* Find a solution for for `gnome-power-manager`. If we cannot drop it or find a replacement, the best thing would probably be to ship it. I noticed that there is a `gnome-power-manager-symbolic` icon in adwaita-icon-theme, [Faba](https://github.com/snwh/faba-icon-theme) (which extends gnome-icon-theme), mate-icon-theme, and Papirus. Maybe that could be a replacement. On the other hand it'd exclude elementary-xfce again.
* Think about `network`. It would definitely be nice to achieve compatibility with adwaita-icon-theme and I guess the icon is easy to replace with something from the specification (maybe just `network-workgroup`?). My `network` icon does not really fit anyway as it's more something I'd associate with a network share rather than networking in general.
* Update dependencies, depending on the actions taken.

infirit, 2018-12-31:
> a) Isn't it a runtime-only dependency, so it should be affected by `$enable_runtime_deps_check`? 

They are runtime only.

> b) faenza and mate do not seem to actually provide pkgconfig configurations at all, so the check will fail in any case if neither adwaita nor gnome are present.

Hmm, I thought they did but just checked to be sure and your right :disappointed: 

<snip>

> Of those, `gnome-power-manager` (`PowerManager` plugin), `menu-editor` (`Menu` plugin), and `network` are not part of the Icon Naming Specification. `modem` is but faenza icon themes do not provide it.

faenza inherits from mate and hicolor so any missing icons will fallback to those. And yea, power manager and menu plugin are somewhat problematic.

<snip>

> * Rethink the checks in configure.ac. Should we just drop them?

Add a message/warning that we need an icon theme conforming to the icon naming spec and list a few examples works for me :smile:.

> * Drop, replace, or just keep (treating it as optional) `menu-editor` in the `Menu` applet plugin. The plugin does not necessarily need an icon; others do not have one as well.

We fallback to the default plugin icon anyway so we could leave it as it is. The ``network`` icon though I think we should change

> * Find a solution for for `gnome-power-manager`. If we cannot drop it or find a replacement, the best thing would probably be to ship it. I noticed that there is a `gnome-power-manager-symbolic` [...]

See above on fallback. But the symbolic ones are special and if I am not mistaken we are not supposed to ask for them. If I find a reference I'll post it.

> * Think about `network`. It would definitely be nice to achieve compatibility with adwaita-icon-theme and I guess the icon is easy to replace with something from the specification (maybe just [...]

There are several network related icons in the spec and I'm sure there will be one that fits.

> * Update dependencies, depending on the actions taken.

Sounds like a plan but while we are discussing icons I want to put device icons as action/decision item. 

We do not have device icons for all types of devices. We fallback to using what is exposed by bluez over dbus and if all fails we use `blueman`. There is a big chance there is a mix of our device icons and the ones provided by the icon theme. Also as they are lowres so with ``GDK_SCALE=2`` they are blurry. I would prefer to drop ours and just rely on what the icon theme provides..

cschramm, 2019-01-01:
Regarding device icons: BlueZ gives us the following names:

```
audio-card
camera-photo
camera-video
computer
input-gaming
input-keyboard
input-mouse
input-tablet
modem
network-wireless
phone
printer
```

All of them are in the Icon Naming Specification, so I think we can just drop our icons an rely on the icon theme, yes.

Regarding Faenza: Neither https://github.com/shlinux/faenza-icon-theme nor https://github.com/mate-desktop/mate-icon-theme-faenza mention anything about another icon theme that's required as fallback. Consequently at least in Debian the `faenza-icon-theme` package has zero dependencies and `mate-icon-theme-faenza` depends on `hicolor-icon-theme` only for the directory structure (contrary to the name there are no actual icons in that package). So in Debian and Co. you will not be guaranteed to have a modem, phone, or video-camera icon if you depend on a Faenza icon theme (except that you cannot have the GTK+ libs around without gnome-icon-theme or adwaita-icon-theme).

I found some more icon names that we use. Together with the names from BlueZ my list is now:

```
application-exit
audio-card
camera-photo
camera-video
computer
dialog-error
dialog-information
dialog-password
dialog-warning
document-new
document-open
document-open-recent
document-properties
edit-clear
edit-copy
edit-delete
edit-find
gnome-power-manager
go-down
go-up
gtk-preferences
help-about
image-missing
info
input-gaming
input-keyboard
input-mouse
input-tablet
list-remove
menu-editor
modem
network
network-offline
network-transmit-receive
network-wireless
network-workgroup
phone
preferences-desktop
preferences-system
printer
process-stop
system-shutdown
view-refresh
```

Of the additional names, `info` and `gtk-preferences` are not part of the Icon Naming Specification and and EFL and Oxygen lack some others. That leaves us with:

||gnome-power-manager|gtk-preferences|info|menu-editor|network|
|-|-|-|-|-|-|
|elementary-xfce|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|:heavy_check_mark:|
|Papirus|:heavy_check_mark:|:heavy_check_mark:||:heavy_check_mark:|:heavy_check_mark:|
|gnome-icon-theme||:heavy_check_mark:|:heavy_check_mark:||:heavy_check_mark:|
|mate-icon-theme||:heavy_check_mark:|:heavy_check_mark:||:heavy_check_mark:|
|GTK+ 2 HighContrast|:heavy_check_mark:|:heavy_check_mark:||||
|adwaita-icon-theme||||||

infirit, 2019-01-01:
On device icons, we can drop some code if we remove the icons. You want me to handle this or will you?

On faenza icons, the index.theme has the inherits line that determines this. For example, https://github.com/shlinux/faenza-icon-theme/blob/master/Faenza/index.theme#L3. See the spec [here](https://standards.freedesktop.org/icon-theme-spec/icon-theme-spec-latest.html#file_formats). If none if the inherits are a dep in debian then you may end up with broken icons, 😕.

We can replace ``gtk-preferences`` with ``preferences-system``. I am not on my machine so can't check, where is ``info`` used? Can we replace it with ``dialog-information``?

I also found the wikimedia page on icon sets quite helpful, https://commons.wikimedia.org/wiki/Comparison_of_icon_sets

cschramm, 2019-01-01:
> On device icons, we can drop some code if we remove the icons. You want me to handle this or will you?

Go ahead. I don't really see where exactly you're heading, to be honest. :wink: This is more complicated than I first thought as we actually distinguish between much more icons than what BlueZ provides.

> On faenza icons, the index.theme has the inherits line that determines this.

Aha. Hm, seem like a packaging bug in Debian then, yes. Anyway, if theme A depends on theme B and them B is already "complete" for us, I don't think mentioning theme A as alternative makes sense as that would mean that we should mention *everything* that extends B (read: gnome-icon-theme).

#978 and #979 replace three of the critical icons, the other two are used as icons for specific applet plugins and we just fall back to the default if they are not present.

I re-opened and merged #974.

Open items:

 * [ ] Drop or replace checks in configure.ac.
 * [ ] Add adwaita-icon-theme and Papirus as alternatives to dependencies.
 * [ ] Drop own device icons.

infirit, 2019-01-01:
>Go ahead. I don't really see where exactly you're heading, to be honest. wink This is more complicated than I first thought as we actually distinguish between much more icons than what BlueZ provides.

I'm quite familiar because of the hidpi changed. Yes it has three layers of checking, now reduced to one. #981 :tada:

PR #1016 (2019-04-28): Replace invalid checks for icon themes with message
Closes #976
commit 9c02e638 (2019-04-28): Replace invalid checks for icon themes with message (#1016)

Closes #976
closed 2019-04-28 by cschramm