blueman-project/blueman #983: Bluetooth connection drops after a few minutes working on Ubuntu 18.04

ndiogene, 2019-01-04:
I use SONY MDR-XB650BT Bluetooth headset and it works fine for a few minutes. However,  all of a sudden the audio stops playing and the Bluetooth connection is lost. Using Blueman, I keep trying to reconnect with no success until I reboot the computer. After reboot, I successfully reconnect but the connection drops again after around 5 minutes. Any help ?

Ubuntu version:

Distributor ID:	Ubuntu
Description:	Ubuntu 18.04.1 LTS
Release:	18.04
Codename:	bionic

Bluetooth version: 5.48

Carl393, 2019-01-05:
Do you have a usb wifi connector? I have a bluetooth dongle as well as a separate usb wifi connector and I had dropping issues as well..

try this:
https://superuser.com/questions/924559/wifi-connection-troubles-solved-why-does-my-fix-work

Carl393, 2019-01-05:
that fixed my bluetooth dropping issues.

infirit, 2019-01-22:
This sound to me like something is going wrong at the kernel driver level. Does your kernel log show any kernel panics related to bluetooth? Also it really helps if you can post the log of blueman-applet as described in https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging

And yes, having two adapters near to each other will result in weird problems, especially when it is discovering new devices.

closed 2019-03-30 by cschramm

paland3, 2020-04-14:
> I use SONY MDR-XB650BT Bluetooth headset and it works fine for a few minutes. However, all of a sudden the audio stops playing and the Bluetooth connection is lost. Using Blueman, I keep trying to [...]

did you find a solution to this?