blueman-project/blueman #987: blueman-manager won't start after update

jeanschneider, 2019-01-22:
blueman: 2.0.7-1 (stable/manjaro repo)
BlueZ: 5.50-3
Distribution: Manjaro
Desktop environment: xfce4

Error code when launching blueman-manager : 
```
[jeans@HP_x360 ~]$ blueman-manager

(blueman-manager:2381): Gtk-WARNING **: 10:14:35.776: Theme parsing error: gtk.css:63:28: The :prelight pseudo-class is deprecated. Use :hover instead.

(blueman-manager:2381): Gtk-WARNING **: 10:14:35.776: Theme parsing error: gtk.css:73:35: The :prelight pseudo-class is deprecated. Use :hover instead.
_________
Load (/usr/lib/python3.7/site-packages/blueman/main/PluginManager.py:60)
['Services', 'PulseAudioProfile'] 
_________
__load_plugin (/usr/lib/python3.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.Services.Services'> 
_________
__load_plugin (/usr/lib/python3.7/site-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.manager.PulseAudioProfile.PulseAudioProfile'> 
_________
pa_context_event (/usr/lib/python3.7/site-packages/blueman/main/PulseAudioUtils.py:341)
1 
/usr/bin/blueman-manager:67: DeprecationWarning: Gtk.Window.get_has_resize_grip is deprecated
  if self.window.get_has_resize_grip():
blueman-manager version 2.0.7 starting
Stale PID, overwriting
_________
pa_context_event (/usr/lib/python3.7/site-packages/blueman/main/PulseAudioUtils.py:341)
2 
_________
on_bluez_name_owner_changed (/usr/bin/blueman-manager:96)
org.bluez owner changed to  :1.2 
_________
get_interface_version (/usr/lib/python3.7/site-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
_________
SetAdapter (/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py:271)
 
_________
on_adapter_changed (/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerToolbar.py:83)
toolbar adapter /org/bluez/hci0 
_________
__init__ (/usr/lib/python3.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
add_device (/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py:317)
adding new device 
_________
do_cache (/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py:496)
Caching new device 48:D6:D5:98:2C:A5 
_________
row_update_event (/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:275)
row update event Fake False 
_________
row_update_event (/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:275)
row update event Trusted 0 
_________
row_update_event (/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerDeviceList.py:275)
row update event Paired 1 
_________
__init__ (/usr/lib/python3.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
add_device (/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py:317)
adding new device 
_________
do_cache (/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py:496)
Caching new device 88:C6:26:EC:DD:D5 
[... 181 lines cut ...]
  if self.window.get_has_resize_grip():
blueman-manager version 2.0.7 starting
Stale PID, overwriting
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/dbus/connection.py", line 604, in msg_reply_handler
    reply_handler(*message.get_args_list(**get_args_opts))
  File "/usr/bin/blueman-manager", line 150, in on_bluez_name_owner_changed
    self.List.DisplayKnownDevices(autoselect=True)
  File "/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py", line 392, in DisplayKnownDevices
    self.device_add_event(Device(device))
  File "/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 207, in device_add_event
    self.AppendDevice(device)
  File "/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py", line 425, in AppendDevice
    self.add_device(device, True)
  File "/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py", line 332, in add_device
    self.row_setup_event(iter, device)
  File "/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 253, in row_setup_event
    description = get_major_class(device.Class).capitalize()
  File "/usr/lib/python3.7/site-packages/blueman/DeviceClass.py", line 239, in get_major_class
    index = (klass >> 8) & 0x1F
TypeError: unsupported operand type(s) for >>: 'str' and 'int'

```

Thanks !

infirit, 2019-01-22:
Yes, this was a bug I introduced in 2.0.7. A new release. 2.0.8, was done for this so use that or apply ad7a67d3d528c232d41ae31554f9fc70c28839e2 yourself.

Closing but feel free to continue the discussion :smile:

closed 2019-01-22 by infirit

jeanschneider, 2019-01-22:
Thanks ! I hope the new version will soon be on the repos !