blueman-project/blueman #988: Can not connect to bluetooth serial converter.

mateakos1, 2019-01-26:
blueman: 2.1_alpha3
BlueZ: 5.50-r1
Distribution: Gentoo
Desktop environment: XFCE

I am using a HC-05 bluetooth serial converter, and since the latest blueman update I can not connect to it.
When right clicking on the device and selecting Serial Port, nothing happens. If I run Wireshark to see the bluetooth traffic, there is none.
Searching and pairing works fine. Connecting to A2DP devices also works.

With blueman-2.1_alpha2 the connection to the HC-05 module is working, and /dev/rfcomm0 is created. The only problem with that version is that rfcomm0 is not removed if the connection breaks abnormally (for example, when removing the power from the serial converter before disconnecting first)

infirit, 2019-01-26:
Quick reply before I have to leave. I'll ask for some logs later.

With abnormal connection break we can not forcefully release the rfcomm device. If some program still has it open it first has to be killed and the rfcomm device will release itself.

mateakos1, 2019-01-27:
blueman-applet
https://pastebin.com/ZSUtwPa2

mateakos1, 2019-01-27:
And this is how the output from blueman-mechanism looks like after I do the same action in the blueman-manager.
`/usr/libexec/blueman-mechanism -d -s --loglevel debug
Jelszó: 
blueman-mechanism 14.36.54 INFO     blueman-mechanism:73 <module>  : Enabled verbose output
blueman-mechanism 14.36.54 INFO     blueman-mechanism:82 <module>  : Starting blueman-mechanism
blueman-mechanism 14.36.54 INFO     blueman-mechanism:154 __init__  : loading Ppp
blueman-mechanism 14.36.54 INFO     blueman-mechanism:154 __init__  : loading Rfcomm
blueman-mechanism 14.36.54 INFO     blueman-mechanism:154 __init__  : loading Network
blueman-mechanism 14.36.54 INFO     blueman-mechanism:154 __init__  : loading RfKill
blueman-mechanism 14.38.02 DEBUG    blueman-mechanism:170 confirm_authorization: (True, False, {})`

mateakos1, 2019-01-27:
And this is the output of blueman-mechanism, when I repeat the same thing:
https://pastebin.com/BP7VHwWL

xbr2019, 2019-01-27:
I turned on, found the device, the connection was successful, there is no sound, disconnected, closed, opened, it does not turn on, I cannot remove the device, I cannot connect with the new device. And silence.Thank you, it's nice, the most important thing is that we are alive.

cschramm, 2019-04-07:
@mateakos1: Unfortunately the pastebin links do not work.

mateakos1, 2019-04-08:
Might be I will return to this problem, but I have no time for it these days. Until then, blueman-2.1_alpha2 works good enough.

closed 2019-06-14 by stale[bot]