blueman-project/blueman #989: Ambiguous wording

FlailAway, 2019-01-28:
blueman-applet: 2.0.5
BlueZ:
Distribution:
Desktop environment:

* [YES ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Your "Device" menu is ambiguous. It has a check-mark alongside "Trust" which suggests that the device **is** already daulted to trusted, but I am guessing it needs the "Untrust" visible for to be **actually** trusted. And, then you have a radio-button for "Untrust"

I could not find any useful help online related to which way this setting should be for the device to be trusted.

infirit, 2019-02-02:
The correct behaviour is, show ``Trust`` when the device is untrusted and show ``Untrust`` when it is trusted.

The Device menu and the the right clicking menu on the device are the same. The right click menu does work properly so something is wrong in the menubar.

infirit, 2019-02-03:
See #989, I will backport to stable later.

commit e9c8691a (2019-04-19): ManagerDeviceMenu: Always generate menu items on property changes

We also need to update the menu that is in the Gtk.MenuBar. Fixes #989.
closed 2019-04-19 by infirit