blueman-project/blueman #994: Connected, but cursor is not moved on Rapoo M100

vlad1777d, 2019-02-16:
blueman: 2.0.5
BlueZ: 5.48.0
Distribution: Linux Mint 19.1
Desktop environment: Cinnamon

* [+] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2019-03-30:
Definitely not a blueman issue, but let's see where we can get. When I do connect a bluetooth mouse `dmsg` looks like this:

```
hidraw: raw HID events driver (C) Jiri Kosina
Bluetooth: HIDP (Human Interface Emulation) ver 1.2
Bluetooth: HIDP socket layer initialized
hid-generic 0005:046D:B008.0001: unknown main item tag 0x0
input: Bluetooth Laser Travel Mouse Mouse as /devices/pci0000:00/0000:00:14.0/usb2/2-11/2-11:1.0/bluetooth/hci0/hci0:256/0005:046D:B008.0001/input/input22
hid-generic 0005:046D:B008.0001: input,hidraw0: BLUETOOTH HID v3.14 Mouse [Bluetooth Laser Travel Mouse] on e8:b1:fc:41:ad:0b
```

I then end up with a new mouse device (`/dev/input/mouse2` in my case) for which I see output if I `cat` it and move the mouse.

I'd check which of these parts differ in your case.

infirit, 2019-06-02:
The only thing I can think of is that you do not have the UHID kernel driver enabled/loaded.

closed 2019-08-08 by stale[bot]