blueman-project/blueman #996: send-to: unable to find service record error

telychko, 2019-03-05:
blueman:  2.0.8
BlueZ: 5.50
Distribution: Arch-Linux
Desktop environment: XFCE4

For debug reason launch blueman-applet via terminal. Then trying to send file to smartphone. Got this error "unable to find service record".

_________
__init__ (/usr/lib/python3.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib/python3.7/site-packages/blueman/main/Device.py:73)
deleting device /org/bluez/hci0/dev_04_B1_67_71_92_AF 
_________
Destroy (/usr/lib/python3.7/site-packages/blueman/main/Device.py:95)
invalidating device /org/bluez/hci0/dev_04_B1_67_71_92_AF 
/usr/bin/blueman-sendto:374: PyGTKDeprecationWarning: Using positional arguments with the GObject constructor has been deprecated. Please specify keyword(s) for "title" or use a class specific constructor. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  "_OK", Gtk.ResponseType.ACCEPT))
/usr/bin/blueman-sendto:374: PyGTKDeprecationWarning: The "buttons" argument must be a Gtk.ButtonsType enum value. Please use the "add_buttons" method for adding buttons. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  "_OK", Gtk.ResponseType.ACCEPT))
/usr/bin/blueman-sendto:388: PyGTKDeprecationWarning: The "buttons" argument must be a Gtk.ButtonsType enum value. Please use the "add_buttons" method for adding buttons. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  d = DeviceSelectorDialog()
_________
on_adapter_property_changed (/usr/lib/python3.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 1 
Sending to 04:B1:67:71:92:AF
_________
on_adapter_property_changed (/usr/lib/python3.7/site-packages/blueman/plugins/applet/DiscvManager.py:96)
prop Discovering 0 
_________
__init__ (/usr/lib/python3.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
/usr/bin/blueman-sendto:280: PyGTKDeprecationWarning: Using positional arguments with the GObject constructor has been deprecated. Please specify keyword(s) for "parent" or use a class specific constructor. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  type=Gtk.MessageType.ERROR, buttons=(Gtk.ButtonsType.CLOSE))
/usr/bin/blueman-sendto:280: PyGTKDeprecationWarning: The keyword(s) "type" have been deprecated in favor of "message_type" respectively. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  type=Gtk.MessageType.ERROR, buttons=(Gtk.ButtonsType.CLOSE))
_________
__del__ (/usr/lib/python3.7/site-packages/blueman/main/Device.py:73)
deleting device /org/bluez/hci0/dev_04_B1_67_71_92_AF 
_________
Destroy (/usr/lib/python3.7/site-packages/blueman/main/Device.py:95)
invalidating device /org/bluez/hci0/dev_04_B1_67_71_92_AF 
_________
__init__ (/usr/lib/python3.7/site-packages/blueman/main/Device.py:26)
caching initial properties 

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2019-03-24:
Sounds like the receiver is not ready to accept files although it advertises Object Push. You can check the available UUIDs of the device with `info 04:B1:67:71:92:AF` in `bluetoothctl`.

telychko, 2019-04-23:
No, it has such capability. See

<details><summary>bluetoothctl check</summary>

```
$ bluetoothctl 
Agent registered
[bluetooth]# show
Controller 7C:E9:D3:E2:B2:02 (public)
	Name: lenovoX121e
	Alias: lenovoX121e
	Class: 0x001c010c
	Powered: yes
	Discoverable: no
	Pairable: yes
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: IrMC Sync                 (00001104-0000-1000-8000-00805f9b34fb)
	UUID: Audio Sink                (0000110b-0000-1000-8000-00805f9b34fb)
	UUID: Vendor specific           (00005005-0000-1000-8000-0002ee000001)
	Modalias: usb:v1D6Bp0246d0532
	Discovering: no
Usage: devices 
[bluetooth]# devices 
Device 04:B1:67:71:92:AF Maksym@XiaomiRedmiNote4x
Device 20:13:12:04:00:CC Bluetooth Mouse
[bluetooth]# paired-devices 
Device 04:B1:67:71:92:AF Maksym@XiaomiRedmiNote4x
Device 20:13:12:04:00:CC Bluetooth Mouse
[bluetooth]# info 04:B1:67:71:92:AF 
Device 04:B1:67:71:92:AF (public)
	Name: Maksym@XiaomiRedmiNote4x
	Alias: Maksym@XiaomiRedmiNote4x
	Class: 0x005a020c
	Icon: phone
	Paired: yes
	Trusted: yes
	Blocked: no
	Connected: no
	LegacyPairing: no
	UUID: OBEX Object Push          (00001105-0000-1000-8000-00805f9b34fb)
	UUID: Headset                   (00001108-0000-1000-8000-00805f9b34fb)
	UUID: Audio Source              (0000110a-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control Target (0000110c-0000-1000-8000-00805f9b34fb)
	UUID: A/V Remote Control        (0000110e-0000-1000-8000-00805f9b34fb)
	UUID: Headset AG                (00001112-0000-1000-8000-00805f9b34fb)
	UUID: PANU                      (00001115-0000-1000-8000-00805f9b34fb)
	UUID: NAP                       (00001116-0000-1000-8000-00805f9b34fb)
	UUID: Handsfree Audio Gateway   (0000111f-0000-1000-8000-00805f9b34fb)
	UUID: SIM Access                (0000112d-0000-1000-8000-00805f9b34fb)
	UUID: Phonebook Access Server   (0000112f-0000-1000-8000-00805f9b34fb)
	UUID: Message Access Server     (00001132-0000-1000-8000-00805f9b34fb)
	UUID: PnP Information           (00001200-0000-1000-8000-00805f9b34fb)
	UUID: Generic Access Profile    (00001800-0000-1000-8000-00805f9b34fb)
	UUID: Generic Attribute Profile (00001801-0000-1000-8000-00805f9b34fb)
	Modalias: bluetooth:v000Fp1200d1436
[bluetooth]# connect 04:B1:67:71:92:AF 
Attempting to connect to 04:B1:67:71:92:AF
[CHG] Device 04:B1:67:71:92:AF Connected: yes
Connection successful
[CHG] Device 04:B1:67:71:92:AF ServicesResolved: yes
[Maksym@XiaomiRedmiNote4x]# disconnect 
Attempting to disconnect from 04:B1:67:71:92:AF
[CHG] Device 04:B1:67:71:92:AF ServicesResolved: no
Successful disconnected
[CHG] Device 04:B1:67:71:92:AF Connected: no
```

</details>

infirit, 2019-04-23:
@telychko Can you post the log of ``blueman-sendo`` as it may shed some light on what is going on. Run it from a terminal and post its output here.

cschramm, 2019-04-23:
To rule out possible issue, could you test with a different sender (application / operating system / device) and / or with a different receiver?

Also, I'd check `systemctl --user status obex.service` after trying to send a file. Maybe it has some issues. Looking at the `bluetoothd` log might help as well.

telychko, 2019-05-04:
Thank for guide. Bug can be closed with. Seems problem in underlayer bluetooth:
$systemctl status bluetooth
May 04 21:07:14 lenovoX121e bluetoothd[433]: avrcp-controller: Protocol not supported (93)
May 04 21:07:14 lenovoX121e bluetoothd[433]: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
May 04 21:07:14 lenovoX121e bluetoothd[433]: audio-avrcp-target: Protocol not supported (93)
May 04 21:07:14 lenovoX121e bluetoothd[433]: L2CAP server failed for Message Notification: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
May 04 21:07:14 lenovoX121e bluetoothd[433]: L2CAP server failed for Message Access: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
May 04 21:07:14 lenovoX121e bluetoothd[433]: L2CAP server failed for Phone Book Access: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
May 04 21:07:14 lenovoX121e bluetoothd[433]: L2CAP server failed for File Transfer: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
May 04 21:07:14 lenovoX121e bluetoothd[433]: L2CAP server failed for Object Push: setsockopt(L2CAP_OPTIONS): Invalid argument (22)
May 04 21:07:14 lenovoX121e bluetoothd[433]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSource
May 04 21:07:14 lenovoX121e bluetoothd[433]: Endpoint registered: sender=:1.68 path=/MediaEndpoint/A2DPSink

That's why sink connection works but obex don't. Will be appreciated if you guide me further.

closed 2019-05-04 by telychko

cschramm, 2019-05-05:
As far as I know, the invalid argument messages just mean that the respective protocol is not supported. Getting that for Object Push points at some issue with your obex service (Arch Linux seems to ship it with the standard bluez package, so it should be there an set up in general). Check `systemctl --user status obex.service`.

telychko, 2019-05-05:
$ LANG= systemctl status --user obex
* obex.service - Bluetooth OBEX service
   Loaded: loaded (/usr/lib/systemd/user/obex.service; enabled; vendor preset: enabled)
   Active: active (running) since Sun 2019-05-05 14:44:12 EEST; 2min 25s ago
 Main PID: 1244 (obexd)
   CGroup: /user.slice/user-1000.slice/user@1000.service/obex.service
           `-1244 /usr/lib/bluetooth/obexd

May 05 14:44:12 lenovoX121e systemd[501]: Starting Bluetooth OBEX service...
May 05 14:44:12 lenovoX121e obexd[1244]: OBEX daemon 5.50
May 05 14:44:12 lenovoX121e systemd[501]: Started Bluetooth OBEX service.

telychko, 2019-05-05:
Seems obex itself is ok. I asked for support on Arch Linux forum.

Mikaka27, 2020-10-04:
Hi @ghost, could you share a link where did you asked on arch forum? I'm facing the same problem as you do.