blueman-project/blueman #997: Blueman, Xubuntu, Lenovo T530 to MDR-xb950n1, update ruined compatibility?

hettingerw, 2019-03-09:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2019-03-24:
Hi, let me rephrase:

You used to disconnect and reconnect your headphones to get their audio working. After some update of some unknown component from an unknown old version to an unknown new version, you're not able to find the disconnect action in blueman anymore.

Is that correct? Then please tell us at least the version of blueman and we'll guide you to the disconnect item.

hettingerw, 2019-03-29:
Hello cshramm!

Huh? How did you get that so backwards? I have fixed the problem by downgrading the blueman app. The new version was the 2.0.5, and that could be easily accessed by checking the Xubuntu repo.

I am unable to get the audio working after disconnect and reconnect. The old app is clunky forcing me to go through this connect (connects as headset), disconnect, and reconnect (as A2DP sink) but it still works. I believe I am using the same "old" one. 2.0.4. I had to change some dependencies and it took some time but hey.

I'm not looking to be guided to the disconnect action. It was not available, being grayed out and not actionable. In essence, the app would try to force me to use the "headset" option, when it would not work (no audio).

I'm not sure if this is an issue with the driver for the bluetooth chip on my laptop, or the driver or whatever for the headset, or both, but I'm satisfied with old version.

Thanks,
Wes

hettingerw, 2019-03-29:
I also want to say, that I believe it's something to do with the driver, because after about an hour of using the headset, there is a notice-able lag and it skips, and then it will disconnect and reconnect.

cschramm, 2019-03-30:
So we're talking about 2.0.4 and 2.0.5.

In 2.0.4 you can get the audio back working by reconnecting the audio service. In 2.0.5 you're not able to disconnect and thus reconnect it, i.e. it looks like it is not connected. Correct?

I do not see a change between 2.0.4 and 2.0.5 (https://github.com/blueman-project/blueman/compare/2.0.4...2.0.5) that might cause that difference. @infirit: Do you? @hettingerw: You did not change any important things like BlueZ or Pulseaudio when downgrading, did you?

infirit, 2019-03-30:
I fixed a lot of bugs I introduced in 2.0.5 but nothing in connect/disconnect plus audio.

A ``blueman-manager`` log from 2.0.5 may yield insight on what is going on.

closed 2019-06-05 by stale[bot]