blueman-project/blueman #998: Error dialog after login when user language changed

zeehio, 2019-03-20:
blueman:
BlueZ:
Distribution: Ubuntu 18.04
Desktop environment: Mate
Related issues: #914, #496 

# Some facts:
Please correct me if I'm wrong:

- `blueman-applet` tries to load the default transfer directory at [plugins/applet/TransferService.py#L149](https://github.com/blueman-project/blueman/blob/master/blueman/plugins/applet/TransferService.py#L149).
- If `org.blueman.transfer.shared-path` does not exist, then `GLib.get_user_special_dir(GLib.UserDirectory.DIRECTORY_DOWNLOAD)` is used and somehow that path is stored at `org.blueman.transfer.shared-path`.
- The directory pointed by `GLib.get_user_special_dir(GLib.UserDirectory.DIRECTORY_DOWNLOAD)` depends on the system language, as given by `$XDG_DOWNLOAD_DIR` in `~/.config/user-dirs.dirs`. 
- After changing the language of my computer, on the first login Ubuntu (and I guess other distributions as well) asks me if I want to move my directories (e.g. from `~/Downloads` to `~/Baixades`, when switching from English to Catalan). And I choose yes.

# The issue:

If blueman has read `GLib.get_user_special_dir(GLib.UserDirectory.DIRECTORY_DOWNLOAD)` and copied the path to `org.blueman.transfer.shared-path` with my user language in English, blueman will continue using `~/Downloads` even after changing the user language and having moved the Downloads directory. blueman asks me that I manually change the path (as mentioned in the related issues). I can manually change the path and I have done that.

However, I would expect that blueman does not store a copy of the value `GLib.get_user_special_dir(GLib.UserDirectory.DIRECTORY_DOWNLOAD)` at `org.blueman.transfer.shared-path`, so whenever I change my user language blueman is able to automatically pick up the new Downloads directory through `GLib`.

Thanks for your time, I hope I have helped clarify the other related issues and provided enough information to fix the issue once and for all.

infirit, 2019-03-20:
Yes that's reasonable. We can make it to only store the path when the user
manually changes it in the UI.

And we should probably allow the user to reset to default from the error
dialog.

infirit, 2019-04-27:
See #1012, I went a slightly different route but I think this is good enough. Feel free to comment if folks disagree :smile:

PR #1012 (2019-04-28): Handle invalid directory for incoming file transfers
Fixes #998
closed 2019-04-28 by infirit