blueman-project/blueman #1000: Bluetooth does not allow search or setup including pairing

beardedragon2, 2019-03-31:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2019-03-31:
What are your results or where do you need help with https://github.com/blueman-project/blueman/wiki/Troubleshooting#no-adapter-or-not-able-to-turn-on-bluetooth?

beardedragon2, 2019-04-01:
Does this help?
[bob@Manjaro-Stable ~]$ systemctl status bluetooth
● bluetooth.service - Bluetooth service
   Loaded: loaded (/usr/lib/systemd/system/bluetooth.service; enabled; vendor preset: disabled)
   Active: active (running) since Sun 2019-03-31 19:08:38 CDT; 10min ago
     Docs: man:bluetoothd(8)
 Main PID: 464 (bluetoothd)
   Status: "Running"
    Tasks: 1 (limit: 4915)
   Memory: 2.2M
   CGroup: /system.slice/bluetooth.service
           └─464 /usr/lib/bluetooth/bluetoothd

Mar 31 19:08:37 Manjaro-Stable systemd[1]: Starting Bluetooth service...
Mar 31 19:08:37 Manjaro-Stable bluetoothd[464]: Bluetooth daemon 5.50
Mar 31 19:08:38 Manjaro-Stable bluetoothd[464]: Starting SDP server
Mar 31 19:08:38 Manjaro-Stable systemd[1]: Started Bluetooth service.
Mar 31 19:08:38 Manjaro-Stable bluetoothd[464]: Bluetooth management interface 1.14 initialized

beardedragon2, 2019-04-01:
Desperate times calls for desperate measures.  Installed Debian based xubuntu in place of Arch based manjaro with the thought it has a different blueman version 2.0.5.  Sadly to no avail.   Still does not work.  I am wondering now if this is not a case of buetooth part gone bad in my computer.   I need it for my hearing aids.  Anyway I ordered a USB dongle to try that out, keep in touch.

beardedragon2, 2019-04-01:
Seems I am not the only one, see this link:
https://forum.manjaro.org/t/bluetooth-woes-no-default-controller-available/81228?u=beardedragon

beardedragon2, 2019-04-02:
Stillworking on trying to find the problem:

[bob@Manjaro ~]$ dmesg

8.611356] Bluetooth: Core ver 2.22
[    8.611368] NET: Registered protocol family 31
[    8.611369] Bluetooth: HCI device and connection manager initialized
[    8.611372] Bluetooth: HCI socket layer initialized
[    8.611373] Bluetooth: L2CAP socket layer initialized
[    8.611377] Bluetooth: SCO socket layer initialized

[10.906103] Bluetooth: hci0: command 0x1001 tx timeout

12.815885] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   12.815886] Bluetooth: BNEP filters: protocol multicast
[   12.815889] Bluetooth: BNEP socket layer initialized

18.799407] Bluetooth: hci0: RTL: HCI_OP_READ_LOCAL_VERSION failed (-110)

beardedragon2, 2019-04-02:
> What are your results or where do you need help with https://github.com/blueman-project/blueman/wiki/Troubleshooting#no-adapter-or-not-able-to-turn-on-bluetooth?

See edited first entry.

cschramm, 2019-04-02:
Seems like you're using some PCI Realtek hardware and might be missing firmware. Check `dmesg | grep firmware`. On Ubuntu firmware-realtek might help. I don't know about Manjaro.

beardedragon2, 2019-04-02:
FYI I decided to use Debian Xfce, it is a fine operating system and I am familiar with it.  Same as xubuntu, but, better and more up to date.

beardedragon2, 2019-04-02:
Solved !!!!!!
Bought bluetooth dongle then installed the following drivers and firmware
bluez-firmware
broadcom-sta-common
broadcom-sta-source

beardedragon2, 2019-04-03:
One last item you might like to know.  I reinstalled Manjaro and it already had the drivers for the dongle because it looks at the hardware when you install it.  It was the bluetooth hardware all along, thanks for the time and effort you put in.

closed 2019-04-03 by beardedragon2