blueman-project/blueman #1001: Exception At Startup

calvinnnh, 2019-03-31:
blueman: 2.0.5
BlueZ:
Distribution: Mint 19.1
Desktop environment: Cinnamon
* [x ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

calvinnnh, 2019-03-31:
[exception.docx](https://github.com/blueman-project/blueman/files/3026797/exception.docx)

cschramm, 2019-04-07:
The file `/var/lib/blueman/network.state` is corrupt. Please provide it so that we can look into what happend.

To get blueman running again, you can just remove the file afterward. Unless you configured a local network access point (NAP), you should be fine.

@infirit: Should we catch the `UnpicklingError` in addition to `IOError` and `UnicodeDecodeError`?

calvinnnh, 2019-04-07:
Thanks very much for help.
[network.state.zip](https://github.com/blueman-project/blueman/files/3051456/network.state.zip)

calvinnnh, 2019-04-07:
That solved the original problem. However; now I've lost the ability to Audio Sink my bluetooth speakers (I don't even get the option to Audio Sink). Prior to this I was getting the error message "Connection failed: blueman.bluez.errors: DbusFaledError: resource temporarily unavailable"

calvinnnh, 2019-04-07:
I hit the wrong button> It is NOT Closed!!!

calvinnnh, 2019-04-07:
Update. I uninstalled the pulse audio bluetooth module, then re-installed. Everything is now working correctly. Thanks for your help. It is now closed as far as I am concerned.

calvinnnh, 2019-04-07:
Issue closed

closed 2019-04-07 by calvinnnh

infirit, 2019-04-07:
@cschramm yeah adding ``UnpicklingError`` to the exceptions is a good idea.

cschramm, 2019-04-08:
Thank you for the sample, @calvinnnh. Unfortunately I have no idea what I'm looking at there. I cannot find any pickle-ish structure in there at all... :thinking:

calvinnnh, 2019-04-08:
@cschramm  Anything more I can do to help? I am really a novice at this stuff. I do know that when I deleted the network.state file and removed and reinstalled Pulseaudio-module-bluetooth as stated above I got everything working again. I have noticed that the little "device" icon in the tray pop-up box is no longer blue colored as in the past but functionally everything seems to be working as in the past. As I stated above before the UnpicklinError I had the "Connection failed: blueman.bluez.errors: DbusFailedError: resource temporarily unavailable" message and could not connect to my speakers even though it appeared that the device was recognized and it appeared that I was paired with the device. I have not tried removing and reinstalling blueman after implementing your suggestions even though I did try this prior to getting in touch with you, but to no avail.

cschramm, 2019-04-27:
@calvinnnh: No thanks. Unfortunately I have no idea what caused your corrupt file, but in any case with #1003 in place, blueman will just reset the file if it's corrupt.

PR #1003 (2019-04-08): [NetConf] Handle UnpicklingError
Fixes #1001
commit 70411358 (2019-04-08): [NetConf] Handle UnpicklingError (#1003)

Fixes #1001