blueman-project/blueman #1009: Type hints?

cschramm, 2019-04-23:
I'm thinking about using type hints as they make some parts of coding (and verifying) a lot easier.

That would raise the minimum Python version to 3.5 and drop the (supposed) support for 3.2, 3.3, and 3.4, however, all of those versions are end-of-life anyway.

Any thoughts?

infirit, 2019-04-23:
I looked at this a while back but dismissed for the time being as pygobject does not implement pep 484,  see https://gitlab.gnome.org/GNOME/pygobject/issues/159. Lazka has since started to generate stubs, see https://github.com/pygobject/pygobject-stubs. I have not used the stubs so not sure how well they actually work.

You can do type hints as comments in python version before 3.2 - 3.5, see https://mypy.readthedocs.io/en/latest/cheat_sheet_py3.html. However I would prefer to use the new syntax introduced in python 3.6.

cschramm, 2019-04-23:
Hm, right, i only really thought about function and method signature but for useful variable type hints we'll need 3.6 of course, which means we'll drop support for 3.5. Fine for me. :+1:

infirit, 2019-04-27:
Ok with me then but after releasing 2.1 please :grin:

cschramm, 2019-04-27:
:+1: So 2.2 will be Python 3.6+

closed 2019-04-27 by cschramm