blueman-project/blueman #1013: Surface Ergonomic Keyboard Cannot Connect

kartik-checkout51, 2019-04-28:
blueman: 2.0.5
BlueZ: 
Distribution: Ubuntu 19.04
Desktop environment: Gnome 3.32.1

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hi Folks,

I'm having trouble connecting my surface ergonomic keyboard with my Ubuntu laptop.  When I try pairing the device using the following sequence of commands:

https://pastebin.com/st3UdL15

I get a message saying `org.bluez.Error.AuthenticationFailed`.  My understanding is that I should be prompted with a PIN code that I should be typing on my keyboard at this point.  I tried this from both the command line and the GUI, without luck.  When I run blueman-applet with debug output i see something like this:

https://pastebin.com/DiyuTq4Y

Also, when I try and follow the troubleshooting steps for *Pairing fails on the remote device's side wihout a pairing dialog showing*, I get the following error message when  running the python script in the terminal: 

```>>> from blueman.gui.Notification import Notification
>>> Notification("Title","Message",0,[["action1",_("Action 1")], ["action2",_("Action 2")]])
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
TypeError: 'Notification' object is not callable
```
Any help would be appreciated!

-K

cschramm, 2019-04-28:
Hi, first thing I'd do is check the bluetoothd logs (should be in `/var/log/daemon.log`) when you get `AuthenticationFailed` to see if it gives you any additional information (otherwise you could stop bluetoothd and start it manually with `-d` for debug logs). Usually a key has to get presented and typed into the keyboard for paring. Both blueman's and the bluethoothctl agent support that, so probably something goes wrong in an earlier phase of pairing.

Regarding notification debugging: Are you sure the import statement is correct? Does typing `Notification` give you `<class 'blueman.gui.Notification.Notification'>` after the import? In any case, I don't expect you to find any issues there, as if the blueman notification would be the problem, it should still work with the bluetoothctl agent.

cschramm, 2019-04-28:
> In any case, I don't expect you to find any issues there, as if the blueman notification would be the problem, it should still work with the bluetoothctl agent.

Strike that. When trying with `bluetoothctl` use `default-agent` to make sure its agent gets instead of e.g. blueman's. *Then* it should work there and if it does, it's a problem with blueman's agent or notification.

kartik-checkout51, 2019-04-28:
Indeed, it doesn't work with default-agent, so the issue is probably lower down in the stack somewhere.

Regarding the Notification issue,  it looks like the problem was me, as I didn't copy the entirety of the python statement to issue the notification.  Thanks for the pointers, and sorry for the false alarm.

closed 2019-04-28 by kartik-checkout51

cschramm, 2019-04-28:
Curious to hear from you in case you find a cause or solution. Pairing a surface ergonomic keyboard in Linux seems to be a constant issue.

kartik-checkout51, 2019-04-28:
After a lot more googling, and debugging, it turns out it was a firmware issue.  `dmesg | grep -i bluetooth` revealed a bunch of errors around not being able to properly load the bluetooth firmware for my machine.  Updating it resolved the issue.