blueman-project/blueman #1018: Turn on bluetooth on boot might need update in Wiki Troubleshooting page

Ding-Fan, 2019-04-29:
https://wiki.archlinux.org/index.php/Bluetooth#Auto_power-on_after_boot

> By default, your Bluetooth adapter will not power on after a reboot. The former method by using `hciconfig hci0 up` is deprecated, see the [release note](http://www.bluez.org/release-of-bluez-5-35/). Now you just need to add the line `AutoEnable=true` in `/etc/bluetooth/main.conf` at the bottom in the [Policy] section:

```
/etc/bluetooth/main.conf
[Policy]
AutoEnable=true
```

infirit, 2019-04-29:
I am confused,  are you asking us (blueman project) to update the arch linux wiki?

Ding-Fan, 2019-04-30:
> I am confused, are you asking us (blueman project) to update the arch linux wiki?

Sorry for my bad English skill. I mean to update this https://github.com/blueman-project/blueman/wiki/Troubleshooting (at the bottom)

infirit, 2019-05-04:
Ah, I see now. Fixes this in the Wiki. Thanks for the report.

closed 2019-05-04 by infirit