blueman-project/blueman #1024: Please clarify/unify the license

mgorny, 2019-05-07:
I've been working on verifying the license for blueman package in Gentoo, in particular establishing the correct GPL versions. What I've been able to find is:

1. `README` says 'GPL' which is ambiguous.
2. `COPYING` file is a copy of GPL-3 license.
3. Icons specify GPL-2 in license tags.
4. `blueman/Functions.py` and `module/libblueman.c` says `Licensed under the GNU General Public License Version 3` (i.e. GPL-3) on top, and immediately follows it with `either version 3 of the License, or (at your option) any later version.` notice.
5. `data/pin-code-database.xml` and `module/modem-prober.*` have 'GPL-2 or newer' notice.

Could you please:
1. Clarify the license in 4., presumably by either removing the misleading statement on top (misleading into believing licenses newer than GPL-3 will not be acceptable) or changing the copyright notice not to say 'or newer'?
2. Clarify the license for files missing copyright notice, and add copyright notices to them as required by GPL terms?
3. Look whether the licensing can be unified, in particular whether GPL-2+ licensed parts can legally be upgraded to GPL-3+?
4. Expand the license information in README to clearly indicate which GPL version(s) are acceptable? Also, make it explicit that some parts are licensed differently.

cschramm, 2019-05-07:
Hi Michał, first and foremost: I'm neither an attorney nor very familiar with the GPL topic, so I'm very open for explanations and advice.

My thoughts on your points:

1. The text in 4. makes sense to me. Things are licensed under GPLv3 and thus can be redistributed under GPLv3 or newer.

2. Adding copyright notices is just a recommendation and I'm not really willing to maintain them.

3. As GPLv2 allows redistribution under newer versions, it seems to be very well possible to unify our license to GPLv3. I will evaluate that. However, for the very same reason it seems absolutely correct to me to distribute the Gentoo package with GPLv3 already.

mgorny, 2019-05-07:
Ok, so first things first. You can use each version of GPL in two ways: as 'GPL-n only', or 'GPL-n or newer'. FSF seems to encourage the latter but it's up to you, and some people distrust future license versions and therefore allow only specific version. In other words, GPL itself does not allow newer versions, unless you explicitly say so.

For this reason, it is important to note which of the options you choose, and appropriate say 'version N or newer', or 'version N only'. Any other form is ambiguous. The best way to do this is use copyright notices, as they make it clear which license applies to which file, and there's no risk that e.g. a contributor will make a wrong assumption.

'GPL-3 or newer' is a subset of 'GPL-2 or newer', so if you have files using both, you can generally say the user needs to agree with 'GPL-3 or newer'. However, if you have files stating 'GPL-3 or newer' and 'GPL-2' (without the 'or newer' clause), the user needs to agree separately on terms of GPL-2 and 'GPL-3 or newer'. Of course, there is also the matter of license compatibility but I don't think we need to worry about it here.

cschramm, 2019-05-07:
I see. So the primary concern here seems to be if the GPLv2 in the SVG files allows "any later version" or not. As the GPLv2 text sounds like these words need to get mentioned to apply and they are not included in the SVG code, I actually doubt that.

infirit, 2019-05-07:
Commit c00237568642f6a48bec2abb1fb017711117bc2f suggest the original authors wanted it to be gpl3 or later. The pindb and /modem-prober are taken from other projects and they are already gpl2 or later so if I understand @mgorny this can be considered then gpl3 or later as well.

So what is missing is mentioning that the vector images are gpl2 only, right..

cschramm, 2019-05-08:
I realized that the relevant commit history is very dense:

```
* | c0023756 - Update copyright <tadas>  <-- updates COPYING to GPLv3
* | 1b26de16 - Fix memory leaks and cleanup a bit <tadas>
* | b8c863ee - few gui tweaks <walmis>
* | 80986fdb - gui test 2 <walmis>
* | 97864238 - update ui template <walmis>
* | 6f0f63e4 - Initial autotools magic <tadas>  <-- adds COPYING with GPLv2
* | a8e7723f - update up/down icons <walmis>
* | 0deb86dd - add a few icons <walmis>  <-- adds the base icons (blueman and blueman-device) with the GPLv2 reference
```

Newer GPLv2 references are just due to exact copies of the base icon or derivative work done by me (blueman-disabled) or @walmis (blueman-active).

So it seems all we need to do to upgrade to GPLv3 is get permission for the blueman, blueman-device, and blueman-active icons from @walmis. I will try to reach out to him (@walmis, if you're reading, please jump in :slightly_smiling_face:).

infirit, 2019-05-08:
Looking at the source of the vector images I doubt he is the actual author and more likely taken from an icon theme like tango. Searching github for ``verbatim-store-n-go-1gb`` and ``bluetooth-applet-48.svg``  results an many hits with varying modifications and licenses. But they all are very likely from the same original source.

Can't we just mention the icons are gpl2 only and the rest gpl3 or later and be done with this?

cschramm, 2019-05-08:
You're right. @holtmann added that icon to bluez-gnome (which is GPLv2+) and it seems to have been created by Andreas Nilsson in the tango project. Might just be based on some other work from him though as the icon does not seem to have ever actually been in the tango icon scheme.

infirit, 2019-05-09:
The [bluetooth icon](https://gitlab.gnome.org/GNOME/gnome-bluetooth/blob/master/icons/hicolor/scalable/apps/bluetooth.svg#L159) in gnome-bluetooth git master still mentions gpl2 so we are pretty much in the same boat as them. 

@mgorny the icon situation between gnome-bluetooth and blueman is the same. And for it gentoo has ``LICENSE="GPL-2+ LGPL-2.1+ FDL-1.1+"``. So if that is incorrect they'll need to fix up the license in the vector image as well as the our blueman icon is pretty much identical as theirs.

PR #1042 (2019-06-09): Add information about the mixed GPL licenses
Closes #1024
closed 2019-06-09 by cschramm