blueman-project/blueman #1025: Can't send files

p1839251, 2019-05-10:
blueman: 2.0.8
BlueZ: 5.50
Distribution: Archlinux
Desktop environment: XFCE

Hello,
I've been struggling with blueman for weeks. I'm using an up-to-date Archlinux with XFCE.
I just can't send files over bluetooth. However, I could send files with 'blueberry'. (I haven't tested thoroughly but it seems I can receive files) 
I found my device and could pair them well. I tried sending files from Thunar and from the blueman-applet as well. Neither of them worked.

I saved the output, and since it's not too long, I'm going to paste it here:
```
/usr/bin/blueman-sendto:374: PyGTKDeprecationWarning: Using positional arguments with the GObject constructor has been deprecated. Please specify keyword(s) for "title" or use a class specific constructor. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  "_OK", Gtk.ResponseType.ACCEPT))
/usr/bin/blueman-sendto:374: PyGTKDeprecationWarning: The "buttons" argument must be a Gtk.ButtonsType enum value. Please use the "add_buttons" method for adding buttons. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  "_OK", Gtk.ResponseType.ACCEPT))
Sending to 55:50:2A:3D:0F:8F
```
And nothing really happens from here. I can't see any incoming files on my phone.
(I've tried multiple devices and I experienced the very same issue)
Let me know if you need more information. I don't know how I could be more helpful.

infirit, 2019-05-10:
The thunar plugin call ``blueman-sendto`` with the path of the file. You can run this fro ma terminal like

```
$ blueman-sendto /path/to/file
```

Can you post the output when you run this from a terminal.

p1839251, 2019-05-11:
Sure thing!
There you go: https://bpaste.net/raw/36cdd3fdae27

infirit, 2019-05-12:
The error means we try to stop discovery but for some reason bluez is already doing this. Does the below patch make it work for you?

```diff
diff --git a/blueman/gui/DeviceList.py b/blueman/gui/DeviceList.py
index 4d91b01b..3f7acf6d 100644
--- a/blueman/gui/DeviceList.py
+++ b/blueman/gui/DeviceList.py
@@ -11,6 +11,7 @@ from blueman.main.FakeDevice import FakeDevice
 from blueman.main.Device import Device
 
 from _blueman import conn_info
+from blueman.bluez.errors import DBusInProgressError
 import blueman.bluez as Bluez
 
 import gi
@@ -308,7 +309,11 @@ class DeviceList(GenericList):
         if progress >= 1.0:
             progress = 1.0
         if self.__discovery_time >= totaltime:
-            self.StopDiscovery()
+            try:
+                self.StopDiscovery()
+            except DBusInProgressError:
+                dprint("Already stopping discovery")
+                pass
             return False
 
         self.emit("discovery-progress", progress)
```

p1839251, 2019-05-12:
Not really :/
I'm still unable to send files:

```
$ blueman-sendto /tmp/test.jpg 
/usr/bin/blueman-sendto:388: PyGTKDeprecationWarning: The "buttons" argument must be a Gtk.ButtonsType enum value. Please use the "add_buttons" method for adding buttons. See: https://wiki.g
nome.org/PyGObject/InitializerDeprecations
  d = DeviceSelectorDialog()
_________
get_interface_version (/usr/lib/python3.7/site-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
_________
SetAdapter (/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py:272)
None 
_________
__init__ (/usr/lib/python3.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
add_device (/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py:322)
adding new device 
_________
do_cache (/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py:501)
Caching new device 55:50:2A:3D:0F:8F 
_________
on_property_changed (/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py:156)
adapter propery changed Discovering 1 
_________
on_device_property_changed (/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py:164)
list: device_prop_ch RSSI -37 /org/bluez/hci0/dev_50_55_27_3D_0F_9F 
_________
__init__ (/usr/lib/python3.7/site-packages/blueman/main/Device.py:26)
caching initial properties 
_________
get_interface_version (/usr/lib/python3.7/site-packages/blueman/bluez/obex/Base.py:20)
Detected BlueZ integrated obexd 
Sending to 55:50:2A:3D:0F:8F
_________
update_progress (/usr/lib/python3.7/site-packages/blueman/gui/DeviceList.py:302)
Already stopping discovery
```

infirit, 2019-05-13:
I was hoping it would be enough sadly not. An obex session should be created when your output stops.

You said you have blueberry installed and it requires gnome-bluetooth. Can you uninstall blueberry and gnome-bluetooth and confirm it work or not? You could probably just kill both but to be safe uninstall.

p1839251, 2019-05-15:
I was very careful and the two programs weren't installed in parallel.
I'd removed blueman when I tried blueberry and I don't have blueberry installed on my computer anymore.
For what it's worth, here's what blueman looks like on Arch:
https://git.archlinux.org/svntogit/community.git/tree/trunk/PKGBUILD?h=packages/blueman
The package is not completely untouched, but I don't really think this would be the problem either.

infirit, 2019-05-15:
That's annoying, looks like i need to create an arch VM and have a look for
myself.

infirit, 2019-05-15:
Have one with xfce4 setup but unfortunately I can't reproduce. Does journalctl show you any errors related to obexd?

p1839251, 2019-05-17:
What I forgot to mention is if I rather persistent, I can occasionally send files over bluetooth.
One file at a time of course. I've managed to send 3 files in this month so far.

I'm providing all the information you asked me:
blueman-applet : https://bpaste.net/show/b5d4c77564d9
blueman-sendto: https://bpaste.net/show/cff82c9438cd
journalctl: https://bpaste.net/show/6b57277be7c0

(I reverted to the unpatched vanilla Arch package)

p1839251, 2019-06-06:
Would I be better off using Blueberry?

closed 2019-08-14 by stale[bot]

cschramm, 2021-12-05:
That is an interesting finding, @abhidesh128. So if you have an (outgoing) file transfer in progress you can break it e.g. with `bluetoothctl scan on`? I can not replicate that on my side but it sounds like a driver / firmware issue that we might want to handle as we also have #1485 in place.

abhidesh128, 2021-12-06:
Hi @cschramm..!
I recently upgraded my Linux distribution and got new versions of both Blueman and Bluez.

It has Blueman 2.1.4 and Bluez 5.55. Everything is working perfectly and smoothly. I am no longer getting exceptions and errors. Devices are discovered correctly and file transfers can be initiated and successfully completed by both sides.

I guess it was not only Blueman's fault, but Bluez also had some issue. Now as both are upgraded, the issues seem to be fixed.

Thank you for your support..!

cschramm, 2021-12-06:
Perfect, thanks, @abhidesh128. I'd still suspect drivers / firmware but those probably got an updates as well. Nothing to care about on our side then, I'd say. :+1: