blueman-project/blueman #1031: Can't configure Blueman to default to High Fidelity Playback (A2DP Sink)

scottgreenup, 2019-05-21:
blueman: 2.0.8-2
BlueZ: 5.50-6
Distribution: Arch Linux
Desktop environment: AwesomeWM

* [X] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

I am using Sony WH-1000MX2 headphones. If I connect to them via Bluetooth (either manually or automatically) using the blueman-applet then they will come up as `Headset Head Unit (HSP/HFP)`. The option for A2DP is there but when I select it, it fails to connect to the sink.

To fix this, I have to disconnect the device then reconnect it selecting `Audio Sink` from the right-click context menu.

I would imagine there could be a way to have it automatically connect and default to the `High Fidelity Playback (A2DP Sink)` mode?

I've tried following the 2nd work-around in the troubleshooting guide at https://wiki.debian.org/BluetoothUser/a2dp by disabling the Bluetooth in PulseAudio. But all I get is a DBUS error in the blueman-applet interface that says protocol not available.

Any ideas or tips would be appreciated.

cschramm, 2019-05-21:
That pulseaudio issue where switching the profile fails (what's the error message?) seems to be related to multiple pulseaudio daemons / users (the GDM user in case of the workaround). If you're really having that exact problem, you should see multiple pulseaudio processes in your system. Did you try swichting the profile with another method than blueman, e.g. `pavucontrol`?

scottgreenup, 2019-05-21:
I have tried using `pavucontrol` but it comes up as "(unavailable)". I'll get screenshots of everything soon.

With regards to having multiple daemons running; it seems I only have one:
```
$ ps aux | grep pulse
${USER}  8151  0.0  0.0   6272  2296 pts/15   S+   17:20   0:00 grep -i --color pulse
${USER} 17234  2.9  0.0 1960936 9708 ?        S<sl 10:37  11:54 /usr/bin/pulseaudio --daemonize=no
${USER} 17254  0.0  0.0 247644  3640 ?        Sl   10:37   0:00 /usr/lib/pulse/gsettings-helper
```
Even digging through `systemctl` shows the same daemon under the `pulseaudio.service`

cschramm, 2019-05-21:
So this does not look like https://wiki.debian.org/BluetoothUser/a2dp#Refused_to_switch_profile_to_a2dp_sink:_Not_connected https://wiki.archlinux.org/index.php/Talk:Bluetooth_headset#GDMs_pulseaudio_instance_captures_bluetooth_headset https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=845938 https://gitlab.freedesktop.org/pulseaudio/pulseaudio/issues/303 ...

In #997 @hettingerw indicates to use a similar workflow to get the audio working (he might have that mentioned issue though as I think Xubuntu uses GDM). Other than that, I have no idea or experience with the issue.

Did you consult pulseaudio yet? (https://www.freedesktop.org/wiki/Software/PulseAudio/Documentation/User/Community/) https://gitlab.freedesktop.org/pulseaudio/pulseaudio/issues/87 might be related but unfortunately it's unanswered.

scottgreenup, 2019-05-29:
Thanks for chasing this up @cschramm!

#997 seems like it's a similar vein of problem, however, I've had this problem for a while now and never chased up on it.

I've tried some things with PulseAudio but I think it's a blueman issue. Therefore (from what I can gather) I don't think https://gitlab.freedesktop.org/pulseaudio/pulseaudio/issues/87 seems to be my issue.

Here is a screen recording so you can see the issue and how I disconnect/reconnect to fix it: https://gfycat.com/flaweddrearycondor . In the video, bluetooth has automatically connected to my headphones when I turn on the headphones. It drops me in to the state you see in the video. I disconnect, then connect to get the A2DP sink.

Here are the logs from `journalctl -b 0 | grep -i -e blue -e pulse` around the time of the issue.

```
May 29 16:58:53 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 0
May 29 16:58:54 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 0
May 29 16:58:54 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 0
May 29 16:58:54 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 0
May 29 16:58:54 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 0
May 29 16:59:00 hostname pulseaudio[1209]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink: Not connected
May 29 16:59:04 hostname pulseaudio[1209]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink: Not connected
May 29 17:00:39 hostname pulseaudio[1209]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink: Not connected
May 29 17:01:20 hostname pulseaudio[1209]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink: Not connected
May 29 17:01:29 hostname pulseaudio[1209]: W: [pulseaudio] module-bluez5-device.c: Refused to switch profile to a2dp_sink: Not connected
May 29 17:01:39 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 257
May 29 17:01:39 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 257
May 29 17:01:44 hostname bluetoothd[581]: /org/bluez/hci0/dev_04_5D_4B_DE_69_DF/fd1: fd(36) ready
May 29 17:01:51 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 0
May 29 17:01:51 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 0
May 29 17:01:51 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 0
May 29 17:01:51 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 0
May 29 17:01:51 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 0
May 29 17:01:52 hostname kernel: Bluetooth: hci0: SCO packet for unknown connection handle 257
```

cschramm, 2019-05-29:
Why do you think it's a blueman issue? You can most probably easily replicate it without blueman:

1. Stop blueman-applet
2. Power on the headphones
3. Check that they are connected but the A2DP sink is unavailable
4. Use bluetoothctl to disconnect and connect the device
5. Check that the A2DP sink is now available

blueman actually does not do anything else in that process than proving you with connect and disconnect (like bluetoothctl, routed to BlueZ) and audio profile switching (like pavucontrol, routed to pulseaudio) facilities.

closed 2019-08-04 by stale[bot]

nicomazz, 2019-10-02:
Hi all!
I have the next version of the headphones mentioned (Sony WH-1000MX3), and I'm having the same problem. 
I don't think it is a problem with Blueman, since also when the applet is stopped, my headphones connect with HSP/HFP profile anyways by default.
However, after all this time, have you found other solutions?

cschramm, 2019-10-03:
Hi, I'm not aware of anything useful. You should probably consult the pulseaudio project to dig into the matter.

camsn0w, 2019-12-30:
> Hi all!
> I have the next version of the headphones mentioned (Sony WH-1000MX3), and I'm having the same problem.
> I don't think it is a problem with Blueman, since also when the applet is stopped, my [...]

I have the Sony WH-1000MX3 too and am having the same problem here it looks to be a Sony MX* problem.

hugokerstens, 2020-05-06:
I'm experiencing the same 'issue' with my Sony WH-1000XM3. However, when connecting the headphones to the same PC but booted into windows, the profile switching works just fine and happens automatically whenever the microphone is required. The same holds for my android phone.

So even though Sony may be at fault here, I figure it's possible to fix this on bluemans side.

cschramm, 2020-05-08:
BlueZ 5.51 brought an improvement for those devices (see https://github.com/blueman-project/blueman/issues/1242#issuecomment-625170430) and there's at least one case in https://github.com/blueman-project/blueman/issues/1209#issuecomment-625155850 where BlueZ 5.53 actually helped with it.

Seb35, 2020-09-23:
I had this issue with a speaker JBL GO 2 and it was just solved by installing BlueZ 5.54 from Debian buster backports.

Thanks to @cschramm for your advices, to @iwamatsu for the packaging, and to other BlueZ folks for fixing this issue (and the whole project).

N3r0m5ns3r, 2021-07-04:
same just got a mx3 and the error msg High Fidelity Playback A2DP Sink not available pops out. I'm am on fedora.
this must be a sony issue right?

pradeeplogan, 2021-10-04:
Hello Guys,

I have Sony WH-1000XM3 and was experiencing similar problem and following the below instruction solved it.

https://askubuntu.com/a/1228816