blueman-project/blueman #1032: Connection Failed: The name org.freedesktop.NetworkManager was not provided by any .service files

sohasanl, 2019-05-22:
blueman:
BlueZ:
Distribution:
Desktop environment:

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2019-05-22:
Please see the troubleshooting page for instructions on how to submit information.

infirit, 2019-05-22:
Blueman uses NetworkManager to provide nap/dun support but we can't hope to provide any assistance if not even the most basic information is provided.

Zeioth, 2019-05-26:
I can reproduce:

**Distro**: Antergos (latest)
**Log**: [here](https://pastebin.com/raw/mUWz3YYH)

Please provide me more info so I can test possible solutions. Cheers.

cschramm, 2019-05-26:
@Zeioth the issue in your log seems unrelated and should be fixed in recent versions (try 2.1.beta1).

Zeioth, 2019-05-26:
@cschramm I did:

    yaourt -S blueman-git
    sudo rm -rf /var/lib/bluetooth
    sudo systemctl enable blueman-mechanism.service 

It works now, thank you.

closed 2019-08-01 by stale[bot]