blueman-project/blueman #1037: 2.1

cschramm, 2019-05-29:
There are several enhancements / features in the 2.1.0 milestone which we could discuss deferring in favor of a quick release as there's no problem with releasing those bits with a 2.2 soon after.

#853 provides a draft for a backend for auto-connect but it'll probably need some more tuning and the necessary GUI parts for #239 does not exist yet and I will not find enough time for it at the moment.

#69 has been in the loop for five years now, so we're not in a hurry with it either. :wink:

#1030 provides fixes for actual bugs, so it's release critical. It should resolve #761 completely, correct?

What about #890? It's not nice but not a regression. Not sure how much effort it requires.

Overall, I think we can boil down the open issues for the 2.1.0 release to just #1030 target a quick 2.2 release.

infirit, 2019-05-29:
All for getting something out of the door.

#890 let me just have a quick look and check if we don't crash.

#1030 agreed and I'll do the missing parts this weekend.

closed 2019-05-29 by cschramm

cschramm, 2019-06-09:
It just pushed the button. :tada: :champagne:

infirit, 2019-06-09:
Nice, saw the notification :+1: :balloon: