blueman-project/blueman #1038: blueman-manager crashed with TypeError in close_rfcomm(): a bytes-like object is required, not 'str'

l-dobrev, 2019-05-30:
blueman: 2.0.5-1ubuntu2
BlueZ: 5.50-0ubuntu2
Distribution: Xubuntu 19.04
Desktop environment: Xfce

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Blueman Manager repeatedly crashes on me. After some digging around I found out the following exception:
```python
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/blueman/gui/manager/ManagerDeviceMenu.py", line 158, in on_disconnect
    appl.disconnect_service(service.device.get_object_path(), service.uuid, port)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 70, in __call__
    return self._proxy_method(*args, **keywords)
  File "/usr/lib/python3/dist-packages/dbus/proxies.py", line 145, in __call__
    **keywords)
  File "/usr/lib/python3/dist-packages/dbus/connection.py", line 651, in call_blocking
    message, timeout)
dbus.exceptions.DBusException: org.freedesktop.DBus.Python.TypeError: Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/dbus/service.py", line 707, in _message_cb
    retval = candidate_method(self, *args, **keywords)
  File "/usr/lib/python3/dist-packages/blueman/main/DbusService.py", line 36, in wrapper
    return method(*args[1:], **kwargs)
  File "/usr/lib/python3/dist-packages/blueman/plugins/mechanism/Rfcomm.py", line 26, in close_rfcomm
    if command in line:
TypeError: a bytes-like object is required, not 'str'
```

Puzzled by this I tried to reproduce the issue and found out the following:
When using **Python-2** the `subprocess.Popen(...).communicate()` returns a tuple of **str** objects.
When using **Python-3** the `subprocess.Popen(...).communicate()` returns a tuple of **bytes** objects.

Hence the issue: with **Python-3** the ../mechanism/Rfcomm.py:[24-27] tries `command in line`, but the line is not `<type 'str'>` but `<class 'bytes'>`.

Work-Around: None to the best of my knowledge.

Naive fix: replace `command in line` with `command in line.__str__()`.

infirit, 2019-05-30:
This was fixed in commit 222d0532cd9164500056cd56fa6542ba17f94ad0 which got released in version 2.0.7.

cschramm, 2019-05-30:
For unknown reasons Ubuntu developers did not migrate blueman packages from Debian for over a year so that 18.10 and even 19.04 got released with the deprecated 2.0.5 version. Maybe just nobody takes care of the Ubuntu package. You might want to file a bug for it but unfortunately nobody really seems to care about [its bugs](https://launchpad.net/ubuntu/+source/blueman/+bugs) as well...

To get the fixes you can [download Debian buster's 2.0.8 package](https://packages.debian.org/buster/amd64/blueman/download). As far as I know it should install and work fine. All dependencies should be available in Ubuntu 19.04.

(@flexiondotorg also has a disco build in [his PPA](https://launchpad.net/~flexiondotorg/+archive/ubuntu/playground) but clearly militates against using the PPA in its description. :smile: )

closed 2019-05-30 by cschramm

flexiondotorg, 2019-05-31:
Yes, my build in the PPA was prep for migrating the newer version from Debian to Ubuntu. Do you recommend we sync against 2.0.8 or 2.1alpha?

cschramm, 2019-06-01:
@flexiondotorg: Cool. :+1: In general I'd recommend syncing with Debian testing / unstable. I'd upload 2.1.beta1 there if we weren't in freeze period at the moment. Since we are, we'll just upload it to experimental instead but you could take it from there. It should show up in the next couple of days. I'm still waiting for my sponsor.

flexiondotorg, 2019-06-01:
Excellent, thanks for the info. I'll sync from experimental 👍