blueman-project/blueman #1040: ManagerStats.py:21:__init__:AttributeError: 'NoneType' object has no attribute 'get_object_path'

suve, 2019-06-04:
blueman: 2.1-beta1, 2.1-alpha3
BlueZ: 5.50
Distribution: Fedora 30/29
Desktop environment: _various_

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

Hi, I'm the maintainer for `blueman` in Fedora. Recently Fedora's automated bug reporting tool filed a couple instances of the following bug occurring:
```
ManagerStats.py:21:__init__:AttributeError: 'NoneType' object has no attribute 'get_object_path'

Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/blueman/main/Manager.py", line 137, in on_dbus_name_appeared
    self.Stats = ManagerStats(self)
  File "/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerStats.py", line 21, in __init__
    self.hci = adapter_path_to_name(blueman.List.Adapter.get_object_path())
AttributeError: 'NoneType' object has no attribute 'get_object_path'
```

Red Hat Bugzilla links:
https://bugzilla.redhat.com/show_bug.cgi?id=1713129
https://bugzilla.redhat.com/show_bug.cgi?id=1693552
https://bugzilla.redhat.com/show_bug.cgi?id=1668128

Users mention that they get this crash when bluetooth is not enabled and they click "yes" when blueman asks if they want to enable bluetooth.

Since I'm the package maintainer, I'd like to ask: apart from asking users to follow the troubleshooting page, and trying to write patches myself - is there anything else I can do to help?

infirit, 2019-06-04:
I have a pretty good idea what is going on and unless @cschramm beats me to it I'll have a fix later this week/weekend.

infirit, 2019-06-04:
Does the below avoid the problem? The curious bit is that even if the adapter is not powered it shows up for me on dbus. So we should always be able to get a valid adapter from ```bluez.Manager.get_adapter(s)``` :confused:.

What ``check_bluetooth_status`` does is check if a bluetooth adapter is powered/blocked and offer the user to power/unblock it.

```diff
diff --git a/blueman/Functions.py b/blueman/Functions.py
index 3b76271e..746cc420 100644
--- a/blueman/Functions.py
+++ b/blueman/Functions.py
@@ -86,6 +86,7 @@ def check_bluetooth_status(message, exitfunc):
             return
 
     applet.SetBluetoothStatus('(b)', True)
+    sleep(1)
     if not applet.GetBluetoothStatus():
         print('Failed to enable bluetooth')
         exitfunc()
```

cschramm, 2019-06-05:
Looking at the code I'd guess that's kind of expected at that point when there is no adapter. :thinking:

infirit, 2019-06-06:
@suve does adding a 1 second delay change anything?

@cschramm Agreed, just not sure how it got there. I am contemplating just making ``blueman-manager``  act on Adapters showing up or disappearing like I always wanted to. But as we are in beta now should this go into 2.1 or 2.2?

- If no adapter is found on dbus show a label saying no adapters are found.
- If an adapter if found but not powered show all devices but make everything insensitive (also needs a button to toggle power state)

suve, 2019-06-07:
I do not observe the issue on my own system, and to check whether the 1 second delay changes anything I'd need to make an update to the Fedora package so the affected users can test it.

I'll see if there's an easy way for users to try this out locally without the need to install an update.

cschramm, 2019-06-07:
@infirit: I think what happens is the following:

`check_bluetooth_status` detects that bluetooth is disabled as that's `PowerManager`'s initial `adapter_state` when there are no adapters (set by `on_manager_state_changed`).

When `check_bluetooth_status` requests the activation `PowerManager` sets `adapter_state` to `True` so that when `check_bluetooth_status` requests the state again, bluetooth is reported to be enabled although nothing really changed except for `PowerManager`'s internal state.

`ManagerDeviceList` has a fallback for when no adapter is available that just leaves the `Adapter` property uninitialized but `ManagerStats` expect it to be set.

The suggested change to the manager behavior sounds good, but I'd schedule it for 2.2.

@suve: Do we have a blueman-applet log in one of the tickets? I did not find one.

infirit, 2019-06-07:
@cschramm During the process you describe I see all adapter and device objects on dbus so for me I should not be possible that ``ManagerStats`` has no adapter set. Do you see ``Adpater1`` object(s) when the adapter is not powered/blocked? If not ``check_bluetooth_status`` would have failed.

ps: the name of ``check_bluetooth_status`` is misleading as it does not really do that. A better name would have been ``check_adapter_status``. Anyway..

cschramm, 2019-06-08:
@infirit: The situation I'm thinking of is that there actually is no adapter, not just on DBus.

infirit, 2019-06-08:
Ah, yes can reproduce and #1039 will fix this.

closed 2019-06-08 by infirit