blueman-project/blueman #1041: Blueman does not recognize being unable to read a file

suve, 2019-06-07:
blueman: 2.1-beta1
BlueZ: 5.50
Distribution: Fedora 30
Desktop environment: LXDE

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

When attempting to send a file that the current user cannot read (e.g. it has `000` permissions), blueman does not recognize that it cannot read the file.

**Expected result:** The bluetooth connection is dropped and the GUI shows an appropriate error message.

**Actual result:** The bluetooth connection is initialized (the receiving device recognizes that a file is being sent) and just hangs indefinitely without any data being sent. Nothing happens GUI-side; the "Sending files via bluetooth" dialog just keeps displaying 0%.

Running `blueman-applet` in the terminal shows that there's an exception being thrown and not caught properly. Pastebin link: https://pastebin.com/raw/hyrTErgv

infirit, 2019-06-07:
The error us actually that we try to pass something to os.path which it does not understand. I fixed this in PR #1030 with the specific commit being https://github.com/blueman-project/blueman/pull/1030/commits/fd34cfac6b47f3b7c87b33132cc2ad3af9a41b23

infirit, 2019-06-07:
btw nice to see blueman has an active maintainer in fedora :+1:

infirit, 2019-06-08:
@suve merged #1030  so this should be fixed now in master and the next (beta) release.

closed 2019-06-08 by infirit