blueman-project/blueman #1045: Hotkey not Working: Ctrl+Q

matthiasbeyer, 2019-06-09:
blueman: 2.0.8
BlueZ: (How to find out / is this relevant?)
Distribution: NixOS
Desktop environment: Sway (Wayland)

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow: No, because not an technical issue but a user-experience-issue.

---

I would like to be able to quit blueman-manager with Cltr-Q, could you please add this feature?

infirit, 2019-06-09:
Most window manager have shortcuts to close a window, usually ctrl+f4.

Other than that blueman-manager does not have keyboard shortcuts which could be added if someone cares enough. Changed title and added enhancement.

matthiasbeyer, 2019-06-09:
Well, in other applications Ctrl-Q works for me,... so I'm not sure whether the window manager is actually responsible here.

Either way: Thanks for not simply closing this!

cschramm, 2020-10-14:
Can you please try #1406? :slightly_smiling_face:

matthiasbeyer, 2020-10-15:
I do not use wayland anymore, so I cannot test, sorry.

Feel free to close this issue.

PR #1406 (2020-10-15): Add standard window / application closing accelerators
Closes #1045
commit 21255a7b (2020-10-15): Add standard window / application closing accelerators

Closes #1045
closed 2020-10-15 by cschramm