blueman-project/blueman #1050: Blueman-manager can't access the secondary bluetooth adapter

adam900710, 2019-06-20:
blueman: blueman 2.1
BlueZ: 5.50
Distribution: Arch
Desktop environment: Cinnamon

Blueman-manager fails to access the secondary bluetooth adapter.

When selecting and switching to the secondary bluetooth adapter, I got the following traceback, and no device shows up.
```
$ blueman-manager 
blueman-manager version 2.1 starting
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 212, in on_finished
    super().device_remove_event(device, tree_iter)
TypeError: super(type, obj): obj must be an instance or subtype of type
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 212, in on_finished
    super().device_remove_event(device, tree_iter)
TypeError: super(type, obj): obj must be an instance or subtype of type
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 212, in on_finished
    super().device_remove_event(device, tree_iter)
TypeError: super(type, obj): obj must be an instance or subtype of type
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 212, in on_finished
    super().device_remove_event(device, tree_iter)
TypeError: super(type, obj): obj must be an instance or subtype of type
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 212, in on_finished
    super().device_remove_event(device, tree_iter)
TypeError: super(type, obj): obj must be an instance or subtype of type
```

After that traceback, no matter which adapter I choose, no device will show up anymore.

cschramm, 2019-06-20:
#1052 should fix the errors but I'm not sure how far that will get you. Did you try a discovery after switching?

infirit, 2019-06-20:
Yeah switching is broken, probably by me :sob:. I'll check this out this weekend.

adam900710, 2019-06-21:
@cschramm I applied the change to `ManagerDeviceList.py`, still the same.
And I just switch the adaptor, no discovery triggered.

That #1052 fixes the traceback, but still not enough to use the secondary adapter.

Anything else I need to test?

infirit, 2019-06-21:
Tracked it down to commit c67ae920c5bfd3932ec6715cc0336139b436f88b

@adam900710 I'll have a patch this weekend you can try out.

infirit, 2019-06-23:
@adam900710 fixes are in #1055. You can get a patch with all changes on https://patch-diff.githubusercontent.com/raw/blueman-project/blueman/pull/1055.diff

adam900710, 2019-06-23:
@infirit  Thank you very much! It works now!

However still a small problem.
Connect device normally has 3 signal bars, but after switching to another adapter and then switching back, connect device no longer shows the signal bar.

Just a small nitpick, as long as adapter switching is work I can't be more happier.
(The damn blueberry can't even support multiple adapters)

infirit, 2019-06-23:
@adam900710 that is not correct and I can reproduce. I'll have a look at and something to the PR for this.

infirit, 2019-06-23:
@adam900710 I pushed a commit c3810278df9e7f67ad9ad0c62b833fe7904e1e5f to handle the bars. You can use the same link to the patch.

@cschramm c3810278df9e7f67ad9ad0c62b833fe7904e1e5f is a bit ugly so if you have another suggestion feel free to comment on the PR.

adam900710, 2019-06-24:
With c381027 everything is fine!

Thanks again for the quick fix!

I'll leave this issue open until the fix get merged.

commit 1878bb86 (2019-07-02): DeviceList: Small refactor in set_adapter to fix switching adapters

fixes #1050
commit 5e02e2b8 (2019-07-02): DeviceList: Small refactor in set_adapter to fix switching adapters

fixes #1050
closed 2019-07-02 by infirit