blueman-project/blueman #1053: can't find adapter

Bob-gt, 2019-06-21:
blueman: 2.04
BlueZ: 5
Distribution: Mint 18.3
Desktop environment: Cinnamon
http://dpaste.com/06BK5HZ  (crash report and debug stuff)
uname-a    4.10.0-38-generic #42~16.04.1-Ubuntu 
lsusb (only bluetooth stuff)
Bus 003 Device 002: ID 0a12:0001 Cambridge Silicon Radio, Ltd Bluetooth Dongle (HCI mode)

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2019-06-21:
Did you check `bluetoothctl list`. Does it list the adapter?

Bob-gt, 2019-06-21:
No. Nothing is listed with that command.
I tried to edit the version of blueman but no luck. It should be 2.04-1

cschramm, 2019-06-22:
So the adapter is not working at all in your system. Searching for `0a12:0001` yields all kinds of issues, https://wiki.ubuntu.com/HardwareSupportComponentsBluetoothUsbAdapters, https://bugs.launchpad.net/ubuntu/+source/linux/+bug/1221995, https://bugzilla.kernel.org/show_bug.cgi?id=103451, https://bugzilla.kernel.org/show_bug.cgi?id=60824, ...

However, they all seem to relate to older kernels than what Linux Mint 18.3 should use (4.10, right?).

Check `dmesg` for messages regarding firmware or any errors. Also check `lsmod` for `btusb`.

Bob-gt, 2019-06-22:
lsmod:
btusb                  45056  0
uname-a    4.10.0-38-generic #42~16.04.1-Ubuntu 
The adapter works in Win 10, so the hardware is ok.
Was the crash report and debug useful?

cschramm, 2019-06-23:
Unfortunately not. The blueman output just looks like there's no adapter and `bt-adapter -i` seems to crash because there's no adapter. `bluetoothctl list` confirms that there's no adapter. So technically there's no adapter, probably because of some firmware, driver, whatever issue that you could possibly find in `dmesg` output.

Bob-gt, 2019-07-06:
dmesg | grep Bluetooth : 
[   13.903561] Bluetooth: Core ver 2.22
[   13.903578] Bluetooth: HCI device and connection manager initialized
[   13.903581] Bluetooth: HCI socket layer initialized
[   13.903583] Bluetooth: L2CAP socket layer initialized
[   13.903590] Bluetooth: SCO socket layer initialized
[   15.333427] Bluetooth: BNEP (Ethernet Emulation) ver 1.3
[   15.333428] Bluetooth: BNEP filters: protocol multicast
[   15.333431] Bluetooth: BNEP socket layer initialized

Any other info I can provide?
Thanks,

infirit, 2019-07-06:
@Bob-gt no, your dongle is not working with the kernel in your distribution. If there are no messages regarding this device in dmesg then there is not much we can do. Maybe ask on the mint forums for help.

closed 2019-07-06 by infirit