blueman-project/blueman #1056: info manager plugin traceback

infirit, 2019-06-23:
Just to make sure I don't forget to fix.

```python
blueman-manager 17.25.13 DEBUG    ManagerDeviceMenu:259 generate  : Sony PLAYSTATION(R)3 Controller
Traceback (most recent call last):
  File "/home/sander/repos/blueman/blueman/plugins/manager/Info.py", line 114, in <lambda>
    item.connect('activate', lambda x: show_info(device, manager_menu.get_toplevel()))
  File "/home/sander/repos/blueman/blueman/plugins/manager/Info.py", line 95, in show_info
    store.append((name, func(device.get(name))))
  File "/usr/lib/python3.7/site-packages/gi/overrides/Gtk.py", line 995, in append
    return self._do_insert(-1, row)
  File "/usr/lib/python3.7/site-packages/gi/overrides/Gtk.py", line 986, in _do_insert
    row, columns = self._convert_row(row)
  File "/usr/lib/python3.7/site-packages/gi/overrides/Gtk.py", line 898, in _convert_row
    result.append(self._convert_value(cur_col, value))
  File "/usr/lib/python3.7/site-packages/gi/overrides/Gtk.py", line 912, in _convert_value
    return GObject.Value(self.get_column_type(column), value)
  File "/usr/lib/python3.7/site-packages/gi/overrides/GObject.py", line 210, in __init__
    self.set_value(py_value)
  File "/usr/lib/python3.7/site-packages/gi/overrides/GObject.py", line 250, in set_value
    (py_value, type(py_value)))
TypeError: Expected string but got 0<class 'int'>
```

infirit, 2019-06-23:
For some reason the lambda for Class ``(lambda x: x and "0x{:06x}".format(x))`` returns an int :confused:

cschramm, 2019-06-23:
I guess it's meant to catch cases where no class info exists, i.e. x is None. The expression returns x if it's int 0 as that's false-ish.

infirit, 2019-06-24:
It will be zero if missing and never None. I'll open a PR with the above change.

closed 2019-07-03 by cschramm