blueman-project/blueman #1062: Adding a new device: Method "QueryPlugins" doesn't exist

suve, 2019-07-01:
blueman: 2.1
BlueZ: 5.50
Distribution: Fedora
Desktop environment: XFCE

* [x] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

This is a bug encountered by one of the users on Fedora. According to them, it occured when they clicked on bluetooth systray menu to try to add a new bluetooth device.

Backtrace:
```
  File "/usr/bin/blueman-assistant", line 270, in <module>
    Assistant(args)
  File "/usr/bin/blueman-assistant", line 52, in __init__
    check_bluetooth_status(_("Bluetooth needs to be turned on for the Bluetooth assistant to work"), lambda: exit())
  File "/usr/lib/python3.7/site-packages/blueman/Functions.py", line 71, in check_bluetooth_status
    if "PowerManager" not in applet.QueryPlugins():
  File "/usr/lib64/python3.7/site-packages/gi/overrides/Gio.py", line 295, in __call__
    None)
gi.repository.GLib.GError: g-dbus-error-quark: GDBus.Error:org.freedesktop.DBus.Error.UnknownObject: Method "QueryPlugins" with signature "" on interface "org.blueman.Applet" doesn't exist
 (41)
```

When the user attempted to reproduce the error with `--loglevel debug`, they encountered different exceptions being thrown. Debug log [**here**](https://bugzilla.redhat.com/attachment.cgi?id=1586301).

Original ticket in RedHat Bugzilla: https://bugzilla.redhat.com/show_bug.cgi?id=1723908

cschramm, 2019-07-01:
Hi @suve, thanks for the report. The backtrace looks like a packaging bug to me as most probably an instance of `blueman-applet` from the old version is still running after the upgrade so that the running old version of the service is incompatible with the starting new version of `blueman-manager`.

cschramm, 2019-07-01:
```
Traceback (most recent call last):
  File "/usr/lib/python3.7/site-packages/blueman/gui/manager/ManagerDeviceList.py", line 212, in on_finished
    super().device_remove_event(device, tree_iter)
TypeError: super(type, obj): obj must be an instance or subtype of type
```

looks like a separate, minor bug we should look into. I'll have a look.

cschramm, 2019-07-01:
The latter got reported in #1050 and is fixed in #1055. We'll do a 2.1.1 bug fix release soon.

closed 2019-07-01 by cschramm