blueman-project/blueman #1066: 2.1 connection issues

rworkman, 2019-07-04:
blueman: 2.1
BlueZ: 5.50
Distribution: Slackware -current (development tree)
Desktop environment: Unknown

This is a third-party report, as I can't reproduce it myself. After we upgraded from blueman-2.0.8 to blueman-2.1 in the -current tree, we got a few reports of unusable bt devices. I've thus far been unable to elicit a good bug report, but the gist of it is that the device pairs and connects but then almost immediately loses the connection.

It's unclear whether this is related, but another comment was "Info menu item doesn't display the info page for connected devices: no page appears at all" but I'm not entirely sure what that's referring to.

I built some test packages with the 2-1-stable commits (up to 68756c54bb7adf1c730b25b320716db6d0a5f211) but they reportedly did not fix the problem.
it along. I'm mainly opening this as a place for others to "me too" and hopefully someone who's actually experiencing the bug(s) will be able to provide more information for troubleshooting.
Sorry I can't be more specific; if I get better information, I'll pass

cschramm, 2019-07-04:
Hi Robby.

> Info menu item doesn't display the info page for connected devices: no page appears at all

I'm pretty sure that one should be fixed in your test packages. 2.1.1 is around the corner (#1065) with the fixes.

> device pairs and connects but then almost immediately loses the connection

That does not sound like a blueman issue but more like a pretty common phenomenon due to issues with BlueZ, drivers, remote devices, ...
I'm not familiar with the Slackware package system. Is there a chance that affected users upgraded BlueZ together with blueman? It should be rather easy to test if it's really a blueman 2.1 regression by just downgrading to 2.0.8 and seeing if the issues remain.

rworkman, 2019-07-04:
I'm waiting on a report from one of the testers to see if the info page is fixed. The other reporter (the one with connection issues) advised that the patched packages did not fix the problem, but reverting to 2.0.8 did fix it. The revert to 2.0.8 did not change any other package contents or status.

cschramm, 2019-07-04:
Interesting. What kinds of services are involved? A log from `blueman-applet` (see https://github.com/blueman-project/blueman/wiki/Troubleshooting#debugging) showing the connection attempt will probably be useful.

userx-bw, 2019-07-04:
I think I am the frist to report this issue. These are the files that are currently installed on my laptop using Slack --current, the only one I downgraded was blueman.

```
/var/log/packages/bluedevil-2.1.1-x86_64-4
/var/log/packages/blueman-2.0.8-x86_64-2
/var/log/packages/bluez-5.50-x86_64-2
/var/log/packages/bluez-firmware-1.2-x86_64-2
```

after downgrading to 2.0.8 it now works again, whereas the upgrade would connect for about a second then disconnect. Having to delete it from the list and redoing the pairing from the start again only for it to repeat itself connecting, pairing, then disconnecting.

rworkman, 2019-07-04:
@userx-bw, the patched 2.1 is here; please install it and provide the logs requested if the problem is still present: http://www.slackware.com/~rworkman/blueman-test/

userx-bw, 2019-07-04:
@rworkman  as they, slackware reverted back to the later version I cannot install a patch on it, can I? So leaves me with not being about to test it to give you feed back, sadly. 

This is a patch I used when it was installed, and it still did not make it work, but worse. The post on it in linuxquestions.org

[userx@slack.it.netters:~
$ ls /var/log/packages/blue*
/var/log/packages/bluedevil-2.1.1-x86_64-4
**/var/log/packages/blueman-2.1-x86_64-1_rlw**
/var/log/packages/bluez-5.50-x86_64-2
/var/log/packages/bluez-firmware-1.2-x86_64-2](url)

nope, it did the connect, then dropped it seconds afterwords.

userx-bw, 2019-07-04:
using the gui, set mouse to be seen:
search...
finds it.
select setup.. Nothing, no other window to start it to pair. 
after a few minutes it goes away off of main window.

using that trouble shooting list from @cschramm post

[$ bluetoothctl list
Controller 34:02:86:24:71:FA BlueZ 5.50 [default]

$ head -1 $(which blueman-applet)
#!/usr/bin/python3
](url)
take crash course in python to figure out how to start a python shell.

[$ python
Python 2.7.16 (default, Apr  3 2019, 21:30:55) 
[GCC 8.3.0] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from blueman.gui.Notification import Notification
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named blueman.gui.Notification
>>> Notification("Title", "Message", 0, [["action1", "Action 1"], ["action2", "Action 2"]]).show()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
NameError: name 'Notification' is not defined](url)

After this I have no idea off the top of my head.

alex14641, 2019-07-04:
I'm running the latest code from Git master. When I access the Info menu item, I get the following debug output:

Traceback (most recent call last):
  File "/usr/lib64/python3.7/site-packages/blueman/plugins/manager/Info.py", line 114, in <lambda>
    item.connect('activate', lambda x: show_info(device, manager_menu.get_toplevel()))
  File "/usr/lib64/python3.7/site-packages/blueman/plugins/manager/Info.py", line 93, in show_info
    store.append((name, device.get(name)))
  File "/usr/lib64/python3.7/site-packages/gi/overrides/Gtk.py", line 995, in append
    return self._do_insert(-1, row)
  File "/usr/lib64/python3.7/site-packages/gi/overrides/Gtk.py", line 986, in _do_insert
    row, columns = self._convert_row(row)
  File "/usr/lib64/python3.7/site-packages/gi/overrides/Gtk.py", line 898, in _convert_row
    result.append(self._convert_value(cur_col, value))
  File "/usr/lib64/python3.7/site-packages/gi/overrides/Gtk.py", line 912, in _convert_value
    return GObject.Value(self.get_column_type(column), value)
  File "/usr/lib64/python3.7/site-packages/gi/overrides/GObject.py", line 210, in __init__
    self.set_value(py_value)
  File "/usr/lib64/python3.7/site-packages/gi/overrides/GObject.py", line 250, in set_value
    (py_value, type(py_value)))
TypeError: Expected string but got {301: [4, 0, 1, 49, 0, 1, 225, 160, 147, 113, 0, 96, 208, 0, 0, 0, 0, 0, 0]}<class 'dict'>

Also, when I turn off the bluetooth headphones and turn them back on, blueman doesn't automatically reconnect.

infirit, 2019-07-05:
@alex14641 #1067 should take care of the traceback. If it does can you post the contents from the Info dialog for 'ManufacturerData', 'ServiceData' and 'AdvertisingData'. Not all may show up and I'll take anything you can provide.

alex14641, 2019-07-05:
@cschramm With the patch, I get no traceback, but still no dialog.

infirit, 2019-07-06:
I setup a slack current vm with @rworkman package. Outside the info plugin issue I can connect my wiiu gamepad without problem.

Start ``blueman-applet`` and ``blueman-manager`` from a terminal with ``--loglevel debug`` and try to do what is failing. Then post the console output here in a [gist](https://gist.github.com).

You may see a message that the protocol is not available which means something is not installed or configured properly. For example bluetooth audio for my phone appears to be not working in the vm I just created.

cschramm, 2019-07-07:
@userx-bw: So you have a paired mouse that you can connect (it's not about pairing, is it?) to in 2.0.8 but not in 2.1, right?

Please run `bluetoothctl devices` to find the address of the mouse and then `bluetoothctl connect <ADDRESS>` to connect to it. Do you get a stable connection when doing that a) if blueman 2.1 is running and b) if it is not?

Please stop the `blueman-applet` process and manually run `blueman-applet --loglevel debug` in your terminal with blueman 2.1, then execute the failing connection attempt and provide us with the terminal output.

alex14641, 2019-07-07:
> @cschramm With the patch, I get no traceback, but still no dialog.

Update: after uninstalling and reinstalling, I'm able to get the info page. The properties you asked for weren't available, But I figured I'd post the results just in case it might be useful.

00:0A:45:0B:92:75
public
ATH-M50xBT    
ATH-M50xBT    
0x240418
0x0000
audio-card
yes
yes
no
no
no
00001101-0000-1000-8000-00805f9b34fb Serial Port
00001108-0000-1000-8000-00805f9b34fb Headset
0000110a-0000-1000-8000-00805f9b34fb Audio Source
0000110b-0000-1000-8000-00805f9b34fb Audio Sink
0000110c-0000-1000-8000-00805f9b34fb Remote Control Target
0000110d-0000-1000-8000-00805f9b34fb Advanced Audio
0000110e-0000-1000-8000-00805f9b34fb Remote Control
0000111e-0000-1000-8000-00805f9b34fb Handsfree
00001124-0000-1000-8000-00805f9b34fb Human Interface Device Service (HID)
00001200-0000-1000-8000-00805f9b34fb PnP Information
bluetooth:v1000p0006d1001
/org/bluez/hci0

userx-bw, 2019-07-07:
> @userx-bw: So you have a paired mouse that you can connect (it's not about pairing, is it?) to in 2.0.8 but not in 2.1, right?
>
> Please run `bluetoothctl devices` to find the address of the mouse and [...]

My mouse at this time is connected via the gui blueman using 2.0.8 this is what that command gives me, the mouse in question is the first one, HP touch to pair mouse, 
```
$ bluetoothctl devices
Device F0:65:DD:74:C3:3E HP Touch to Pair Mouse
Device E4:E1:30:8D:D2:BC Alcatel A574BL
Device 30:21:00:F0:00:C8 CX9015
```
me swap out 2.0.8 with 2.1 and see what it does. 
installed 2.1 restarting bluetooth
```
Let # Homepage: https://github.com/blueman-project/blueman
Executing install script for blueman-2.1-x86_64-2_rlw.txz.
Package blueman-2.1-x86_64-2_rlw.txz installed.
Package blueman-2.0.8-x86_64-2 upgraded with new package ./blueman-2.1-x86_64-2_rlw.txz.
userx@slack.it.netters:~/Downloads
$ sudo  /etc/rc.d/rc.bluetooth restart
Starting Bluetooth services:  bluetoothd hciconfig sdptool hciattach
```
The mouse at this point is staying connected, but the blueman-apple no longer shows itself when the icon in the system tray is clicked on. I get nothing, not even a flash of the window opening then disappearing. 
that is a change in the prder of the devices on 2.1 , I see it now has HP second on the list. 
```
$ bluetoothctl devices
Device E4:E1:30:8D:D2:BC Alcatel A574BL
Device F0:65:DD:74:C3:3E HP Touch to Pair Mouse
Device 30:21:00:F0:00:C8 CX9015
```
DMESG
```

I assume input27 is before I replaced driver the second one input29 is after I updated to 2.1 I did not look until afterwards. 

[   69.576141] input: HP Touch to Pair Mouse Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:256/0005:03F0:044C.0003/input/input27
[   69.576338] hid-generic 0005:03F0:044C.0003: input,hidraw2: BLUETOOTH HID v1.08 Mouse [HP Touch to Pair Mouse] on 34:02:86:24:71:fa

[  117.084546] EXT4-fs (sdc1): mounted filesystem with ordered data mode. Opts: (null)
[  117.258671] EXT4-fs (mmcblk0p1): mounted filesystem with ordered data mode. Opts: (null)
[ 1817.692750] hid-generic 0005:03F0:044C.0004: unknown main item tag 0x0

[ 1817.692945] input: HP Touch to Pair Mouse Mouse as /devices/pci0000:00/0000:00:14.0/usb1/1-4/1-4:1.0/bluetooth/hci0/hci0:256/0005:03F0:044C.0004/input/input29
[ 1817.693225] hid-generic 0005:03F0:044C.0004: input,hidraw2: BLUETOOTH HID v1.08 Mouse [HP Touch to Pair Mouse] on 34:02:86:24:71:fa

```
Killing blueman-apple and stating it in the terminal - disconnected mouse deleted it, then put mouse in discover mode to try and re-connect mouse, this is the debug info

7.693225] hid-generic 0005:03F0:044C.0004: input,hidraw2: BLUETOOTH HID v1.08 Mouse [HP Touch to Pair Mouse] on 34:02:86:24:71:fa
userx@slack.it.netters:~/Downloads
$ 
userx@slack.it.netters:~/Downloads
$ 
userx@slack.it.netters:~/Downloads
$ blueman-applet  --loglevel debug
blueman-applet version 2.1 starting
blueman-applet 14.19.27 INFO     PluginManager:85 load_plugin: ['NMPANSupport', 'ExitItem', 'DiscvManager', 'DBusService', 'NetUsage', 'AuthAgent', 'NMDUNSupport', 'DhcpClient', 'KillSwitch', 'Networking', 'StandardItems', 'PowerManager', 'GameControllerWakelock', 'TransferService', 'StatusIcon', 'Menu', 'RecentConns', 'PPPSupport', 'ShowConnected', 'SerialManager']
blueman-applet 14.19.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 14.19.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 14.19.27 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Address': '34:02:86:24:71:FA', 'AddressType': 'public', 'Name': 'BlueZ 5.50', 'Alias': 'BlueZ 5.50', 'Class': 0, 'Powered': True, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['0000110e-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001133-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '00001112-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0532'}
blueman-applet 14.19.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 14.19.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 14.19.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 14.19.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 14.19.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 14.19.27 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 14.19.27 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
[... cut ...]
/usr/bin/blueman-assistant:219: PyGTKDeprecationWarning: Using positional arguments with the GObject constructor has been deprecated. Please specify keyword(s) for "label" or use a class specific constructor. See: https://wiki.gnome.org/PyGObject/InitializerDeprecations
  rbs.append(Gtk.RadioButton(_("Don't connect")))
blueman-applet 14.20.07 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Discovering': False}
blueman-applet 14.20.07 DEBUG    DiscvManager:82 on_adapter_property_changed: prop Discovering False
blueman-applet 14.20.11 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0/dev_F0_65_DD_74_C3_3E {'ServicesResolved': False, 'Connected': False}

rworkman, 2019-07-07:
@userx-bw, how about making a fresh start of the system with my 2.1 package installed? Unpair/"forget" all bt devices, restart the computer, and begin the process fresh. Provide logs from that and/or report success/failure.

userx-bw, 2019-07-07:
> @userx-bw, how about making a fresh start of the system with my 2.1 package installed? Unpair/"forget" all bt devices, restart the computer, and begin the process fresh. Provide logs from that and/or report success/failure.

It did as before, no change, connects, mouse still does not move cursor, then it disconnects a few seconds later. 

$ blueman-applet --loglevel debug
blueman-applet version 2.1 starting
blueman-applet 18.22.21 INFO     PluginManager:85 load_plugin: ['NMPANSupport', 'ExitItem', 'DiscvManager', 'DBusService', 'NetUsage', 'AuthAgent', 'NMDUNSupport', 'DhcpClient', 'KillSwitch', 'Networking', 'StandardItems', 'PowerManager', 'GameControllerWakelock', 'TransferService', 'StatusIcon', 'Menu', 'RecentConns', 'PPPSupport', 'ShowConnected', 'SerialManager']
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Menu.Menu'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'>
blueman-applet 18.22.21 DEBUG    Base:74 do_g_properties_changed: /org/bluez/hci0 {'Address': '34:02:86:24:71:FA', 'AddressType': 'public', 'Name': 'BlueZ 5.50', 'Alias': 'BlueZ 5.50', 'Class': 1835276, 'Powered': True, 'Discoverable': True, 'DiscoverableTimeout': 0, 'Pairable': True, 'PairableTimeout': 0, 'Discovering': False, 'UUIDs': ['0000110e-0000-1000-8000-00805f9b34fb', '0000110a-0000-1000-8000-00805f9b34fb', '00001200-0000-1000-8000-00805f9b34fb', '00001112-0000-1000-8000-00805f9b34fb', '00001108-0000-1000-8000-00805f9b34fb', '00001133-0000-1000-8000-00805f9b34fb', '0000110c-0000-1000-8000-00805f9b34fb', '0000112f-0000-1000-8000-00805f9b34fb', '00001132-0000-1000-8000-00805f9b34fb', '0000110b-0000-1000-8000-00805f9b34fb', '00001106-0000-1000-8000-00805f9b34fb', '00001104-0000-1000-8000-00805f9b34fb', '00001800-0000-1000-8000-00805f9b34fb', '00001105-0000-1000-8000-00805f9b34fb', '00001801-0000-1000-8000-00805f9b34fb'], 'Modalias': 'usb:v1D6Bp0246d0532'}
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DBusService.DBusService'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ExitItem.ExitItem'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'>
blueman-applet 18.22.21 WARNING  PluginManager:147 __load_plugin: Not loading DhcpClient because its conflict has higher priority
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.PowerManager.PowerManager'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.KillSwitch.KillSwitch'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.Networking.Networking'>
blueman-applet 18.22.21 INFO     Networking:31 load_nap_settings: Loading NAP settings
blueman-tray version 2.1 starting
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.StandardItems.StandardItems'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.TransferService.TransferService'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.RecentConns.RecentConns'>
blueman-applet 18.22.21 WARNING  PluginManager:147 __load_plugin: Not loading PPPSupport because its conflict has higher priority
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'>
blueman-applet 18.22.21 INFO     PluginManager:156 __load_plugin: loading <class 'blueman.plugins.applet.SerialManager.SerialManager'>
blueman-applet 18.22.21 INFO     KillSwitch:106 io_event  : killswitch registered 1
blueman-applet 18.22.21 INFO     KillSwitch:122 io_event  : State: True
blueman-applet 18.22.21 INFO     PowerManager:172 update_power_state: off False | foff False | on True | current state True | new state True
blueman-applet 18.22.21 INFO     KillSwitch:87 _on_connman_vanished: net.connman vanished
blueman-applet 18.22.21 INFO     Applet:51 _on_dbus_name_appeared: org.bluez :1.1
blueman-applet 18.22.21 INFO     BluezAgent:97 register_agent: Register Agent
blueman-applet 18.22.21 INFO     Networking:67 set_nap   : set nap False
blueman-applet 18.22.21 INFO     RecentConns:91 initialize: rebuilding menu
blueman-applet 18.22.21 INFO     TransferService:210 _on_dbus_name_appeared: org.bluez.obex :1.23
blueman-applet 18.22.21 INFO     ShowConnected:38 enumerate_connections: Found 0 existing connections
blueman-applet 18.22.21 INFO     AgentManager:15 on_registered: /org/bluez/obex/agent/blueman
blueman-applet 18.22.22 INFO     PowerManager:123 request_power_state: Requesting True
[... cut ...]
Device 7B:75:DD:74:9B:14 7B-75-DD-74-9B-14
Device 6D:4E:7E:E2:A9:F0 6D-4E-7E-E2-A9-F0
Device 73:0C:28:88:AE:3B 73-0C-28-88-AE-3B
Device 6C:28:03:DC:B0:45 6C-28-03-DC-B0-45
Device 6E:D2:81:24:27:02 6E-D2-81-24-27-02
Device 53:16:D0:F7:73:6C 53-16-D0-F7-73-6C
Device 43:B3:B8:E0:0F:FF 43-B3-B8-E0-0F-FF
Device 56:AF:A2:96:24:E9 56-AF-A2-96-24-E9
Device 6E:7C:A3:BD:1D:25 6E-7C-A3-BD-1D-25
Device 40:D7:B5:E4:95:2B 40-D7-B5-E4-95-2B
Device 5F:B1:DA:09:75:88 5F-B1-DA-09-75-88
Device 48:74:CB:57:FC:EA 48-74-CB-57-FC-EA
```

cschramm, 2019-07-10:
@userx-bw

If connecting is the issue, please keep the mouse paired all the time.

Starting with a paired mouse which you do not unpair / remove at any time try connecting with `bluetoothctl connect F0:65:DD:74:C3:3E` and see if you get a working connection with...

a) blueman-applet 2.1 running.
b) blueman-applet 2.0.8 or not running at all.

userx-bw, 2019-07-10:
$ ls /var/log/packages/blue*
/var/log/packages/bluedevil-2.1.1-x86_64-4
/var/log/packages/blueman-2.1-x86_64-2_rlw
/var/log/packages/bluez-5.50-x86_64-2
/var/log/packages/bluez-firmware-1.2-x86_64-2

blueman 2.1
$ bluetoothctl connect F0:65:DD:74:C3:3E
Attempting to connect to F0:65:DD:74:C3:3E
Failed to connect: org.bluez.Error.Failed

$ ls /var/log/packages/blue*
/var/log/packages/bluedevil-2.1.1-x86_64-4
/var/log/packages/blueman-2.0.8-x86_64-2
/var/log/packages/bluez-5.50-x86_64-2
/var/log/packages/bluez-firmware-1.2-x86_64-2

blueman-applet 2.0.8

$ bluetoothctl connect F0:65:DD:74:C3:3E
Attempting to connect to F0:65:DD:74:C3:3E
Failed to connect: org.bluez.Error.Failed

userx-bw, 2019-07-10:
OK that post did not update my edit, so belay what that says.... it connected. I just switched versions so this is going to be backwards as I just put 2.0.8 in again, and my mouse it working. 

using the command line with 2.1 and putting mouse into pair mode it connected.

userx-bw, 2019-07-12:
senise I've been back in Slack, I've created a script to connect my mouse via 2.1 and well, if the system sleeps/screen goes black, and upon hitting the space bar, my mouse is no longer connected, having to use the script which is nothing more than the command to connect it. this is the latest when I booted into slack

```
$ connmouse
Attempting to connect to F0:65:DD:74:C3:3E
[CHG] Device F0:65:DD:74:C3:3E Connected: yes
[CHG] Device F0:65:DD:74:C3:3E Paired: yes

[CHG] Device F0:65:DD:74:C3:3E Paired: no
[CHG] Device F0:65:DD:74:C3:3E Connected: no
Failed to connect: org.bluez.Error.Failed

```

I'm going back to 2.0.8

alex14641, 2019-07-14:
Sorry for the late update. I've installed blueman 2.1.1, and the Info menu item now works.

@infirit 

My other issue was a reconnect problem, not a connect problem, but it turns out that was a problem with the headphones: I had updated the firmware at the same time I update blueman to 2.1. Reconnecting works with another set of headphones.

closed 2019-09-19 by stale[bot]