blueman-project/blueman #1069: Files sent to computer with blueman end up in wrong folder

insomniacno1, 2019-07-06:
blueman: 2.0.4
BlueZ: bluez (5.37-0ubuntu5.1) xenial-security; urgency=medium
Distribution: Peach OSI 16.04 LTS 64bit (Ubuntu 16.04.6 LTS) Xubuntu
Kernel: Linux 4.4.0-154-generic
Desktop environment: XFCE 4 Gnome
Creator: https://www.peachosi.com/ and can be contacted by sending a message here https://www.facebook.com/peachosi

* [ ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

infirit, 2019-07-06:
This means hat blueman is not handling the incomming file transfer but gnome-bluetooth is. Upon receiving a file blueman-applet will show it on the console and print some debug info. Most likely you do not see this so below is the start of a transfer I did just now. Yours may look different as you are on a different version.

```
blueman-applet 15.24.12 INFO     Manager:64 _on_object_added: /org/bluez/obex/server/session1
blueman-applet 15.24.12 INFO     Manager:56 _on_object_added: /org/bluez/obex/server/session1/transfer0
blueman-applet 15.24.12 DEBUG    Transfer:25 do_g_properties_changed: /org/bluez/obex/server/session1/transfer0 Status queued
blueman-applet 15.24.12 DEBUG    Transfer:25 do_g_properties_changed: /org/bluez/obex/server/session1/transfer0 Session /org/bluez/obex/server/session1
blueman-applet 15.24.12 DEBUG    Transfer:25 do_g_properties_changed: /org/bluez/obex/server/session1/transfer0 Transferred 0
```

insomniacno1, 2019-07-06:
Okay, not being an expert here, but what is the solution to this?

infirit, 2019-07-06:
Do not run gnome-bluetooth and blueman at the same time. Either disable gnome-bluetooth in xfce session 's autostart application or uninstall gnome-bluetooth completely.

insomniacno1, 2019-07-07:
Okay, the strange part is that this behaviour only started this year around februar - That is when I noticed a file I sent by bluetooth to my computer, disapeared - later I found it in /home/user/.cache/obexd/ . 

I got the Local services in blueman-applet and blueman-manager set to Downloads and it worked for a while, until it didn't. Now it was starting to annoy me and I went looking on the net for a solution which is described in first post - which worked, until it didn't. 

Later I noticed it happen just after an update. Before that, files sent by Bluetooth went into the folder I had set in Local services, which was Public or Pictures or Downloads. 

I never saw the issue on Peach OSI 14.04 LTS and never had it in 2018 on 16.04 LTS.

And I checked gnome-bluetooth is installed(from distributor), but not running.

So are you sure about your diagnose or could it be something else, like a bug in blueman? 
As a general rule I don't uninstall something that other things might rely on.

After all blueman-applet runs on startup and in Local services it is set to Downloads(or whatever folder I have chosen) so according to blueman settings the files received should go to Downloads - but they don't, they go to /home/user/.cache/obexd/

I'm sorry but for me, not being a programmer but troubleshooter and since this issue only started in february, that looks like an programming error in either blueman or any of the other parts it relies on - not something(gnome-bluetooth) that has been there forever. 

So are there any part of this that just got an update and might cause this behaviour or could it be in the kernel itself? 

I just got updated from 4.4.0-151-generic to 4.4.0-154-generic - and in between the fix in the first post and the update, the files went to right place - Can't say that it happen with all updates, but the problem go back to february and there has been alot of updates in between.

With kind regards

Jan

cschramm, 2019-07-07:
As @infirit said, blueman and gnome-bluetooth compete to handle incoming transfers and it seems like gnome-bluetooth wins. You could check by looking at blueman's output, as @infirit described. If you tried a transfer during in the above log, it did not get handled by blueman.

insomniacno1, 2019-07-07:
@infirit could you give me the command you used to make the log as it do not show in your answer.

Thank you

insomniacno1, 2019-07-22:
Did this thread die?

cschramm, 2019-07-22:
Yes, you did not provide further information or questions. The relevant information is in https://github.com/blueman-project/blueman/issues/1069#issuecomment-508926407. @infirit most probably ran `blueman-applet --loglevel debug` and is on newer version but you should see similar output regarding the file transfer like I said in https://github.com/blueman-project/blueman/issues/1069#issuecomment-509003853.

insomniacno1, 2019-07-23:
Actually I did ask @infirit  one more time to provide the command he used, so I could try.
And I told you that I would love to try the command but was waiting for @infirit 

Both of these lasted a few minutes to some hours and got deleted from this thread - don't ask me why or how, but they were there when I closed the page.

insomniacno1, 2019-07-23:
So I ran the command but didn't get the same as @infirit so don't think that was the command he used

user@user:~$ blueman-applet --loglevel debug
blueman-applet version 2.0.4 starting
Stale PID, overwriting
_________
Load (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:60)
['StandardItems', 'PPPSupport', 'ShowConnected', 'NetUsage', 'Menu', 'RecentConns', 'KillSwitch', 'Networking', 'AuthAgent', 'ExitItem', 'NMDUNSupport', 'DBusService', 'PowerManager', 'StatusIcon', 'SerialManager', 'AppIndicator', 'DhcpClient', 'DiscvManager', 'NMPANSupport', 'Headset', 'TransferService', 'GameControllerWakelock'] 
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/BlueZInterface.py:13)
Detected BlueZ 5 
/usr/lib/python3/dist-packages/blueman/plugins/applet/AppIndicator.py:8: PyGIWarning: AppIndicator3 was imported without specifying a version first. Use gi.require_version('AppIndicator3', '0.1') before import to ensure that the right version gets loaded.
  from gi.repository import AppIndicator3 as girAppIndicator
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StatusIcon.StatusIcon'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AuthAgent.AuthAgent'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.GameControllerWakelock.GameControllerWakelock'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Networking.Networking'> 
_________
load_nap_settings (/usr/lib/python3/dist-packages/blueman/plugins/applet/Networking.py:36)
Loading NAP settings 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.ShowConnected.ShowConnected'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Menu.Menu'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.AppIndicator.AppIndicator'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.Headset.Headset'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DBusService.DBusService'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMPANSupport.NMPANSupport'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.NMDUNSupport.NMDUNSupport'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.DiscvManager.DiscvManager'> 
_________
update_menuitems (/usr/lib/python3/dist-packages/blueman/plugins/applet/DiscvManager.py:123)
warning: Adapter is None 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.PowerManager.PowerManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.SerialManager.SerialManager'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.StandardItems.StandardItems'> 
_________
__load_plugin (/usr/lib/python3/dist-packages/blueman/main/PluginManager.py:133)
loading <class 'blueman.plugins.applet.TransferService.TransferService'> 
_________
get_interface_version (/usr/lib/python3/dist-packages/blueman/bluez/obex/Base.py:20)
Detected BlueZ integrated obexd 
[... 206 lines cut ...]
adding action
_________
_on_properties_changed (/usr/lib/python3/dist-packages/blueman/bluez/obex/Transfer.py:54)
/org/bluez/obex/server/session15/transfer14 Status complete 
_________
_on_session_removed (/usr/lib/python3/dist-packages/blueman/bluez/obex/Manager.py:57)
/org/bluez/obex/server/session15 
_________
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_70_3A_51_96_27_4B 
_________
Destroy (/usr/lib/python3/dist-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_70_3A_51_96_27_4B 
_________
__init__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:26)
caching initial properties 
_________
__del__ (/usr/lib/python3/dist-packages/blueman/main/Device.py:72)
deleting device /org/bluez/hci0/dev_70_3A_51_96_27_4B 
_________
Destroy (/usr/lib/python3/dist-packages/blueman/main/Device.py:94)
invalidating device /org/bluez/hci0/dev_70_3A_51_96_27_4B

insomniacno1, 2019-07-23:
For me it looks like its ONLY blueman that handles this and not gnome-bluetooth

infirit, 2019-07-23:
@insomniacno1 that is exactly what we need to see what is going on. The output is different as the current version 2.1 has diverged a lot and that is what I am using.

From a terminal what is the output of ``gsettings get org.blueman.transfer shared-path``.  The output will show you what path blueman will use to move the incoming file to. If this is not what you expect run ``blueman-services`` and configure it in the "Transfer" section.

insomniacno1, 2019-07-23:
@infirit I did everything as you asked and it is now set to Downloads folder and the files go there. But as I explained before, I did this before and just after any update it was back to the obexd folder eventhough it was set to Downloads in blueman-services. I will leave it like this and then get back to you after next update, if it changes again.

infirit, 2019-07-23:
You did not share what the gsettings command returned, an empty string, a path to your cache dir or something completely different? If you still have the output post it otherwise include this when you report back :smile: .

You can monitor changes to this setting with ``gsettings monitor org.blueman.transfer shared-path``.

insomniacno1, 2019-07-24:
@infirit gsettings get org.blueman.transfer shared-path shows whatever is configured in blueman-services whether that is obexd or Downloads - right now '/home/th/Downloads'

gsettings monitor org.blueman.transfer shared-path does nothing, it doesn't even go back to prompt, it just freeze the cursor so that Ctrl + C is only option.

I updated yesterday and it seems to be stable for now, so I will say its solved.

Thank you for the help :-)

With kind regards

Jan

closed 2019-07-24 by cschramm

iago-pssjd, 2021-04-26:
Hi @cschramm @infirit 

I am having a similar issue. If I run `gsettings monitor org.blueman.transfer shared-path` it does not return anything (it hangs up), and if I run `gsettings get org.blueman.transfer shared-path`, if I do it as a user, then I get `/home/user/Downloads` (as I configure through GUI), but if I run it as root I get `''`.  Can I configure this the Transfer Incoming folder through command line? How shouls I do it?

Thanks!

cschramm, 2021-05-06:
@iago-pssjd: If `gsettings get org.blueman.transfer shared-path` gives you the expected path, it's already configured. If it does not work, please open a new issue with the output from `blueman-applet --loglevel debug` when receiving a file.

iago-pssjd, 2021-05-07:
That's it, my version is exactly 2.0.4