blueman-project/blueman #1070: Too many devices found, which one is mine?

PenelopeFudd, 2019-07-08:
blueman: 2.0.5
BlueZ: 5.48
Distribution: Ubuntu 18.04
Desktop environment: Gnome

* [ x ] I have consulted the [Troubleshooting page](https://github.com/blueman-project/blueman/wiki/Troubleshooting) and done my best effort to follow.

cschramm, 2019-07-10:
I don't think there's much we can do about that from a technical perspective (i.e. I'm not aware of a way to resolve the name proactively).

If your "air space" is that crowded, the best idea I have is to start the search, wait for most of the other devices to appear and only then make your device visible (by powering it on, putting it into pairing mode or whatever) so that it is (one of) the last to appear.

PenelopeFudd, 2019-09-08:
Is there a way to assign names or otherwise mark the other devices prior to turning mine on?

Is there a way to rapidly ping (e.g. once a second) a particular device?  Then I can put my device in the refrigerator or some other faraday cage, or take the battery out, and instantly see the signal cut off.

I just don't know enough about bluetooth to know what is/is not possible.  If there was just a bluetooth cheat sheet showing the things that can be done and the equivalent in wifi or ethernet terminology, I might be better able to help myself.  For instance, is there a way to use something like tcpdump with bluetooth?  How about a ping command?

Thanks.

infirit, 2019-09-10:
You can try to rename the device in blueman-manager if you know the mac address. It may not be remembered by BlueZ but worth a try. These devices probably expose a name in GATT but it is not exposed through the BlueZ dbus api.

BlueZ has some tools to do scans and one of them is ``hcitool lescan`` (may need to run as root). However this also does not show me any names for these devices however my Android phone does.

It sucks but until BlueZ exposes this in the dbus api we can't do any about this.

closed 2019-11-16 by stale[bot]